britkat1980/giv_tcp #205: GivTCP-Beta 2.4.642 update issue

rupi08, 2024-08-13:
updated from 2.4.324 to 2.4.642, after restart no options are available in configuration tab, just has title "Options". 
In log nothing happens after the following:
2024-08-12 22:51:59,595 - startup - [INFO] - ======= Self Run is off, so no data collection is happening =======
[2024-08-12 22:51:59 +0100] [76] [INFO] Starting gunicorn 22.0.0
[2024-08-12 22:51:59 +0100] [76] [INFO] Listening at: http://0.0.0.0:6345 (76)
[2024-08-12 22:51:59 +0100] [76] [INFO] Using worker: sync
[2024-08-12 22:51:59 +0100] [77] [INFO] Booting worker with pid: 77
[2024-08-12 22:51:59 +0100] [78] [INFO] Booting worker with pid: 78
[2024-08-12 22:51:59 +0100] [79] [INFO] Booting worker with pid: 79

britkat1980, 2024-08-15:
As per the log line you need to open up the web gui config page, check settings and then turn on Self-Run then it will connect. Settings are no longer under options, but web based. (to allow for auto discovery)

rupi08, 2024-08-17:
tried to update to v2.4.651, got following error : 
2024-08-16 21:19:31,834 - startup - [INFO] - Setting up invertor: 1
Traceback (most recent call last):
  File "/app/startup.py", line 557, in <module>
    createsettingsjson(inv)
  File "/app/startup.py", line 175, in createsettingsjson
    outp.write("    inverter_type=\""+str(setts["Model_"+str(inv)])+"\"\n")
                                          ~~~~~^^^^^^^^^^^^^^^^^^^
KeyError: 'Model_1'

britkat1980, 2024-08-17:
update to Beta656 and this should be solved now

sammort, 2024-08-18:
Updating to beta656 and I see this error instead:

```
2024-08-18 07:35:22,368 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-08-18 07:35:24,368 - startup - [INFO] - Searching for Inverters
2024-08-18 07:35:24,369 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-08-18 07:35:50,595 - startup - [INFO] - No inverters found...
2024-08-18 07:35:50,600 - startup - [INFO] - Setting up invertor: 1
Traceback (most recent call last):
  File "/app/startup.py", line 548, in <module>
    createsettingsjson(inv)
  File "/app/startup.py", line 155, in createsettingsjson
    setts["Model_"+str(inv)]= inverter_type['Model'].name.capitalize()
                              ~~~~~~~~~~~~~^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

Running as a standalone container not a home assistant addon.

poolski, 2024-08-21:
Interestingly, after I cleared out all of the `.pkl` files and extraneous other bits, the addon started up properly. I'd installed the BETA version alongside the existing GivTCP addon which I think was causing the issue.

closed 2024-08-21 by britkat1980