britkat1980/giv_tcp #206: Import/Export values on dashboard

louismckee19, 2024-08-16:
Hi all,

Have an issue where the import and export energy readings show differently within the energy dashboard then they do as an MQTT entity as shown below:

[image]
[image]

Happens with both total and today values.

Any thoughts?

Thanks!

poolski, 2024-08-21:
Some observations just based on those two screenshots:

- The MQTT entity seems to be a couple orders of magnitude out - nearly 4.9MwH? How big is your array?!
- The graph shows point-in-time values for the import energy over time but the Energy Dashboard shows the **totals** so the value is "correct" from its perspective.

I would suggest that you need to check what's going on with your MQTT entities and why they're reporting _thousands_ of kWh rather than _tens_ or even single-digit kWh readings.

poolski, 2024-08-26:
Hmm, actually I'm able to reproduce this now with Beta 2.4.698 

[image]
I'm getting very strange Energy Dashboard spikes after running 2.4.698 

[image]
[image]

poolski, 2024-08-27:
Interestingly, if I swap the `_today_kwh` battery sensors for their `total_kwh` counterparts, the issue goes away.

britkat1980, 2024-08-29:
I always use the Total energy entities in the dashboard as HA expects an increasing total and it determines its own daily values from that.

norfolkmustard, 2024-09-24:
I'm seeing some oddities in the lifetime numbers too now, regularly jumps to exactly 5,648.6kWh

[image]

With some spikes in the today values

[image]

On beta
[image]

poolski, 2024-09-26:
I've been seeing a lot of spikes in the battery charge/discharge sensors recently, which can really throw the statistics off once they're calculated: 

[image]

norfolkmustard, 2024-09-26:
I've since found the alternative entities for "battery stack" a lot more stable

sensor.givtcp_battery_stack_1_stack_charge_energy_today_kwh

sensor.givtcp_battery_stack_1_stack_discharge_energy_today_kwh

E.g. the purple spikes don't have equivalents in the battery stack 1 entity

[image]

poolski, 2024-09-26:
That's interesting - I don't have any sensors apart from **BMS Temperature** and **BMS Voltage** on Battery Stack 1...

norfolkmustard, 2024-09-26:
Suggested earlier too, that the total/lifetime entities are perhaps better to use for the energy dashboard, and the ones in the battery_stack_1 area also seem stable. 

sensor.givtcp_battery_stack_1_stack_charge_energy_total_kwh

sensor.givtcp_battery_stack_1_stack_discharge_energy_total_kwh

e.g.

[image]

poolski, 2024-09-26:
Hmm, clearing the MQTT messages didn't do anything. Still only have Temp + Voltage sensors for the battery. I wonder if it's to do with the battery itself. 

MQTT doesn't show any stack charge/discharge stats for the battery at all, only those two

norfolkmustard, 2024-09-26:
Hmm. Mine's an All-in-one gen1 /6kW

Givtcp has one area per battery module (so x 4. HX99 - HX102 below) and one for the combined stack. 

[image]

These entities are in stack 1

[image]

poolski, 2024-09-26:
Mine's a Gen3 Hybrid 5.5kW + 9.5kWh battery separated, so that may be why.

gcoan, 2024-09-26:
Having used the main battery charge/discharge sensors in my energy dashboard for ages I would prefer to not swap to different battery stack sensors as that'll lose all the existing history I have in the dashboard.

I've gone back to 2.4.9 for now with the teething troubles on v3, but I think my battery stack only has temperature and voltage.  Gen 1 hybrid so I think that might be why, the AIO has different data

poolski, 2024-09-26:
For what it's worth, I think the Energy Dashboard will backfill if the "new" sensor has historic data 🤷🏻

gcoan, 2024-09-26:
Yes you are absolutely right the dashboard will back fill with the long term statistics as long as they exist for the sensor, my comment about the issue of new sensors was related to the 'battery stack' charge/discharge sensors which I thought were new in v3, and then they wouldn't have had any history with them.

Anyway, for me, moot point as I think these are AIO sensors and I have a Gen 1 hybrid.

I have been experimenting with adding the total sensors to the energy dashboard and seeing what they look like.  Using these sensors has one immediate advantage that it fills in gaps in my data when I've had comms issues with the inverter.

Overall it seems that I get far fewer spikes than with the today sensors although there are about one a month from December through to April this year,  but nothing more recently 🤷‍♂️
Also notice that the spikes are often around midnight suggesting there is/was a problem with counts not being reset properly in some way
[image]

I'll play around a bit more but given the small number I should be able to fix these data spikes, and battery charge/discharge totals look to be the way to go.

gcoan, 2024-09-27:
there is *definitely* a correlation between rebooting Home Assistant and spikes appearing in the GivTCP MQTT sensors.

I upgraded from HA 2024.9.2 to 2024.9.3 tonight and afterwards noticed that my house load sensor which is calculated based on battery charge, discharge, import and export data had jumped up.

Looking at the underlying data sensors from both of my inverters, at the point of HA restart:
- import energy spiked downwards and then recovered back unchanged
- export energy was flat throughout no change on either inverter

[image]

- battery charge today on both inverters jumped up, one by 6.2kWh, the other by 3.7kWh
- battery discharge today on one inverter jumped up by 0.9kWh and the other remained constant

[image]

I have 'retain MQ data' turned on now in GivTCP but that doesn't seem to have avoided the issue.  Whilst I can and will swap over to using the total sensors for the energy dashboard, I need the today figures to be accurate for my daily house load sensor.

norfolkmustard, 2024-09-27:
You could create new daily-resetting utility meter helper entities from the _total_kwh to generate daily values. 

https://www.home-assistant.io/integrations/utility_meter

[image]

gcoan, 2024-09-28:
> You could create new daily-resetting utility meter helper entities from the _total_kwh to generate daily values

Yeah I was thinking exactly the same thought myself yesterday evening.  I could create daily utility meter sensors for the key GivTCP entities that I need to calculate daily house load and use the appropriate GivTCP total sensor as the source.

May well go down this route; just need to decide whether to keep just 1 day of history of these sensors so I can calculate today's house load each day,  or whether I use these as a long term replacement for my daily battery charge, discharge, solar, etc sensors and then move (or recreate) all the long term sensor statistics over onto the new sensors.

Oh, and one side issue I have found with adding the total kWh sensors to the energy dashboard is that the displayed title is the full 'total kwh' entity name which is a bit misleading.  There isn't the option in the energy dashboard to overwrite the battery or solar sensor names as you can now do for the detailed item consumption sensors on the dashboard.
Could of course rename the sensor names in HA but then it'd be a bit confusing to have 'inverter today', 'inverter total' and 'utility meter today' sensors all named similar things.

Of course if I do create a 'utility meter today' sensor from the total then this could be used in the energy dashboard as well, but would need the long term statistics moving over to be useful.

What a pain.  Does feel like a lot of mucking around to cover up an issue with the underlying GivTCP today sensors

britkat1980, 2024-10-24:
Fixed in 3.0.4

closed 2024-10-24 by britkat1980