britkat1980/giv_tcp #207: FileNotFoundError: [Errno 2] No such file or directory: '/config/GivTCP/allsettings.json'

ProximusAl, 2024-08-17:
Latest docker is throwing this:

Traceback (most recent call last):
  File "/app/startup.py", line 449, in <module>
    shutil.copyfile("settings.json",SFILE)
  File "/usr/local/lib/python3.11/shutil.py", line 258, in copyfile
    with open(dst, 'wb') as fdst:
         ^^^^^^^^^^^^^^^

ProximusAl, 2024-08-17:
Just to be clear, this is on giv_tcp-ma docker version 2.4.3 from docker hub. 
I’ve rolled back to 2.4.2 and all working again

britkat1980, 2024-08-17:
That's very odd, 2.4.3 is production and that code snippet is from Dev, no idea why that's happened! I'll need to roll back 2.4.3 somehow

dandwhelan, 2024-08-18:
Yeah, I've got the same issue. It keeps looping around with the same error. I also have the correct Invertor IP Address within my config file, but the addon insists on searching my network for the inverter.

Container stopped
18/08/2024
09:20:54
Container started
2024-08-18 09:20:54,917 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-08-18 09:20:56,918 - startup - [INFO] - Searching for Inverters
2024-08-18 09:20:56,918 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-08-18 09:21:22,324 - startup - [INFO] - No inverters found...
Traceback (most recent call last):
  File "/app/startup.py", line 449, in <module>
    shutil.copyfile("settings.json",SFILE)
  File "/usr/local/lib/python3.11/shutil.py", line 258, in copyfile
    with open(dst, 'wb') as fdst:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/config/GivTCP/allsettings.json'
18/08/2024
09:21:22
Container stopped

stefanwest1, 2024-08-18:
Same issue here. Rolling back to 2.4.2 fixes it

ChavasRegal, 2024-08-18:
how can i roll back to 2.4.2 when adding on home assistant via adons?

britkat1980, 2024-08-18:
I have re-pushed the original 2.4.3 build to 2.4.7, so you should see an update option in Addons so you can install 2.4.4.

Apologies for this mistake, not sure how it happened!

ProximusAl, 2024-08-18:
Just updated to 2.4.7 MA and it seems to be working.

ProximusAl, 2024-08-18:
Resolved with 2.4.7

closed 2024-08-18 by ProximusAl

gcoan, 2024-08-18:
> I have re-pushed the original 2.4.3 build to 2.4.7, so you should see an update option in Addons so you can install 2.4.4.
> 
> Apologies for this mistake, not sure how it happened!

@britkat1980 I'm using HASS add-on store.  Saw the givtcp 2.4.7 'release' and at the top of the release note says that the future v3 release will be pushed to https://github.com/britkat1980/ha-addons so we need to the add on store, but I just get an error when I try to do this:

[image]

and then tried again:

[image]

So am lost as to whether I will get the v3 release or not?

britkat1980, 2024-08-18:
You’ve got it there already. It’s the one that says GivTCP-TEST. I’ll rename it once I push v3 in the coming days