britkat1980/giv_tcp #208: EVC not connecting - Missing Serial Number

APoller1, 2024-08-17:
Added an EVC to my Gen 1 inverter running old firmware. 
It gives no serial number error for the EVC.

Does the EVC only work with the new firmware?
When I set the inverter to new firmware the inverter gets 'no serial number ' errors.

Checked IP address.

APoller1, 2024-08-17:
Tried again - Now get connection refused.

2024-08-17 15:36:12,471 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-08-17 15:36:12,481 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-08-17 15:36:14,496 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-08-17 15:36:18,944 - startup - [CRITICAL] - Inverter SD2244G430 which is a Gen 1 - Hybrid has been found at: 192.168.1.112
2024-08-17 15:36:18,946 - startup - [INFO] - Searching for Inverters again
2024-08-17 15:36:18,958 - startup - [CRITICAL] - Running Redis
2024-08-17 15:36:18,965 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/08/17 15:36:18 [notice] 69#69: using the "epoll" event method
2024/08/17 15:36:18 [notice] 69#69: nginx/1.20.2
2024/08/17 15:36:18 [notice] 69#69: OS: Linux 6.6.31-haos-raspi
2024/08/17 15:36:18 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/08/17 15:36:18 [notice] 69#69: start worker processes
2024/08/17 15:36:18 [notice] 69#69: start worker process 74
2024/08/17 15:36:18 [notice] 69#69: start worker process 75
2024/08/17 15:36:18 [notice] 69#69: start worker process 76
2024/08/17 15:36:18 [notice] 69#69: start worker process 77
2024-08-17 15:36:19,028 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-08-17 15:36:19,029 - startup - [CRITICAL] - Running Invertor (192.168.1.112) read loop every 10s
2024-08-17 15:36:19,036 - startup - [CRITICAL] - Running EVC read loop every 10s
2024-08-17 15:36:19,044 - startup - [CRITICAL] - Subscribing MQTT Broker for EVC control
2024-08-17 15:36:19,076 - startup - [CRITICAL] - Setting chargeMode loop to manage different charge modes every 60s
2024-08-17 15:36:19,077 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-08-17 15:36:19,096 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2024-08-17 15:36:19 +0100] [84] [INFO] Starting gunicorn 21.2.0
[2024-08-17 15:36:19 +0100] [84] [INFO] Listening at: http://0.0.0.0:6345 (84)
[2024-08-17 15:36:19 +0100] [84] [INFO] Using worker: sync
[2024-08-17 15:36:19 +0100] [85] [INFO] Booting worker with pid: 85
[2024-08-17 15:36:19 +0100] [86] [INFO] Booting worker with pid: 86
[2024-08-17 15:36:19 +0100] [87] [INFO] Booting worker with pid: 87
2024-08-17 15:36:21,873 - Inv1 - mqtt_client_evc -  [CRITICAL] - Connecting to MQTT broker for EVC control- core-mosquitto
2024-08-17 15:36:22,172 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.102, 502) failed: [Errno 111] Connection refused
2024-08-17 15:36:22,939 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2024-08-17 15:36:27,631 - Inv1 - read        -  [CRITICAL] - First time running so saving AC Charge status
2024-08-17 15:36:27,635 - Inv1 - read        -  [CRITICAL] - Publishing Home Assistant Discovery messages
2024-08-17 15:36:32,183 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.102, 502) failed: [Errno 111] Connection refused
2024-08-17 15:36:42,191 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.102, 502) failed: [Errno 111] Connection refused
2024-08-17 15:36:52,204 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.102, 502) failed: [Errno 111] Connection refused
2024-08-17 15:37:02,219 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.102, 502) failed: [Errno 111] Connection refused
2024-08-17 15:37:12,231 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.102, 502) failed: [Errno 111] Connection refused

britkat1980, 2024-08-17:
Connection refused, which means either the ip address is wrong, or your evc is connected via ethernet. Unfortunately modbus is only open on wifi.

APoller1, 2024-08-17:
It's on Wifi and it's the correct IP Address - I turned it off to see it go off line.
Though Give don't give it a name you can find or provide the MAC address on the case!

britkat1980, 2024-08-17:
OK, have you enabled Local Control on the portal? That needs to be turned on to open the modbus port. (Security requirement)

APoller1, 2024-08-17:
Yep - Is there an easy way to check I have the correct IP address?
I turned it off and it disconnected previously.
------------------------------
*From:* britkat1980 ***@***.***>
*Sent:* 17 August 2024 17:19
*To:* britkat1980/giv_tcp ***@***.***>
*Cc:* APoller1 ***@***.***>; Author ***@***.***>
*Subject:* Re: [britkat1980/giv_tcp] EVC not connecting - Missing Serial
Number (Issue #208)

OK, have you enabled Local Control on the portal? That needs to be turned
on to open the modbus port. (Security requirement)

britkat1980, 2024-08-17:
If you can install the beta version it should find it for you, which will confirm it’s the right one!
Add https://github.com/britkat1980/ha-addons to the addon store repository in HA and then you can install the beta version.

APoller1, 2024-08-19:
Ran the current version and it gave me an error.
Loaded the Beta and it found the EVC, but didn't load it, once I'd selected to load it, it worked.
Stopped the Beta
Ran the current version worked fine, must be something missing in the set up...
Thanks for all your help - Resolved.
Let me know where to send a coffee?

closed 2024-08-19 by APoller1