britkat1980/giv_tcp #209: Beta 2.4.669 - Controls don't seem to actually work

Gavin-Carpenter, 2024-08-19:
Not entirely sure with which beta version this problem started with but for the last few I've been unable to set controls properly in HA.

For example if I toggle "enable discharge schedule" to off in my dashboard it turns off for a few seconds but then turns back on again. 

Can't seem to see any errors or details in either HA or the add-on logs.

The non-beta version works fine.

Held off reporting this as thought it was something already aware of and would be fixed shortly.

Thanks for making and maintaining this add-on. Everything else works great as far as I've seen other than the EVC not staying connected which I think is a firmware issue and the rare weird spike/drop in some of the energy stats which I just filter out.

britkat1980, 2024-08-19:
What inverter do you have?

Gavin-Carpenter, 2024-08-20:
Model: All-In-One
FW Version: D0.612-A0.612
BMS Version: 10

Model: Gateway
FW Version: A0.008

FYI just tried beta 679 out of interest and get this error in the logs and no entities updating:

Error in inital detect/refresh: ('TypeError', 'read.py', 46)

britkat1980, 2024-08-21:
Beta693 should work now

Gavin-Carpenter, 2024-08-22:
Yup, all looks to be working now, can see set success messages in the logs, etc.

Thanks for the quick fix!

closed 2024-08-22 by Gavin-Carpenter