britkat1980/giv_tcp #213: GivTCP entity sensor state class total_increasing, but its state is not strictly increasing.

rholligan, 2024-08-24:
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:303
integration: Sensor (documentation, issues)
First occurred: 23 August 2024 at 10:25:10 (5 occurrences)
Last logged: 23 August 2024 at 10:30:10

Entity sensor.givtcp_xxx_self_consumption_energy_today_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 6.2 (6.3) with last_updated set to 2024-08-23T09:23:51.674930+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.givtcp_xxx_load_energy_today_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 2.5 (2.6) with last_updated set to 2024-08-23T09:23:51.676623+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.givtcp_xxx_load_energy_total_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 5756.4 (5756.5) with last_updated set to 2024-08-23T09:23:51.691124+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.givtcp_xxx_self_consumption_energy_total_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 1645.4 (1645.5) with last_updated set to 2024-08-23T09:23:51.693740+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.givtcp_xxx_soc_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 5.333 (5.428) with last_updated set to 2024-08-23T09:26:57.431117+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

gcoan, 2024-08-24:
I think this is fixed in the next release or the beta version. There was a prior github issue reporting this

gcoan, 2024-08-24:
See previous issues #195 and #199

britkat1980, 2024-08-29:
Fixed in dev builds and slated for v3 release

closed 2024-08-29 by britkat1980