britkat1980/giv_tcp #214: PV Automated Load Manager has stopped doing anything

stevedundee2, 2024-08-26:
I have GivTCP running in Home Assistant and configured to use Solar Forecast Automation. Since 17th August 2024 it has stopped doing anything. When it runs at 23.30 each day, the log shows only the "PV Automated Load Manager" line and the target SOC doesn't get changed (from 77%).

The current GivTCP log has just 2 lines at 23.30:

> 2024-08-19 23:20:20,639 - palm_soc - [INFO] - PALM... PV Automated Load Manager: v1.1.0SoC
> 2024-08-19 23:20:20,639 - palm_soc - [INFO] - Timestamp: 19-08-2024 23:20:20 +0100

On and before 17th August there were multiple lines:

> 2024-08-17 23:20:02,538 - palm_soc - [INFO] - PALM... PV Automated Load Manager: v1.1.0SoC
> 2024-08-17 23:20:02,539 - palm_soc - [INFO] - Timestamp: 17-08-2024 23:20:02 +0100
> 2024-08-17 23:20:02,539 - palm_utils - [INFO] - Processing load history for day -1
> 2024-08-17 23:20:03,183 - palm_utils - [INFO] - Processing load history for day -2
> 2024-08-17 23:20:03,834 - palm_utils - [INFO] - Processing load history for day -3
> 2024-08-17 23:20:04,516 - palm_utils - [INFO] - Processing load history for day -4
> 2024-08-17 23:20:05,694 - palm_utils - [INFO] - Processing load history for day -5
> 2024-08-17 23:20:06,451 - palm_utils - [INFO] - Processing load history for day -6
> 2024-08-17 23:20:07,161 - palm_utils - [INFO] - Processing load history for day -7
> 2024-08-17 23:20:07,999 - palm_utils - [INFO] - Load Calc Summary: [0.1, 0.2, 0.7, 0.7, 1.5, 1.8, 1.5, 0.8, 1.1, 1.1, 0.2, 0.0, 0.1, 0.1, 0.0, 0.2, 0.2, 0.1, 0.1, 0.1, 0.1, 0.2, 0.2, 0.2, 0.5, 0.3, 0.3, 0.3, 0.4, 0.4, 0.4, 0.3, 0.2, 0.2, 0.2, 0.1, 0.3, 0.2, 0.2, 0.2, 0.1, 0.2, 0.1, 0.1, 0.1, 0.0, -0.0, 0.0]
> 2024-08-17 23:20:09,034 - palm_utils - [INFO] - No second array
> 2024-08-17 23:20:09,035 - palm_utils - [INFO] - Successful Solcast download.
> 2024-08-17 23:20:09,036 - palm_utils - [INFO] - Applying BST offset to Solcast data
> 2024-08-17 23:20:09,046 - palm_utils - [INFO] - PV Estimate 10% (hrly, 7 days) / kWh; 17-08-2024 23:20:09; [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.028, 0.056, 0.1, 0.161, 0.227, 0.285, 0.324, 0.357, 0.387, 0.422, 0.465, 0.498, 0.525, 0.551, 0.577, 0.594, 0.592, 0.566, 0.516, 0.478, 0.455, 0.401, 0.33, 0.259, 0.19, 0.112, 0.052, 0.018, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0][9.864, 5.221, 0.0, 0.0, 0.0, 0.0]

... and so on.

Oddly these logs are missing an event which I think is the cause of this issue. There was an HA OS update around the 18th. That evening the target SOC was set to 100 (I think) and the next night it set it to 77%, where it has remained. But no GivTCP logs show this.

My Home Assistant is running on a Raspberry Pi:

- Core 2024.8.3
- Supervisor 2024.08.0
- Operating System 13.1
- Frontend 20240809.0

My GivTCP addon is 2.4.7 from repo https://github.com/britkat1980/giv_tcp. It has been this for a while.

gcoan, 2024-08-26:
Might be worth turning Debug mode on in GivTCP config and see what that says in the logs.

My guess is that Solcast is rejecting the API call.

There's two reasons for this:
1. Solcast has (a couple of months ago) started introducing rate limiting for hobbyist accounts, returning response code 429 if it's too busy.  The solution that's been adopted in the Solcast integration is to delay for a random period of time and then retry.  Sometimes it can take up to 10 minutes of pauses and retries before Solcast will accept the API call

2. Solcast has removed the GetUserAllowance endpoint.  This happened a couple of weeks ago so could well be why the Palm integration has suddenly stopped polling Solcast.
See https://github.com/springfall2008/batpred/issues/1362 and https://github.com/BJReplay/ha-solcast-solar/discussions/128#discussioncomment-10335517 including a reply from Solcast themselves that this API call was actually deprecated a while ago but only removed very recently.

Short term you could install the Solcast integration and poll solcast directly, then with your own automations charge the battery.  Or use Predbat.
Recognise either of these are a big change from just using GivTCP

stevedundee2, 2024-08-26:
Hi

Thanks for replying. You could be right, though the log lines before the Solcast download relate to my local load I think and they are missing too. 

Any idea how I switch on debug mode for GivTCP? I can't see a switch.

My Givenergy system also has Solcast configured. That seems to work. I have no idea how it works though (eg whether it calls Solcast only when I look or opn a schedule or whatever). I wonder whether it is using up an allowance.

gcoan, 2024-08-26:
Debug mode for GivTCP is in the configuration of the add-on:

[image]

If you have Solcast configured in the GivEnergy dashboard then that will be consuming your API calls as well.  Old accounts have a limit of 50 calls a day, newer (last few years) the limit is 10.
Whenever you display the solar forecast on the givenergy dashboard it will make an API call, or 2 if you have two arrays (sites) defined.  Refreshing the page will do the same so you can quickly run out of API calls.

cjkendo, 2024-08-28:
Hi

I’m not sure if it is related but I am also having issues with the PALM element of the add on. 

This last error on the log is:
  [INFO    ] - Timestamp: 28-08-2024 23:20:06 +0100
Traceback (most recent call last):
  File "/app/GivTCP_1/palm_soc.py", line 105, in <module>
    inverter: GivEnergyObj = GivEnergyObj()
  File "/app/GivTCP_1/palm_utils.py", line 92, in __init__
    self.cmd_list = stgs.GE_Command_list['data']
AttributeError: module 'palm_settings' has no attribute 'GE_Command_list'

The rest of the add on appears to be working as expect. 
I have tried restarting, deletion and reinstall all to no success. This is only since updating to the 2.4.7.

stevedundee2, 2024-08-30:
Interesting. I turned on debug mode for 5 minutes either side of the run a couple of nights ago. Hopefully I'll get time to look at it soon.

stevedundee2, 2024-09-01:
> Hi
> 
> I’m not sure if it is related but I am also having issues with the PALM element of the add on.
> 
> This last error on the log is: [INFO ] - Timestamp: 28-08-2024 23:20:06 +0100 Traceback (most recent call last): File "/app/GivTCP_1/palm_soc.py", line 105, in inverter: GivEnergyObj = GivEnergyObj() File "/app/GivTCP_1/palm_utils.py", line 92, in **init** self.cmd_list = stgs.GE_Command_list['data'] AttributeError: module 'palm_settings' has no attribute 'GE_Command_list'
> 
> The rest of the add on appears to be working as expect. I have tried restarting, deletion and reinstall all to no success. This is only since updating to the 2.4.7.

I see that message about GE_Command_list too, but only when I view the log from within the Addon config. There seems to be no signs of it in the rotated log files.

GivTCP 2.4.7 seems a bit odd. It isn't listed as a release in github and when you click to the change history in the add on config in HA version 2.4.7 seems to have the wrong date.  I see there is a v3 coming up, with its own new repo. I wonder whether the developer is just over stretched and we just need to be patient for the next release?

Though... I'd really like to restore to 2.4.3 (what happened to 2.4.4 etc?) to see whether that fixes things. Any idea how?

Meantime, my debug log lines have mysteriously disappeared. I'll try that again tonight and study them tomorrow evening.

gcoan, 2024-09-01:
> GivTCP 2.4.7 seems a bit odd. It isn't listed as a release in github and when you click to the change history in the add on config in HA version 2.4.7 seems to have the wrong date

2.4.7 was issued to fix a docker release problem I think. It's code wise the same as 2.4.3. 
No idea why there is a gap in versions but if it's not working for you then no issue to restore to the earlier version. V3 is coming as you say

iainmoonie, 2024-09-03:
Hi folks - having the same issue. The SOC has stuck at 81% since 21st August, seems to be the same issue as @stevedundee2 at 23.20 when it usually sets the SOC for the following day.

----

2024-09-02 23:20:56,076 - Inv1 - palm_soc    -  [INFO    ] - PALM... PV Automated Load Manager: v1.1.0SoC
2024-09-02 23:20:56,077 - Inv1 - palm_soc    -  [INFO    ] - Timestamp: 02-09-2024 23:20:56 +0100
Traceback (most recent call last):
  File "/app/GivTCP_1/palm_soc.py", line 105, in <module>
    inverter: GivEnergyObj = GivEnergyObj()
  File "/app/GivTCP_1/palm_utils.py", line 92, in __init__
    self.cmd_list = stgs.GE_Command_list['data']
AttributeError: module 'palm_settings' has no attribute 'GE_Command_list'

----

I've tried turning off the 'old firmware' button as have done updates to GIVTCP to 2.4.7 and updates to the GivEnergy firmware in recent days (i think?)

[image]

stevedundee2, 2024-09-07:
I think I have to give up with debug logs. I run them, I can see the [DEBUG] lines, I go to sleep (the actions I am debugging happen at 11.20pm after all), and when I come back to them they are gone. No rotated logs contain them (or are even big enough to contain them).

@iainmoonie did that switch work? 

@gcoan any idea how I can revert to 2.4.3? 

Perhaps I just need to wait for v3. Does anyone know how long that is likely to be? I appreciate it will be ready when it's ready, but how close is it? Meantime I am pretty good at guessing a reasonable value based on the met office weather forecast.

iainmoonie, 2024-09-07:
My amateur diagnosis did not work... unbelievably 😅

I've switched to Predbat and happy that I have tbh, it's got so much more functionality.

It's not such a big switch in fairness given that you can now install Predbat via HA add-on, you've already got solcast API set up, all I required was the octopus integration.

stevedundee2, 2024-09-08:
Predbat sounds interesting. I'm heading up Ben Chonzie today, but I'm looking forward to giving it a try this evening.

stevedundee2, 2024-09-11:
I've installed predbat and am spending some time getting to grips with it. Which mode do you use it in? I tried "charge and discharge" but my myenergi devices see the forced discharges as solar excess and mop up the power which reduces the income that predbat is aiming for. I'm going to ask about that on the predbat forums.

Thanks for the pointer!

gcoan, 2024-09-11:
I'd suggest starting off with predbat set to read only when you're getting used to its plan. Most people Normally use Charge and Discharge mode but you can use just Charge if you don't want predbat to discharge the battery. 

The myeddi, ideally you need to set it to a mode so it's not monitoring the export apart from the periods where you want it to. Predbat can control the myeddi if you want it to

salewis38, 2024-09-25:
> Hi folks - having the same issue. The SOC has stuck at 81% since 21st August, seems to be the same issue as @stevedundee2 at 23.20 when it usually sets the SOC for the following day.
> 
> 2024-09-02 23:20:56,076 - Inv1 - palm_soc - [INFO ] - PALM... PV Automated Load Manager: v1.1.0SoC 2024-09-02 23:20:56,077 - Inv1 - palm_soc - [INFO ] - Timestamp: 02-09-2024 23:20:56 +0100 Traceback (most recent call last): File "/app/GivTCP_1/palm_soc.py", line 105, in inverter: GivEnergyObj = GivEnergyObj() File "/app/GivTCP_1/palm_utils.py", line 92, in **init** self.cmd_list = stgs.GE_Command_list['data'] AttributeError: module 'palm_settings' has no attribute 'GE_Command_list'
> 
> I've tried turning off the 'old firmware' button as have done updates to GIVTCP to 2.4.7 and updates to the GivEnergy firmware in recent days (i think?)
> 
> ![image](https://private-user-images.githubusercontent.com/41291062/363865171-a2257e43-91e8-4710-8c2d-571f6ca1aee4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyODgyNjUsIm5iZiI6MTcyNzI4Nzk2NSwicGF0aCI6Ii80MTI5MTA2Mi8zNjM4NjUxNzEtYTIyNTdlNDMtOTFlOC00NzEwLThjMmQtNTcxZjZjYTFhZWU0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI1VDE4MTI0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNkMzRlMGE5ZjYyN2MwZTE1YjRkYzNiZjgxY2EwMWYyN2I4ZTJhMDYzNzhkMmU4YjQ3NzExODcxN2JhMmNhMjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mTz0XJjB7lZXcJso8Bz5FQFHEax92fi4eIHK4_LAwaE)

There is a missing entry in the latest palm_settings.py file. It's a simple fix and I'll push the correction.