britkat1980/giv_tcp #215: Battery_Charge_Rate not being sent to MQTT

crg-n, 2024-08-26:
Hi, apologies if this is known or intentional.

On the latest beta and dev there seems to be some kind of issue with the "Battery_Charge_Rate" not being output on readData. PredBat is crashing because it is not able to find this in its REST request. Here is the output of manually reading the data from the same readData point that PredBat calls, with the "Control" exercpt below:

`"Control": { "Active_Power_Rate": 100, "Battery_Calibration": "Off", "Battery_Charge_Rate_AC": 100, "Battery_Discharge_Rate": 6000, "Battery_Discharge_Rate_AC": 100, "Battery_Power_Cutoff": 4, "Battery_Power_Reserve": 4, "Battery_pause_mode": "Disabled", "Charge_Target_SOC_1": 100, "Charge_Target_SOC_10": 100, "Charge_Target_SOC_2": 100, "Charge_Target_SOC_3": 100, "Charge_Target_SOC_4": 100, "Charge_Target_SOC_5": 100, "Charge_Target_SOC_6": 100, "Charge_Target_SOC_7": 100, "Charge_Target_SOC_8": 100, "Charge_Target_SOC_9": 100, "Discharge_Target_SOC_1": 4, "Discharge_Target_SOC_10": 4, "Discharge_Target_SOC_2": 4, "Discharge_Target_SOC_3": 4, "Discharge_Target_SOC_4": 4, "Discharge_Target_SOC_5": 4, "Discharge_Target_SOC_6": 4, "Discharge_Target_SOC_7": 4, "Discharge_Target_SOC_8": 4, "Discharge_Target_SOC_9": 4, "Eco_Mode": "enable", "Enable_Charge_Schedule": "disable", "Enable_Discharge_Schedule": "disable", "Force_Charge": "Normal", "Force_Charge_Num": 0, "Force_Export": "Normal", "Force_Export_Num": 0, "Mode": "Eco", "Reboot_Addon": "disable", "Reboot_Invertor": "disable", "Target_SOC": 100, "Temp_Pause_Charge": "Normal", "Temp_Pause_Charge_Num": 0, "Temp_Pause_Discharge": "Normal", "Temp_Pause_Discharge_Num": 0 }`

Weirdly "Battery_Discharge_Rate" is still there so I'm not sure what is happening. I'm on a GivEnergy AIO with Gateway.
Startup logs are:

`2024-08-26 22:35:39,154 - startup - [INFO] - ========================== STARTING GivTCP================================
2024-08-26 22:35:41,165 - startup - [INFO] - Searching for Inverters
2024-08-26 22:35:41,165 - startup - [INFO] - Scanning network for GivEnergy Devices...
2024-08-26 22:36:00,729 - startup - [INFO] - Inverter CH2344G*** which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.32
2024-08-26 22:36:01,898 - startup - [INFO] - Inverter GW2335G*** which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.33
2024-08-26 22:36:01,905 - startup - [INFO] - Inverter CH2344G*** already found in settings file (slot 1), checking IP address is unchanged...
2024-08-26 22:36:01,907 - startup - [INFO] - Inverter GW2335G*** already found in settings file (slot 2), checking IP address is unchanged...
2024-08-26 22:36:01,907 - startup - [INFO] - Setting up invertor: 1
2024-08-26 22:36:01,923 - startup - [INFO] - ==============================================================
2024-08-26 22:36:01,923 - startup - [INFO] - ====             Web Gui Config is at                     ====
2024-08-26 22:36:01,923 - startup - [INFO] - ====     http://192.168.*.*:****/config.html             ====
2024-08-26 22:36:01,923 - startup - [INFO] - ==============================================================
2024-08-26 22:36:01,923 - startup - [INFO] - Running Invertor 1 (CH2344G***) read loop every 30/120s
[2024-08-26 22:36:02 +0100] [77] [INFO] Starting gunicorn 22.0.0
[2024-08-26 22:36:02 +0100] [77] [INFO] Listening at: http://0.0.0.0:**** (77)
[2024-08-26 22:36:02 +0100] [77] [INFO] Using worker: sync
[2024-08-26 22:36:02 +0100] [80] [INFO] Booting worker with pid: 80
[2024-08-26 22:36:02 +0100] [81] [INFO] Booting worker with pid: 81
[2024-08-26 22:36:02 +0100] [82] [INFO] Booting worker with pid: 82
2024-08-26 22:36:02,492 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-08-26 22:36:02,495 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-08-26 22:36:12,193 - Inv1 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2024-08-26 23:27:01,326 - Inv1 - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2024-08-26 23:30:31,572 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of night rate tariff (Dynamic)`

I don't believe this is an issue on my end but I could be mistaken. Cheers.

gcoan, 2024-08-27:
Think this has been reported in in #186 (predbat issue https://github.com/springfall2008/batpred/issues/1380)

crg-n, 2024-08-27:
Hi @gcoan, I did see that issue but I couldn't find any mention of anyone having this specific problem in there. No mention of "Battery_Charge_Rate" which is the only problem I seem to have. It seemed to be more about readings, calculations and control of 3 phase inverters. It could definitely be an issue related to similar changes on the inverter end, but I wasn't confident enough it was related to tag onto that issue.

britkat1980, 2024-08-27:
This is an error in Dev Version of GivTCP, fixed in 716 (to be tested). I think it only happened when charge rate is set to Zero. Install 716 and confirm fix

crg-n, 2024-08-27:
Fix confirmed! Many thanks

closed 2024-08-27 by crg-n