britkat1980/giv_tcp #216: 3 phase commands testing. Enable_Charge_Schedule

LeeFarm11, 2024-08-28:
Hi.
Thought I would try to work through some of the commands at IPADDRESS:8099/REST1/api using GivTCP-Dev.
First command on the list is enableChargeSchedule.
On 3 phase I'm not sure what this command does because the schedule seems always active. In the Giv Portal to use the schedule to charge, I set the start and stop times and set Enable Force Charge and Enable AC Charge to On.

I clicked it ON at 09:33:00.
At approx 09:34:25 it moved itself to OFF.

```
2024-08-28 09:32:54,788 - evc - [DEBUG] - Publish all to MQTT
2024-08-28 09:32:54,789 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-08-28 09:33:00,341 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= enable
2024-08-28 09:33:00,504 - read - [DEBUG] - Write Request recieved
2024-08-28 09:33:00,505 - read - [DEBUG] - Command: enableChargeSchedule was sent: {'state': 'enable'}
2024-08-28 09:33:00,506 - write - [DEBUG] - Enabling Charge Schedule
2024-08-28 09:33:00,648 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G049/Control/Enable_Charge_Schedule - enable
2024-08-28 09:33:00,650 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-08-28 09:33:00,670 - write - [DEBUG] - Pushing control update to pkl cache: Enable_Charge_Schedule - enable
2024-08-28 09:33:00,672 - write - [DEBUG] - Enabling Charge Schedule was a success
2024-08-28 09:33:00,673 - write - [INFO] - Enabling Charge Schedule was a success
2024-08-28 09:33:00,674 - read - [DEBUG] - responses is: [{'id': 'enableChargeSchedule', 'result': '{"result": "Enabling Charge Schedule was a success"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Enabling Charge Schedule was a success"}'}]
2024-08-28 09:33:04,802 - evc - [DEBUG] - Socket is already open
2024-08-28 09:33:04,805 - evc - [DEBUG] - Socket is still open
2024-08-28 09:33:04,817 - evc - [DEBUG] - Dealt with Charger
2024-08-28 09:33:04,817 - evc - [DEBUG] - Publish all to MQTT
2024-08-28 09:33:04,818 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-08-28 09:33:09,509 - read - [DEBUG] - Running partial refresh
2024-08-28 09:33:11,980 - read - [DEBUG] - Data get was successful, now running handler if needed: 
2024-08-28 09:33:11,980 - read - [DEBUG] - Running processData
2024-08-28 09:33:11,982 - read - [DEBUG] - Beginning parsing of Inverter data
2024-08-28 09:33:11,982 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-08-28 09:33:12,003 - read - [DEBUG] - Getting Battery Details
2024-08-28 09:33:12,006 - read - [DEBUG] - Building battery output: 
2024-08-28 09:33:12,006 - read - [DEBUG] - Battery HY2343G302 added
2024-08-28 09:33:12,007 - read - [DEBUG] - Building battery output: 
2024-08-28 09:33:12,008 - read - [DEBUG] - Battery HY2343G257 added
2024-08-28 09:33:12,008 - read - [DEBUG] - Building battery output: 
2024-08-28 09:33:12,009 - read - [DEBUG] - Battery HY2343G446 added
2024-08-28 09:33:12,009 - read - [DEBUG] - Building battery output: 
2024-08-28 09:33:12,010 - read - [DEBUG] - Battery HY2343G025 added
2024-08-28 09:33:12,010 - read - [DEBUG] - Building battery output: 
2024-08-28 09:33:12,011 - read - [DEBUG] - Battery HY2343G259 added
2024-08-28 09:33:12,011 - read - [DEBUG] - Building battery output: 
2024-08-28 09:33:12,012 - read - [DEBUG] - Battery HY2343G144 added
2024-08-28 09:33:12,013 - read - [DEBUG] - Getting mode control figures
2024-08-28 09:33:12,014 - read - [DEBUG] - Calculating Mode...
2024-08-28 09:33:12,014 - read - [DEBUG] - Mode is: Eco (Paused)
2024-08-28 09:33:12,015 - read - [DEBUG] - Force Export is not Running
2024-08-28 09:33:12,015 - read - [DEBUG] - Getting TimeSlot data
2024-08-28 09:33:12,016 - read - [DEBUG] - Getting TimeSlot data
2024-08-28 09:33:12,019 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-08-28 09:33:12,037 - read - [DEBUG] - Running the data cleansing process
2024-08-28 09:33:12,039 - read - [DEBUG] - Data cleansed for: HY2343G302
2024-08-28 09:33:12,040 - read - [DEBUG] - Data cleansed for: HY2343G257
2024-08-28 09:33:12,041 - read - [DEBUG] - Data cleansed for: HY2343G446
2024-08-28 09:33:12,041 - read - [DEBUG] - Data cleansed for: HY2343G025
2024-08-28 09:33:12,042 - read - [DEBUG] - Data cleansed for: HY2343G259
2024-08-28 09:33:12,043 - read - [DEBUG] - Data cleansed for: HY2343G144
2024-08-28 09:33:12,044 - read - [DEBUG] - Data cleansed for: Battery_Stack_1
2024-08-28 09:33:12,044 - read - [DEBUG] - Data cleansed for: Battery_Details
2024-08-28 09:33:12,045 - read - [DEBUG] - Data cleansed for: Power
2024-08-28 09:33:12,045 - read - [DEBUG] - Data cleansed for: Power
2024-08-28 09:33:12,045 - read - [DEBUG] - Data cleansed for: TD2343G049
2024-08-28 09:33:12,046 - read - [DEBUG] - Data cleansed for: Meter_ID1
2024-08-28 09:33:12,046 - read - [DEBUG] - Data cleansed for: Meter_Details
2024-08-28 09:33:12,046 - read - [DEBUG] - Data cleansed for: Today
2024-08-28 09:33:12,047 - read - [DEBUG] - Data cleansed for: Total
2024-08-28 09:33:12,047 - read - [DEBUG] - Data cleansed for: Energy
2024-08-28 09:33:12,047 - read - [DEBUG] - Data cleansed for: Timeslots
2024-08-28 09:33:12,048 - read - [DEBUG] - Data cleansed for: Control
[... 346 lines cut ...]
2024-08-28 09:34:25,136 - mqtt - [DEBUG] - Publishing: GivEnergy/TD2343G049/raw
2024-08-28 09:34:25,137 - mqtt - [DEBUG] - Prepping HY2343G025 for publishing
2024-08-28 09:34:25,137 - mqtt - [DEBUG] - Prepping HY2343G144 for publishing
2024-08-28 09:34:25,138 - mqtt - [DEBUG] - Prepping HY2343G257 for publishing
2024-08-28 09:34:25,139 - mqtt - [DEBUG] - Prepping HY2343G259 for publishing
2024-08-28 09:34:25,140 - mqtt - [DEBUG] - Prepping HY2343G302 for publishing
2024-08-28 09:34:25,141 - mqtt - [DEBUG] - Prepping HY2343G446 for publishing
2024-08-28 09:34:25,148 - mqtt - [DEBUG] - Prepping Stack_0 for publishing
2024-08-28 09:34:25,152 - mqtt - [DEBUG] - Prepping HV_Battery_Stacks for publishing
2024-08-28 09:34:25,153 - mqtt - [DEBUG] - Prepping invertor for publishing
2024-08-28 09:34:25,155 - mqtt - [DEBUG] - Prepping Meter_ID_1 for publishing
2024-08-28 09:34:25,156 - mqtt - [DEBUG] - Prepping meters for publishing
2024-08-28 09:34:29,167 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-08-28 09:34:29,203 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-08-28 09:34:35,049 - evc - [DEBUG] - Socket is already open
2024-08-28 09:34:35,051 - evc - [DEBUG] - Socket is still open
2024-08-28 09:34:35,064 - evc - [DEBUG] - Dealt with Charger
2024-08-28 09:34:35,064 - evc - [DEBUG] - Publish all to MQTT
2024-08-28 09:34:35,065 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-08-28 09:34:45,070 - evc - [DEBUG] - Socket is already open
2024-08-28 09:34:45,084 - evc - [DEBUG] - Socket is still open
2024-08-28 09:34:45,095 - evc - [DEBUG] - Dealt with Charger
2024-08-28 09:34:45,096 - evc - [DEBUG] - Publish all to MQTT
2024-08-28 09:34:45,096 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
```

britkat1980, 2024-08-28:
OK thanks, it seems that that control is only for single phase and so is replaced with "Enable AC Charge".
Do you know the difference between "Enable AC Charge" and "Enable Force Charge"?

LeeFarm11, 2024-08-28:
I just tried testing both of these switches, but sorry, not conclusive. While testing, the PV generation was approximately matching the House Load. Both around 1kW. Might be better to test when there is no PV generation, and also when there is more PV than load.

Base situation - One of the time periods is set to cover the current time, with both switches Off, with Charge Power Rate set to 100, Charge Up To % set to 100 and Enable Force Discharge is Off. This is Normal/ Eco.

If I then set only 'Enable Force Charge' to On, it appears to send all PV to the Battery and power the house from the Grid.

If I set 'Enable Force Charge' to Off, and set 'Enable AC Charge' to On, I am not sure what is different to both switches being Off, except that the column 'AC Charge Power' in Inverter Data is showing around 1kW, but Battery is neither charging nor discharging and power is neither being sent to the Grid nor pulled from the Grid.

If I set both switches to On, then the battery is charged at full power from the Grid.

LeeFarm11, 2024-08-29:
Hi @britkat1980 
Just for your info there was a Portal/ App update last night. Now the Slot numbers are all messed up. I guess only for 3 phase..... In the Portal, on the Remote Settings screen, all is still the same as it was. However, On the Inverter, Settings page and on the App, the slot numbers are all messed up. See this post of Facebook - https://www.facebook.com/groups/1517262158723613/posts/1955699554879869/

britkat1980, 2024-08-29:
I have removed charge and discharge schedule controls as the only apply to single phase.

closed 2024-08-29 by britkat1980