britkat1980/giv_tcp #218: predbat unable to charge or discharge via GivTCP for 3 phase Inverter

LeeFarm11, 2024-09-03:
Hi @britkat1980 
Please see issue here -

https://github.com/springfall2008/batpred/issues/1426

It seems that predbat NEEDS enable_charge_shedule and enable_discharge_schedule but they are not available (directly) on 3 phase Inverter.......

LeeFarm11, 2024-09-03:
Hi. Now throwing an error and looping.

2024-09-03 14:38:46,851 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-03 14:38:48,889 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-03 14:38:48,889 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-03 14:38:50,138 - Startup - startup     -  [INFO    ] - GivEVC 11288853538258 found at: 192.168.1.102
2024-09-03 14:38:57,724 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.101:8899
2024-09-03 14:38:57,724 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-03 14:38:58,900 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-03 14:39:04,482 - Startup - startup     -  [INFO    ] - Inverter TD2343G049 which is a Gen1 - Hybrid_3ph with 6 batteries has been found at: 192.168.1.101
2024-09-03 14:39:04,493 - Startup - startup     -  [INFO    ] - Inverter TD2343G049 already found in settings file (slot 1), checking IP address is unchanged...
2024-09-03 14:39:04,500 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-03 14:39:04,600 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-03 14:39:04,601 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-03 14:39:04,601 - Startup - startup     -  [INFO    ] - ====     http://192.168.1.204:8099/config.html             ====
2024-09-03 14:39:04,602 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-03 14:39:04,606 - Startup - startup     -  [INFO    ] - Running Invertor 1 (TD2343G049) read loop every 30/120s
2024-09-03 14:39:04,615 - Startup - startup     -  [INFO    ] - Running EVC read loop every 10s
2024-09-03 14:39:04,625 - Startup - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2024-09-03 14:39:06 +0100] [57] [INFO] Starting gunicorn 23.0.0
[2024-09-03 14:39:06 +0100] [57] [INFO] Listening at: http://0.0.0.0:6345 (57)
[2024-09-03 14:39:06 +0100] [57] [INFO] Using worker: sync
[2024-09-03 14:39:06 +0100] [71] [INFO] Booting worker with pid: 71
[2024-09-03 14:39:06 +0100] [72] [INFO] Booting worker with pid: 72
[2024-09-03 14:39:06 +0100] [73] [INFO] Booting worker with pid: 73
2024-09-03 14:39:09,380 - Inv1 - evc         -  [DEBUG   ] - Socket isn't yet open
2024-09-03 14:39:09,448 - Inv1 - evc         -  [DEBUG   ] - Socket is still open
2024-09-03 14:39:09,459 - Inv1 - evc         -  [DEBUG   ] - Dealt with Charger
2024-09-03 14:39:09,467 - Inv1 - evc         -  [DEBUG   ] - Opening settings for first run evc
2024-09-03 14:39:09,471 - Inv1 - evc         -  [DEBUG   ] - Settings available evc
2024-09-03 14:39:09,477 - Inv1 - evc         -  [DEBUG   ] - serial number aready exists: "    serial_number_evc="11288853538258"
"
2024-09-03 14:39:09,480 - Inv1 - evc         -  [DEBUG   ] - removing lockfile
2024-09-03 14:39:09,482 - Inv1 - evc         -  [CRITICAL] - Publishing Home Assistant Discovery messages
2024-09-03 14:39:10,630 - Inv1 - HA_Discovery -  [DEBUG   ] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 14:39:10,643 - Inv1 - HA_Discovery -  [DEBUG   ] - Publishing: GivEnergy/11288853538258/Stats/Timeout_Error
2024-09-03 14:39:11,090 - Inv1 - selector_events -  [DEBUG   ] - Using selector: EpollSelector
2024-09-03 14:39:11,099 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-03 14:39:11,102 - Inv1 - GivLUT      -  [DEBUG   ] - Opening Modbus Connection to 192.168.1.101
2024-09-03 14:39:11,111 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-03 14:39:13,725 - Inv1 - HA_Discovery -  [DEBUG   ] - 0 discovery messages changed and removed
 INFO  Accepting connections at http://localhost:3000
2024-09-03 14:39:18,740 - Inv1 - HA_Discovery -  [DEBUG   ] - Connecting to broker: core-mosquitto
2024-09-03 14:39:18,740 - Inv1 - HA_Discovery -  [DEBUG   ] - In wait loop (sendDiscoMsg)
2024-09-03 14:39:18,944 - Inv1 - HA_Discovery -  [DEBUG   ] - In wait loop (sendDiscoMsg)
2024-09-03 14:39:19,148 - Inv1 - HA_Discovery -  [DEBUG   ] - In wait loop (sendDiscoMsg)
2024-09-03 14:39:19,350 - Inv1 - HA_Discovery -  [DEBUG   ] - In wait loop (sendDiscoMsg)
2024-09-03 14:39:19,557 - Inv1 - HA_Discovery -  [DEBUG   ] - In wait loop (sendDiscoMsg)
2024-09-03 14:39:19,746 - Inv1 - HA_Discovery -  [DEBUG   ] - connected OK Returned code=Success
2024-09-03 14:39:19,760 - Inv1 - HA_Discovery -  [DEBUG   ] - Publishing MQTT: core-mosquitto
2024-09-03 14:39:23,779 - Inv1 - HA_Discovery -  [DEBUG   ] - Sent 29 discovery messages
2024-09-03 14:39:23,779 - Inv1 - HA_Discovery -  [DEBUG   ] - Found 1809 discovery messages
2024-09-03 14:39:23,783 - Inv1 - evc         -  [DEBUG   ] - Publish all to MQTT
2024-09-03 14:39:23,783 - Inv1 - evc         -  [DEBUG   ] - MQTT Connection appears closed, re-opening
2024-09-03 14:39:23,783 - Inv1 - evc         -  [DEBUG   ] - Opening MQTT Connection to core-mosquitto
2024-09-03 14:39:23,787 - Inv1 - evc         -  [DEBUG   ] - connected OK Returned code=Success
2024-09-03 14:39:23,788 - Inv1 - evc         -  [DEBUG   ] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 14:39:23,789 - Inv1 - evc         -  [DEBUG   ] - Subscribing to GivEnergy/control/#
[... cut ...]
2024-09-03 14:41:04,075 - Inv1 - evc         -  [DEBUG   ] - Socket is already open
2024-09-03 14:41:04,077 - Inv1 - evc         -  [DEBUG   ] - Socket is still open
2024-09-03 14:41:04,089 - Inv1 - evc         -  [DEBUG   ] - Dealt with Charger
2024-09-03 14:41:04,090 - Inv1 - evc         -  [DEBUG   ] - Publish all to MQTT
2024-09-03 14:41:04,090 - Inv1 - evc         -  [DEBUG   ] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 14:41:06,935 - Inv1 - read        -  [DEBUG   ] - Running partial refresh

britkat1980, 2024-09-03:
Try Dev748

LeeFarm11, 2024-09-03:
Thanks now no error.
I did a manual Force Charge via Giv Portal to get some energy in the battery, but now it is 100% so can't a Force Charge but I tried a Force Dicharge via predbat switch.

At around 17:15 I told predbat to do a Manual Force Discharge at 17:30
It shows in the predbat plan.
But it seems it failed to do any discharge. Not sure if it helps but I have included the predbat log file and the GivTCP log file.

[image]

[image]

I can see in Giv Portal that Discharge Down To % was set. And Discharge Slot 1 End Time was set to 18:01.
But Enable Force Discharge was not set to On. I think it needs to be set to On to allow Discharge
DC Discharge 1 Lower SOC % Limit was not modified ( I normally set them all to 30%)

[image]

[image]

[image]

[predbat(7).log](https://github.com/user-attachments/files/16852332/predbat.7.log)
[log_inv_1(1).log](https://github.com/user-attachments/files/16852349/log_inv_1.1.log)

For Charging, will test later when import is cheap, instead of Enable Force Discharge being set to On, I think both Enable Force Charge and Enable AC Charge need to be On.
I will manually do a Force Discharge now and then predbat will probably want to do a normal scheduled Force Charge from 21:00. If not I will Force it to, and will report back.

britkat1980, 2024-09-03:
What I have implemented is that enable_charge_schedule (for 3ph) toggles both AC-Charge-Enable and Force-Charge-Enable forcing both on or off as per request
Enable_discharge_schedule does the same for Force-Discharge-Enable

LeeFarm11, 2024-09-03:
Hi. 

But during the above it seems the predbat 'Manual force discharge' did not toggle On the Giv 'Enable Force Discharge'. Print above from Giv Portal confirms it did not get switched on, even though 'DC Discharge 1 End Time' was set and 'Discharge Down To %' was set.

britkat1980, 2024-09-03:
If you manually toggle enable_discharge_schedule in HA does it change Force Discharge Enable in the portal?

gcoan, 2024-09-03:
> Print above from Giv Portal confirms it did not get switched on, even though 'DC Discharge 1 End Time' was set and 'Discharge Down To %' was set.

@LeeFarm11 be careful looking at the GivEnergy Portal as this is a copy of what it thinks the inverter is set to, based on what it last sent to the inverter or what it last read from the inverter.
GivTCP is writing directly to the inverter via modbus control and the portal is completely unaware of this happening.
You're best to check what's happening by looking at the HA controls directly as to what Predbat set, and then the GivTCP logfile (possibly with debug mode on) as to what it then sent to the inverter.

LeeFarm11, 2024-09-03:
With Predbat set to Read Only, I switched on 'Enable Discharge Schedule' at 20:35:40. 19 seconds later, at 20:35:59, it switched itself back to Off.

2024-09-03 20:35:32,389 - evc - [DEBUG] - Publish all to MQTT
2024-09-03 20:35:32,389 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 20:35:41,106 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableDischargeSchedule= enable
2024-09-03 20:35:41,155 - read - [DEBUG] - Write Request recieved
2024-09-03 20:35:41,156 - read - [DEBUG] - Command: enableDischargeSchedule was sent: {'state': 'enable'}
2024-09-03 20:35:41,156 - write - [DEBUG] - Enabling Disharge Schedule
2024-09-03 20:35:41,302 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G049/Control/Enable_Discharge_Schedule - enable
2024-09-03 20:35:41,304 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-09-03 20:35:42,397 - evc - [DEBUG] - Socket is already open
2024-09-03 20:35:42,402 - evc - [DEBUG] - Socket is still open
2024-09-03 20:35:42,415 - evc - [DEBUG] - Dealt with Charger
2024-09-03 20:35:42,418 - evc - [DEBUG] - Publish all to MQTT
2024-09-03 20:35:42,418 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 20:35:42,711 - write - [DEBUG] - Pushing control update to pkl cache: Enable_Discharge_Schedule - enable
2024-09-03 20:35:42,791 - write - [INFO] - Setting Discharge Schedule to enable was a success
2024-09-03 20:35:42,793 - read - [DEBUG] - responses is: [{'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 4 was a success"}'}, {'id': 'setBatteryPauseMode', 'result': '{"result": "Setting Battery Pause Mode to Disabled was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 18:01 was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 64 was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 71 was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 78 was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 00:00 was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 4 was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}]
2024-09-03 20:35:50,313 - read - [DEBUG] - Running partial refresh
2024-09-03 20:35:52,422 - evc - [DEBUG] - Socket is already open
2024-09-03 20:35:52,439 - evc - [DEBUG] - Socket is still open
2024-09-03 20:35:52,452 - evc - [DEBUG] - Dealt with Charger
2024-09-03 20:35:52,453 - evc - [DEBUG] - Publish all to MQTT
2024-09-03 20:35:52,454 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 20:35:54,242 - read - [DEBUG] - Data get was successful, now running handler if needed: 
2024-09-03 20:35:54,243 - read - [DEBUG] - Running processData
2024-09-03 20:35:54,243 - read - [DEBUG] - Beginning parsing of Inverter data
2024-09-03 20:35:54,244 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-09-03 20:35:54,888 - read - [DEBUG] - Getting Battery Details

There was nothing in the log relating to switching back off.

LeeFarm11, 2024-09-03:
Hi @gcoan 
I check the Inverter settings in Portal only after doing a fresh read of each individual record.

LeeFarm11, 2024-09-03:
I just tried setting Enable Discharge Schedule to On again.
Same result in Log File. Seems to be trying to be making up to 4 sets of changes all at once - 
Setting battery reserve 4, then 64, then 71, then 78, then 4.
Setting DischargeSlotEnd 1 to 18:01, then to 00:00

Some variables not being cleared maybe?

2024-09-03 20:51:10,586 - read - [DEBUG] - Command: enableDischargeSchedule was sent: {'state': 'enable'}
2024-09-03 20:51:10,586 - write - [DEBUG] - Enabling Disharge Schedule
2024-09-03 20:51:14,256 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G049/Control/Enable_Discharge_Schedule - enable
2024-09-03 20:51:14,257 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-09-03 20:51:15,014 - evc - [DEBUG] - Socket is already open
2024-09-03 20:51:15,017 - evc - [DEBUG] - Socket is still open
2024-09-03 20:51:15,029 - evc - [DEBUG] - Dealt with Charger
2024-09-03 20:51:15,030 - evc - [DEBUG] - Publish all to MQTT
2024-09-03 20:51:15,030 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 20:51:15,491 - write - [DEBUG] - Pushing control update to pkl cache: Enable_Discharge_Schedule - enable
2024-09-03 20:51:15,581 - write - [INFO] - Setting Discharge Schedule to enable was a success
2024-09-03 20:51:15,581 - read - [DEBUG] - responses is: [{'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 4 was a success"}'}, {'id': 'setBatteryPauseMode', 'result': '{"result": "Setting Battery Pause Mode to Disabled was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 18:01 was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 64 was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 71 was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 78 was a success"}'}, {'id': 'setBatteryMode', 'result': '{"result": "Setting Timed Export mode was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 00:00 was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 4 was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}]
2024-09-03 20:51:25,042 - evc - [DEBUG] - Socket is already open
2024-09-03 20:51:25,050 - evc - [DEBUG] - Socket is still open
2024-09-03 20:51:25,057 - evc - [DEBUG] - Dealt with Charger

LeeFarm11, 2024-09-03:
Upgraded to Dev 2.4.750

Did same again. This time less messages (may be due to restart of GivTCP clearing variables?) but same result, turned itself off a few seconds later.

2024-09-03 21:03:18,105 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 21:03:20,029 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableDischargeSchedule= enable
2024-09-03 21:03:20,070 - read - [DEBUG] - Write Request recieved
2024-09-03 21:03:20,071 - read - [DEBUG] - Command: enableDischargeSchedule was sent: {'state': 'enable'}
2024-09-03 21:03:20,071 - write - [DEBUG] - Enabling Disharge Schedule
2024-09-03 21:03:20,214 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G049/Control/Enable_Discharge_Schedule - enable
2024-09-03 21:03:20,216 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-09-03 21:03:20,245 - write - [DEBUG] - Pushing control update to pkl cache: Enable_Discharge_Schedule - enable
2024-09-03 21:03:20,247 - write - [INFO] - Setting Discharge Schedule to enable was a success
2024-09-03 21:03:20,248 - read - [DEBUG] - responses is: [{'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}]
2024-09-03 21:03:28,116 - evc - [DEBUG] - Socket is already open
2024-09-03 21:03:28,119 - evc - [DEBUG] - Socket is still open
2024-09-03 21:03:28,131 - evc - [DEBUG] - Dealt with Charger
2024-09-03 21:03:28,132 - evc - [DEBUG] - Publish all to MQTT
2024-09-03 21:03:28,133 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 21:03:38,140 - evc - [DEBUG] - Socket is already open
2024-09-03 21:03:38,143 - evc - [DEBUG] - Socket is still open
2024-09-03 21:03:38,160 - evc - [DEBUG] - Dealt with Charger
2024-09-03 21:03:38,160 - evc - [DEBUG] - Publish all to MQTT
2024-09-03 21:03:38,162 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 21:03:45,517 - read - [DEBUG] - Running full refresh
2024-09-03 21:03:48,168 - evc - [DEBUG] - Socket is already open
2024-09-03 21:03:48,172 - evc - [DEBUG] - Socket is still open
2024-09-03 21:03:48,184 - evc - [DEBUG] - Dealt with Charger
2024-09-03 21:03:48,184 - evc - [DEBUG] - Publish all to MQTT
2024-09-03 21:03:48,185 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 21:03:53,497 - read - [DEBUG] - Data get was successful, now running handler if needed: 
2024-09-03 21:03:53,497 - read - [DEBUG] - Running processData
2024-09-03 21:03:53,497 - read - [DEBUG] - Beginning parsing of Inverter data
2024-09-03 21:03:53,497 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-09-03 21:03:53,522 - read - [DEBUG] - Getting Battery Details
2024-09-03 21:03:53,524 - read - [DEBUG] - Building battery output: 
2024-09-03 21:03:53,525 - read - [DEBUG] - Battery HY2343G302 added
2024-09-03 21:03:53,525 - read - [DEBUG] - Building battery output: 
2024-09-03 21:03:53,526 - read - [DEBUG] - Battery HY2343G257 added
2024-09-03 21:03:53,527 - read - [DEBUG] - Building battery output: 
2024-09-03 21:03:53,528 - read - [DEBUG] - Battery HY2343G446 added
2024-09-03 21:03:53,528 - read - [DEBUG] - Building battery output: 
2024-09-03 21:03:53,529 - read - [DEBUG] - Battery HY2343G025 added
2024-09-03 21:03:53,530 - read - [DEBUG] - Building battery output: 
2024-09-03 21:03:53,530 - read - [DEBUG] - Battery HY2343G259 added
2024-09-03 21:03:53,531 - read - [DEBUG] - Building battery output: 
2024-09-03 21:03:53,532 - read - [DEBUG] - Battery HY2343G144 added
2024-09-03 21:03:53,533 - read - [DEBUG] - Getting mode control figures
2024-09-03 21:03:53,534 - read - [DEBUG] - Calculating Mode...
2024-09-03 21:03:53,534 - read - [DEBUG] - Mode is: Eco (Paused)
2024-09-03 21:03:53,535 - read - [DEBUG] - Force Export is not Running

LeeFarm11, 2024-09-03:
I enabled predbat (Read Only to Off) for a few minutes and it decided to try to do a Freeze Charge at 73% but failed. Then I set it back to Read Only to On.
Then I tried the GivTCP Enable Charge Schedule at 21:17:00
It switched itself back off again around 21:17:34

Here is the log. Again now multiple messages. Setting Battery Reserve to 73 and then to 4. Again supports variable somewhere not being cleared? 

NOTE: We are trying to Charge.
The Log says -
responses is: [{'id': 'enableDischargeSchedule', 'result': '{"result": "**_Setting Discharge Schedule to enable_** was a success"}'

Also note that battery reserve (Discharge Down To %) was successfully set to 4. I know because I cannot set it to 4 via the Portal. I can only set it down to 5.

2024-09-03 21:17:00,389 - evc - [DEBUG] - Publish all to MQTT
2024-09-03 21:17:00,389 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-03 21:17:00,614 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= enable
2024-09-03 21:17:00,660 - read - [DEBUG] - Write Request recieved
2024-09-03 21:17:00,661 - read - [DEBUG] - Command: enableChargeSchedule was sent: {'state': 'enable'}
2024-09-03 21:17:00,661 - write - [DEBUG] - Enabling Charge Schedule
2024-09-03 21:17:04,317 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G049/Control/Enable_Charge_Schedule - enable
2024-09-03 21:17:04,319 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-09-03 21:17:04,425 - write - [DEBUG] - Pushing control update to pkl cache: Enable_Charge_Schedule - enable
2024-09-03 21:17:04,438 - write - [INFO] - Setting Charge Schedule to enable was a success
2024-09-03 21:17:04,440 - read - [DEBUG] - responses is: [{'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'setBatteryPauseMode', 'result': '{"result": "Setting Battery Pause Mode to PauseDischarge was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 73 was a success"}'}, {'id': 'setBatteryPauseMode', 'result': '{"result": "Setting Battery Pause Mode to Disabled was a success"}'}, {'id': 'setBatteryReserve', 'result': '{"result": "Setting battery reserve 4 was a success"}'}]
2024-09-03 21:17:10,447 - evc - [DEBUG] - Socket is already open
2024-09-03 21:17:10,449 - evc - [DEBUG] - Socket is still open
2024-09-03 21:17:10,461 - evc - [DEBUG] - Dealt with Charger
2024-09-03 21:17:10,461 - evc - [DEBUG] - Publish all to MQTT

LeeFarm11, 2024-09-05:
Hi. Is there anything else I can provide to help solve these issues? Thanks.

britkat1980, 2024-09-06:
Try beta30.0c I wasn't passing inverter type to the commands, so it wasn't using the right registers!

LeeFarm11, 2024-09-06:
Testing with beta3.0.0c

Still a problem with Enable Charge Schedule

CHARGE -
Turn ON Force AC Charge Enable and Turn ON Force Charge Enable-
Both controls work and Enable Charge Schedule switches itself ON.
**Perfect.**

Then turn OFF either of Force AC Charge Enable or Force Charge Enable-
The individual Control is OFF and Enable Charge Schedule turns itself OFF.
**Perfect.**

With all 3 OFF, turn ON Enable Charge Schedule.
Nothing happens. And Log File shows loads of nonsense. Like variables not being cleared.
**Not good.**

```
2024-09-06 22:30:49,439 - evc - [DEBUG] - Publish all to MQTT
2024-09-06 22:30:49,439 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-06 22:30:51,633 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= enable
2024-09-06 22:30:51,859 - read - [DEBUG] - Write Request recieved
2024-09-06 22:30:51,860 - read - [DEBUG] - Command: enableChargeSchedule was sent: {'state': 'enable'}
2024-09-06 22:30:51,860 - write - [DEBUG] - Enabling Charge Schedule
2024-09-06 22:30:51,861 - write - [ERROR] - Setting charge schedule enable failed: ('TypeError', 'write.py', 204)
2024-09-06 22:30:51,865 - read - [DEBUG] - responses is: [{'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 10:00 was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 14:46 was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 00:00 was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 14:50 was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 00:00 was a success"}'}, {'id': 'setChargeSlotStart', 'result': '{"result": "Setting Charge Slot 1 Start to: 00:00 was a success"}'}, {'id': 'setChargeSlotEnd', 'result': '{"result": "Setting Charge Slot End 1 to: 00:00 was a success"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setForceDischarge', 'result': '{"result": {}}'}, {'id': 'forceExport', 'result': '{"result": "Export successfully forced for 5.0 minutes"}'}, {'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 1 was a success"}'}, {'id': 'forceCharge', 'result': '{"result": "Charge successfully forced for 2.0 minutes"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'setForceDischarge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'setForceDischarge', 'result': '{"result": {}}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}]
2024-09-06 22:30:59,457 - evc - [DEBUG] - Socket is already open
2024-09-06 22:30:59,487 - evc - [DEBUG] - Socket is still open
```

LeeFarm11, 2024-09-06:
Testing with beta 3.0.0c

Something not right with Enable Discharge Schedule.

DISCHARGE -
Enable Discharge Schedule and Force Discharge Enable.
When either one is turn ON or OFF, it stays ON or OFF and the other follows.
**Perfect.**

EXCEPT look at the log -
```
2024-09-06 22:24:35,980 - write - [DEBUG] - Enabling Disharge Schedule
2024-09-06 22:24:36,116 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G049/Control/Enable_Discharge_Schedule - enable
2024-09-06 22:24:36,117 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-09-06 22:24:37,775 - write - [DEBUG] - Pushing control update to pkl cache: Enable_Discharge_Schedule - enable
2024-09-06 22:24:37,898 - write - [INFO] - Setting Discharge Schedule to enable was a success
2024-09-06 22:24:37,900 - read - [DEBUG] - responses is: [{'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 10:00 was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 14:46 was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 00:00 was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 14:50 was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 00:00 was a success"}'}, {'id': 'setChargeSlotStart', 'result': '{"result": "Setting Charge Slot 1 Start to: 00:00 was a success"}'}, {'id': 'setChargeSlotEnd', 'result': '{"result": "Setting Charge Slot End 1 to: 00:00 was a success"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setForceDischarge', 'result': '{"result": {}}'}, {'id': 'forceExport', 'result': '{"result": "Export successfully forced for 5.0 minutes"}'}, {'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 1 was a success"}'}, {'id': 'forceCharge', 'result': '{"result": "Charge successfully forced for 2.0 minutes"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'setForceDischarge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}]
2024-09-06 22:24:38,317 - evc - [DEBUG] - Socket is already open
2024-09-06 22:24:38,319 - evc - [DEBUG] - Socket is still open
```

LeeFarm11, 2024-09-06:
I tried testing Force Charge 2, and it seems to work. 
At the end of the 2 minutes Battery Charge Rate AC is automatically moved from 100 to 1. Force AC Charge Enable and Force AC Enable remain ON, and the Start and End times in Charge Slot 1 remain with the 2 minute gap.

However, I then tried running it again and this time I cancelled half way through.

Cancelling was not successful. It tried to cancel. I saw Battery Charge Rate AC drop from 100% to 0%, but then it jumped back up to 100%. It ended only when the End Time in Charge Slot 1 was reached. But this time Battery Charge Rate AC did NOT drop back to 1. It remained at 100%. 

Here is a partial Log file.
[GivTCP Part Log.txt](https://github.com/user-attachments/files/16913187/GivTCP.Part.Log.txt)

LeeFarm11, 2024-09-06:
I tried setting charge start and end time which worked but Log file has issues - 

```
2024-09-06 23:16:27,925 - evc - [DEBUG] - Publish all to MQTT
2024-09-06 23:16:27,926 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-06 23:16:34,025 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setChargeStart1= 04:30:00
2024-09-06 23:16:34,076 - read - [DEBUG] - Write Request recieved
2024-09-06 23:16:34,076 - read - [DEBUG] - Command: setChargeSlotStart was sent: {'start': '04:30', 'slot': 1}
2024-09-06 23:16:34,077 - write - [DEBUG] - Setting Charge Slot 1 Start to: 04:30
2024-09-06 23:16:34,209 - write - [DEBUG] - Pushing control update to mqtt: GivEnergy/TD2343G049/Timeslots/Charge_start_time_slot_1 - 04:30:00
2024-09-06 23:16:34,211 - GivLUT - [DEBUG] - Opening regcache at: /config/GivTCP/regCache_1.pkl
2024-09-06 23:16:35,871 - write - [DEBUG] - Pushing control update to pkl cache: Charge_start_time_slot_1 - 04:30:00
2024-09-06 23:16:36,000 - write - [INFO] - Setting Charge Slot 1 Start to: 04:30 was a success
2024-09-06 23:16:36,001 - read - [DEBUG] - responses is: [{'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 10:00 was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 14:46 was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 00:00 was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 14:50 was a success"}'}, {'id': 'setDischargeSlotEnd', 'result': '{"result": "Setting Discharge Slot End 1 to: 00:00 was a success"}'}, {'id': 'setChargeSlotStart', 'result': '{"result": "Setting Charge Slot 1 Start to: 00:00 was a success"}'}, {'id': 'setChargeSlotEnd', 'result': '{"result": "Setting Charge Slot End 1 to: 00:00 was a success"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setForceDischarge', 'result': '{"result": {}}'}, {'id': 'forceExport', 'result': '{"result": "Export successfully forced for 5.0 minutes"}'}, {'id': 'setChargeTarget2', 'result': '{"result": "Setting Charge Target 1 was a success"}'}, {'id': 'forceCharge', 'result': '{"result": "Charge successfully forced for 2.0 minutes"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule disable failed: (\'TypeError\', \'write.py\', 208)"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'setForceDischarge', 'result': '{"result": {}}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge', 'result': '{"result": {}}'}, {'id': 'setACCharge', 'result': '{"result": {}}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to enable was a success"}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'setForceDischarge', 'result': '{"result": {}}'}, {'id': 'enableDischargeSchedule', 'result': '{"result": "Setting Discharge Schedule to disable was a success"}'}, {'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 204)"}'}, {'id': 'forceCharge', 'result': '{"result": "Charge successfully forced for 2.0 minutes"}'}, {'id': 'forceCharge', 'result': '{"result": "Charge successfully forced for 2.0 minutes"}'}, {'id': 'cancelJob', 'result': '{"result": "Cancelling scheduled task as requested"}'}]
2024-09-06 23:16:37,933 - evc - [DEBUG] - Socket is already open
2024-09-06 23:16:37,939 - evc - [DEBUG] - Socket is still open
2024-09-06 23:16:37,950 - evc - [DEBUG] - Dealt with Charger
2024-09-06 23:16:37,950 - evc - [DEBUG] - Publish all to MQTT
2024-09-06 23:16:37,951 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-06 23:16:47,963 - evc - [DEBUG] - Socket is already open
2024-09-06 23:16:47,967 - evc - [DEBUG] - Socket is still open
[... cut ...] ult': '{"result": "Charge successfully forced for 2.0 minutes"}'}, {'id': 'cancelJob', 'result': '{"result": "Cancelling scheduled task as requested"}'}, {'id': 'setChargeSlotStart', 'result': '{"result": "Setting Charge Slot 1 Start to: 04:30 was a success"}'}, {'id': 'setChargeSlotEnd', 'result': '{"result": "Setting Charge Slot End 1 to: 05:30 was a success"}'}]
2024-09-06 23:17:38,133 - evc - [DEBUG] - Socket is already open
2024-09-06 23:17:38,136 - evc - [DEBUG] - Socket is still open

```

LeeFarm11, 2024-09-09:
Hi @britkat1980

Are you able to 
1) Make Enable Charge Schedule turn on both Force AC Charge Enable and Force Charge Enable?
2) Fix Cancelling Force Charge X
3) Fix the issues with the Log File multiple repetition of previous carried out commands?

Thanks very much.

LeeFarm11, 2024-09-22:
Hi @britkat1980 
Is there any progress on this please? Is there any newer version worth testing yet?
Thanks.

britkat1980, 2024-09-24:
I've made an underlying change to the stability of entities which use the "enable" data type (switches). Which might help your issue. Can you try Dev3.0.2 and let me know if it moves us forward.

Will look at the Force Charge time select compatibility with TPH in the coming days

LeeFarm11, 2024-09-24:
With Dev 3.0.2

1) (node:60) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
(Use `node --trace-deprecation ...` to show where the warning was created)

2) 2024-09-24 22:28:46,341 - Inv1 - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'UnboundLocalError'>, UnboundLocalError("cannot access local variable 'inverter' where it is not associated with a value"), <traceback object at 0x7f22b1b02200>), 1439)

Then additional errors probably relating to above.

```
2024-09-24 22:27:57,822 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-24 22:27:57,822 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-24 22:27:59,298 - Startup - startup     -  [INFO    ] - GivEVC 11288853538258 found at: 192.168.1.102
2024-09-24 22:28:08,331 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.101:8899
2024-09-24 22:28:08,331 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-24 22:28:09,506 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-24 22:28:15,037 - Startup - startup     -  [INFO    ] - Inverter TD2343G049 which is a Gen1 - Hybrid_3ph with 6 batteries has been found at: 192.168.1.101
2024-09-24 22:28:15,057 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-24 22:28:15,146 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-24 22:28:15,147 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-24 22:28:15,148 - Startup - startup     -  [INFO    ] - ====     http://192.168.1.204:8099/config.html             ====
2024-09-24 22:28:15,149 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-24 22:28:15,150 - Startup - startup     -  [INFO    ] - Running Invertor 1 (TD2343G049) read loop every 30/120s
2024-09-24 22:28:15,160 - Startup - startup     -  [INFO    ] - Running EVC read loop every 10s
2024-09-24 22:28:15,169 - Startup - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2024-09-24 22:28:17 +0100] [57] [INFO] Starting gunicorn 23.0.0
[2024-09-24 22:28:17 +0100] [57] [INFO] Listening at: http://0.0.0.0:6345 (57)
[2024-09-24 22:28:17 +0100] [57] [INFO] Using worker: sync
[2024-09-24 22:28:17 +0100] [72] [INFO] Booting worker with pid: 72
[2024-09-24 22:28:17 +0100] [73] [INFO] Booting worker with pid: 73
[2024-09-24 22:28:17 +0100] [74] [INFO] Booting worker with pid: 74
(node:60) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
(Use `node --trace-deprecation ...` to show where the warning was created)
2024-09-24 22:28:20,768 - Inv1 - evc         -  [DEBUG   ] - Socket isn't yet open
2024-09-24 22:28:20,810 - Inv1 - evc         -  [DEBUG   ] - Socket is still open
2024-09-24 22:28:20,820 - Inv1 - evc         -  [DEBUG   ] - Dealt with Charger
2024-09-24 22:28:20,825 - Inv1 - evc         -  [DEBUG   ] - Opening settings for first run evc
2024-09-24 22:28:20,825 - Inv1 - evc         -  [DEBUG   ] - Settings available evc
2024-09-24 22:28:20,827 - Inv1 - evc         -  [DEBUG   ] - serial number aready exists: "    serial_number_evc="11288853538258"
"
2024-09-24 22:28:20,833 - Inv1 - evc         -  [DEBUG   ] - removing lockfile
2024-09-24 22:28:20,835 - Inv1 - evc         -  [CRITICAL] - Publishing Home Assistant Discovery messages
 INFO  Accepting connections at http://localhost:3000
2024-09-24 22:28:27,789 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-24 22:28:27,796 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-24 22:28:27,813 - Inv1 - HA_Discovery -  [DEBUG   ] - Publishing: GivEnergy/11288853538258/Charger
2024-09-24 22:28:27,818 - Inv1 - HA_Discovery -  [DEBUG   ] - Publishing: GivEnergy/11288853538258/Stats/Timeout_Error
2024-09-24 22:28:28,857 - Inv1 - HA_Discovery -  [DEBUG   ] - Foundmessages= 0
2024-09-24 22:28:28,859 - Inv1 - HA_Discovery -  [DEBUG   ] - Moremessages= 317
2024-09-24 22:28:29,861 - Inv1 - HA_Discovery -  [DEBUG   ] - Foundmessages= 317
2024-09-24 22:28:29,863 - Inv1 - HA_Discovery -  [DEBUG   ] - Moremessages= 1365
2024-09-24 22:28:30,864 - Inv1 - HA_Discovery -  [DEBUG   ] - Foundmessages= 1365
2024-09-24 22:28:30,865 - Inv1 - HA_Discovery -  [DEBUG   ] - Moremessages= 1934
2024-09-24 22:28:31,866 - Inv1 - HA_Discovery -  [DEBUG   ] - Foundmessages= 1934
2024-09-24 22:28:31,866 - Inv1 - HA_Discovery -  [DEBUG   ] - Moremessages= 1934
2024-09-24 22:28:32,867 - Inv1 - HA_Discovery -  [DEBUG   ] - Foundmessages= 1934
2024-09-24 22:28:32,867 - Inv1 - HA_Discovery -  [DEBUG   ] - Moremessages= 1934
2024-09-24 22:28:33,868 - Inv1 - HA_Discovery -  [DEBUG   ] - Foundmessages= 1934
2024-09-24 22:28:33,868 - Inv1 - HA_Discovery -  [DEBUG   ] - Moremessages= 1934
2024-09-24 22:28:34,868 - Inv1 - HA_Discovery -  [DEBUG   ] - Foundmessages= 1934
2024-09-24 22:28:34,869 - Inv1 - HA_Discovery -  [DEBUG   ] - Moremessages= 1934
2024-09-24 22:28:35,870 - Inv1 - HA_Discovery -  [DEBUG   ] - Foundmessages= 1934
2024-09-24 22:28:35,870 - Inv1 - HA_Discovery -  [DEBUG   ] - Moremessages= 1934
2024-09-24 22:28:35,879 - Inv1 - HA_Discovery -  [DEBUG   ] - 2 discovery messages changed and removed
[... cut ...]
2024-09-24 22:29:21,365 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f22b069b9c0>), 1633)
2024-09-24 22:29:21,366 - Inv1 - read        -  [DEBUG   ] - processData result: processData Error processing registers: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f22b069b9c0>), 1633)
2024-09-24 22:29:21,366 - Inv1 - read        -  [DEBUG   ] - Running pubFromPickle
2024-09-24 22:29:21,368 - Inv1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1809)
```

LeeFarm11, 2024-09-25:
Now Upgraded (!!) to Dev 3.0.1 but looks worse.

```
2024-09-25 07:34:35,600 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-25 07:34:37,634 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-25 07:34:37,634 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-25 07:34:53,936 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.101:8899
2024-09-25 07:34:53,937 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-25 07:34:58,357 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-25 07:35:03,896 - Startup - startup     -  [INFO    ] - Inverter TD2343G049 which is a Gen1 - Hybrid_3ph with 6 batteries has been found at: 192.168.1.101
2024-09-25 07:35:03,918 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-25 07:35:04,084 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-25 07:35:04,085 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-25 07:35:04,085 - Startup - startup     -  [INFO    ] - ====     http://192.168.1.204:8099/config.html             ====
2024-09-25 07:35:04,087 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-25 07:35:04,090 - Startup - startup     -  [INFO    ] - Running Invertor 1 (TD2343G049) read loop every 30/120s
2024-09-25 07:35:04,096 - Startup - startup     -  [INFO    ] - Running EVC read loop every 10s
2024-09-25 07:35:04,110 - Startup - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2024-09-25 07:35:07 +0100] [57] [INFO] Starting gunicorn 23.0.0
[2024-09-25 07:35:07 +0100] [57] [ERROR] Connection in use: ('', 6345)
[2024-09-25 07:35:07 +0100] [57] [ERROR] connection to ('', 6345) failed: [Errno 98] Address in use
[2024-09-25 07:35:08 +0100] [57] [ERROR] Connection in use: ('', 6345)
[2024-09-25 07:35:08 +0100] [57] [ERROR] connection to ('', 6345) failed: [Errno 98] Address in use
[2024-09-25 07:35:09 +0100] [57] [ERROR] Connection in use: ('', 6345)
[2024-09-25 07:35:09 +0100] [57] [ERROR] connection to ('', 6345) failed: [Errno 98] Address in use
[2024-09-25 07:35:10 +0100] [57] [ERROR] Connection in use: ('', 6345)
[2024-09-25 07:35:10 +0100] [57] [ERROR] connection to ('', 6345) failed: [Errno 98] Address in use
(node:60) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
(Use `node --trace-deprecation ...` to show where the warning was created)
[2024-09-25 07:35:11 +0100] [57] [ERROR] Connection in use: ('', 6345)
[2024-09-25 07:35:11 +0100] [57] [ERROR] connection to ('', 6345) failed: [Errno 98] Address in use
2024-09-25 07:35:12,562 - Inv1 - evc         -  [DEBUG   ] - Socket isn't yet open
2024-09-25 07:35:12,578 - Inv1 - evc         -  [DEBUG   ] - Socket is closed
[2024-09-25 07:35:12 +0100] [57] [ERROR] Can't connect to ('', 6345)
 INFO  Accepting connections at http://localhost:3000
2024-09-25 07:35:16,917 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-25 07:35:16,920 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-25 07:35:22,581 - Inv1 - evc         -  [DEBUG   ] - Socket isn't yet open
2024-09-25 07:35:22,584 - Inv1 - evc         -  [DEBUG   ] - Error: ('ConnectionException', 'evc.py', 123)
2024-09-25 07:35:32,584 - Inv1 - evc         -  [DEBUG   ] - Socket isn't yet open
2024-09-25 07:35:32,587 - Inv1 - evc         -  [DEBUG   ] - Error: ('ConnectionException', 'evc.py', 123)
2024-09-25 07:35:42,588 - Inv1 - evc         -  [DEBUG   ] - Socket isn't yet open
2024-09-25 07:35:42,590 - Inv1 - evc         -  [DEBUG   ] - Socket is closed
2024-09-25 07:35:47,088 - Inv1 - read        -  [DEBUG   ] - Running full refresh
2024-09-25 07:35:47,092 - Inv1 - read        -  [DEBUG   ] - Running processData
2024-09-25 07:35:47,092 - Inv1 - read        -  [DEBUG   ] - Beginning parsing of Inverter data
2024-09-25 07:35:47,093 - Inv1 - read        -  [ERROR   ] - Error processing Three Phase data: ((<class 'UnboundLocalError'>, UnboundLocalError("cannot access local variable 'inverter' where it is not associated with a value"), <traceback object at 0x7f599835fb80>), 1417)
2024-09-25 07:35:47,093 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f5997e80c40>), 1595)
2024-09-25 07:35:47,096 - Inv1 - read        -  [DEBUG   ] - processData result: processData Error processing registers: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f5997e80c40>), 1595)
2024-09-25 07:35:47,097 - Inv1 - read        -  [DEBUG   ] - Running pubFromPickle
2024-09-25 07:35:47,098 - Inv1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1771)
```

britkat1980, 2024-09-25:
Looks like you have two versions of GivTCP running. Stop the prod version and just run dev 3.0.1. I’ll look at the inverter type error once I’ve got the kids to school

britkat1980, 2024-09-25:
try Dev3.0.3

LeeFarm11, 2024-09-25:
Oops. Sorry. Yes I failed to stop Beta. Now running Dev 3.0.4

Some error/warning messages in Log File, but seems to be running.

Initially I see that Force Charge Enable and Force AC Charge Enable are On (I have been controlling via Giv Portal), but Enable Charge Schedule is showing Off. 
[image]

If I try to turn On Enable Charge Schedule, the switch simply goes back to Off.

```
2024-09-25 12:41:14,803 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= enable
2024-09-25 12:41:14,856 - read - [DEBUG] - Write Request recieved
2024-09-25 12:41:14,857 - read - [DEBUG] - Command: enableChargeSchedule was sent: {'state': 'enable'}
2024-09-25 12:41:14,861 - read - [DEBUG] - responses is: [{'id': 'enableChargeSchedule', 'result': '{"result": "Setting charge schedule enable failed: (\'TypeError\', \'write.py\', 225)"}'}]
2024-09-25 12:41:22,682 - read - [DEBUG] - Running partial refresh
2024-09-25 12:41:24,429 - evc - [DEBUG] - Socket is already open
2024-09-25 12:41:24,432 - evc - [DEBUG] - Socket is still open
2024-09-25 12:41:24,446 - evc - [DEBUG] - Dealt with Charger
2024-09-25 12:41:24,452 - evc - [DEBUG] - Publish all to MQTT
2024-09-25 12:41:24,456 - evc - [DEBUG] - Publishing: GivEnergy/11288853538258/Charger
2024-09-25 12:41:25,379 - read - [DEBUG] - Data get was successful, now running handler if needed: 
2024-09-25 12:41:25,379 - read - [DEBUG] - Running processData
2024-09-25 12:41:25,379 - read - [DEBUG] - Beginning parsing of Inverter data
2024-09-25 12:41:25,434 - read - [DEBUG] - Getting Battery Details
2024-09-25 12:41:25,438 - read - [DEBUG] - Building battery output: 
2024-09-25 12:41:25,440 - read - [DEBUG] - Battery HY2343G302 added
2024-09-25 12:41:25,440 - read - [DEBUG] - Building battery output: 
2024-09-25 12:41:25,442 - read - [DEBUG] - Battery HY2343G257 added
2024-09-25 12:41:25,442 - read - [DEBUG] - Building battery output: 
2024-09-25 12:41:25,444 - read - [DEBUG] - Battery HY2343G446 added
2024-09-25 12:41:25,445 - read - [DEBUG] - Building battery output: 
2024-09-25 12:41:25,446 - read - [DEBUG] - Battery HY2343G025 added
2024-09-25 12:41:25,447 - read - [DEBUG] - Building battery output: 
2024-09-25 12:41:25,448 - read - [DEBUG] - Battery HY2343G259 added
2024-09-25 12:41:25,449 - read - [DEBUG] - Building battery output: 
2024-09-25 12:41:25,450 - read - [DEBUG] - Battery HY2343G144 added
2024-09-25 12:41:25,452 - read - [DEBUG] - Getting mode control figures
2024-09-25 12:41:25,453 - read - [DEBUG] - Calculating Mode...
2024-09-25 12:41:25,454 - read - [DEBUG] - Mode is: Eco (Paused)
2024-09-25 12:41:25,455 - read - [DEBUG] - Force Export is not Running
2024-09-25 12:41:25,456 - read - [DEBUG] - Getting TimeSlot data
2024-09-25 12:41:25,456 - read - [DEBUG] - Getting TimeSlot data
2024-09-25 12:41:25,503 - read - [DEBUG] - cache len= 18
2024-09-25 12:41:25,503 - read - [DEBUG] - Running the data cleansing process
2024-09-25 12:41:25,507 - read - [DEBUG] - Stack_SOC_Difference is outside of allowable bounds so using old value. Out of bounds value is: -3. Min limit: 0. Max limit: 100
```

LeeFarm11, 2024-09-25:
Here are the warning messages I spotted 

```
2024-09-25 12:30:49,782 - read - [DEBUG] - cache len= 0
2024-09-25 12:30:49,782 - read - [DEBUG] - Data Cleansing not carried out: cache too small
2024-09-25 12:30:49,782 - read - [DEBUG] - Starting ratecalcs...
```

```
2024-09-25 12:31:02,380 - HA_Discovery - [DEBUG] - In wait loop (sendDiscoMsg)
2024-09-25 12:31:02,542 - evc - [DEBUG] - Socket is already open
2024-09-25 12:31:02,543 - evc - [DEBUG] - Error: ('ConnectionException', 'evc.py', 123)
2024-09-25 12:31:02,581 - HA_Discovery - [DEBUG] - In wait loop (sendDiscoMsg)
```

```
2024-09-25 12:31:09,834 - read - [DEBUG] - Publish all to MQTT
2024-09-25 12:31:09,834 - mqtt - [DEBUG] - MQTT Connection appears closed, re-opening
2024-09-25 12:31:09,835 - mqtt - [DEBUG] - Opening MQTT Connection to core-mosquitto
2024-09-25 12:31:09,839 - mqtt - [DEBUG] - connected OK Returned code=Success
```

```
2024-09-25 12:32:17,710 - read - [DEBUG] - Running the data cleansing process
2024-09-25 12:32:17,712 - read - [DEBUG] - Stack_SOC_Difference is outside of allowable bounds so using old value. Out of bounds value is: -2. Min limit: 0. Max limit: 100
2024-09-25 12:32:17,713 - read - [DEBUG] - Data cleansed for: HY2343G302
```

```
2024-09-25 12:32:17,710 - read - [DEBUG] - Running the data cleansing process
2024-09-25 12:32:17,712 - read - [DEBUG] - Stack_SOC_Difference is outside of allowable bounds so using old value. Out of bounds value is: -2. Min limit: 0. Max limit: 100
2024-09-25 12:32:17,713 - read - [DEBUG] - Data cleansed for: HY2343G302
2024-09-25 12:32:17,714 - read - [DEBUG] - Data cleansed for: HY2343G257
2024-09-25 12:32:17,716 - read - [DEBUG] - Data cleansed for: HY2343G446
2024-09-25 12:32:17,717 - read - [DEBUG] - Data cleansed for: HY2343G025
2024-09-25 12:32:17,718 - read - [DEBUG] - Data cleansed for: HY2343G259
2024-09-25 12:32:17,718 - read - [DEBUG] - Data cleansed for: HY2343G144
2024-09-25 12:32:17,719 - read - [DEBUG] - Data cleansed for: Battery_Stack_1
2024-09-25 12:32:17,719 - read - [DEBUG] - Data cleansed for: Battery_Details
2024-09-25 12:32:17,720 - read - [DEBUG] - Data cleansed for: Power
2024-09-25 12:32:17,720 - read - [DEBUG] - Data cleansed for: Power
2024-09-25 12:32:17,721 - read - [DEBUG] - Power_Factor is Zero so using old value
2024-09-25 12:32:17,721 - read - [DEBUG] - Data cleansed for: TD2343G049
2024-09-25 12:32:17,722 - read - [DEBUG] - Frequency is Zero so using old value
2024-09-25 12:32:17,722 - read - [DEBUG] - Phase_1_Power_Factor is Zero so using old value
2024-09-25 12:32:17,722 - read - [DEBUG] - Phase_2_Power_Factor is Zero so using old value
2024-09-25 12:32:17,722 - read - [DEBUG] - Phase_3_Power_Factor is Zero so using old value
2024-09-25 12:32:17,723 - read - [DEBUG] - Data cleansed for: Meter_ID1
2024-09-25 12:32:17,723 - read - [DEBUG] - Data cleansed for: Meter_Details
2024-09-25 12:32:17,723 - read - [DEBUG] - Export2_Energy_Today_kWh is Zero so using old value
2024-09-25 12:32:17,724 - read - [DEBUG] - Data cleansed for: Today
2024-09-25 12:32:17,724 - read - [DEBUG] - Export2_Energy_Total_kWh is Zero so using old value
2024-09-25 12:32:17,724 - read - [DEBUG] - Data cleansed for: Total
2024-09-25 12:32:17,725 - read - [DEBUG] - Data cleansed for: Energy
2024-09-25 12:32:17,725 - read - [DEBUG] - Data cleansed for: Timeslots
2024-09-25 12:32:17,725 - read - [DEBUG] - Battery_Power_Cutoff is Zero so using old value
2024-09-25 12:32:17,726 - read - [DEBUG] - Data cleansed for: Control
2024-09-25 12:32:17,727 - read - [DEBUG] - Data cleansed for: Stats
2024-09-25 12:32:17,727 - read - [DEBUG] - Data Cleansing not carried out: cache too small
2024-09-25 12:32:17,727 - read - [DEBUG] - Starting ratecalcs...
```

LeeFarm11, 2024-09-25:
Now with Dev 3.0.8

Still showing similar errors to above.

And now predbat will not run. Suggesting a problem with sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh
Please see predbat below. And GivTCP log file attached.

[log_inv_1(2).log](https://github.com/user-attachments/files/17137671/log_inv_1.2.log)

```
-----------------------------------------------------------
Bootstrap Predbat
Startup
 Add-on: Predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
Predbat files are installed correctly for version v8.4.9
**** Starting Standalone Predbat ****
2024-09-25 20:28:11.730893: Loading apps.yaml
2024-09-25 20:28:12.029108: Info: Connected to Home Assistant at http://supervisor/core
2024-09-25 20:28:12.039133: Info: Add-on slug is 6adb4f0d_predbat
2024-09-25 20:28:12.039270: Creating task: <coroutine object HAInterface.socketLoop at 0x7efffb217e60>
2024-09-25 20:28:12.040269: Info: Web Socket task started
2024-09-25 20:28:12.040406: Creating task: <coroutine object WebInterface.start at 0x7efffb217d80>
2024-09-25 20:28:12.041662: Info: Start socket for url http://supervisor/core/api/websocket
2024-09-25 20:28:12.045558: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
Web interface started
2024-09-25 20:28:12.087958: Info: Web Socket active
2024-09-25 20:28:12.246057: Warn: Regular expression argument: geserial2 unable to match re:sensor.givtcp2_(.+)_soc_kwh, now will disable
2024-09-25 20:28:12.252467: Warn: Regular expression argument: car_charging_energy unable to match re:(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy), now will disable
2024-09-25 20:28:12.259174: Warn: Regular argument car_charging_planned expression re:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status) failed to match - disabling this item
2024-09-25 20:28:12.265182: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
2024-09-25 20:28:12.270942: Warn: Regular expression argument: octopus_saving_session unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|)), now will disable
2024-09-25 20:28:12.276941: Warn: Regular expression argument: metric_octopus_import unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate), now will disable
2024-09-25 20:28:12.282847: Warn: Regular expression argument: metric_octopus_export unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate), now will disable
 System: Home Assistant OS 13.1  (amd64 / qemux86-64)
 Home Assistant Core: 2024.9.2
 Home Assistant Supervisor: 2024.09.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[20:28:13] INFO: Configuring timezone (Europe/London)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Watching ['./prediction.py', './hass.py', './apps.yaml', './web.py', './config.py', './download.py', './predbat.py', './utils.py', './ha.py', './unit_test.py', './inverter.py'] for changes
2024-09-25 20:28:19.872676: Error: Unable to fetch history for sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh
2024-09-25 20:28:19.905526: Info: record_status Error: Unable to fetch history from sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh
2024-09-25 20:28:19.905721: Error: Exception raised 
2024-09-25 20:28:19.913074: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 11096, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9919, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 8971, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/predbat.py", line 1062, in minute_data_load
    raise ValueError
ValueError

2024-09-25 20:28:19.953226: Info: record_status Error: Exception raised 
2024-09-25 20:28:19.953359: Error: 
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 11102, in update_time_loop
    raise e
  File "/config/predbat.py", line 11096, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9919, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 8971, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/predbat.py", line 1062, in minute_data_load
    raise ValueError
ValueError
```

LeeFarm11, 2024-09-28:
Hi @britkat1980 
Due to the reuse of version numbers in the Dev history I am unable to roll back to a GivTCP Dev version which allows Predbat to start. I am only using predbat in Monitor mode (obvs), but I can't even do that at the moment. 
Any ideas please?

gcoan, 2024-09-28:
> Due to the reuse of version numbers in the Dev history I am unable to roll back to a GivTCP Dev version which allows Predbat to start.

Did you take a backup of the original add-on version as part of the install process that you could restore ?

Alternatively if you have a full HA backup taken any time beforehand that has a working GivTCP version in, you can select that backup and do a partial restore of just the GivTCP v3 add-on and that works as well.

LeeFarm11, 2024-09-28:
Yes I always leave the box ticked to save a backup.
I have then gone back to select the relevant restore points, but if I try to restore back to any of the version installed a few weeks ago, it doesn't restore it to the version I had previously. Instead it restores it to the latest version with that number which was not the version I previously had.

LeeFarm11, 2024-09-28:
I don't update every time there is a new version.
And I don't have a full HA backup. Probably should schedule that. :)

gcoan, 2024-09-28:
> Yes I always leave the box ticked to save a backup.
>
> I have then gone back to select the relevant restore points, but if I try to restore back to any of the version installed a few weeks ago, it [...]

ah ok, I thought it would have saved the day for you

r/e doing full backups, I use the Google Drive backup add-on to both take a daily backup and to save it securely on a Google cloud server.  You can create a new dedicated Google account for HA backups and get 15Gb of free cloud storage which is enough for several generations of backups.

LeeFarm11, 2024-09-28:
Now running Dev 3.0.12 (the version of today 28 September. Not the Dev 3.0.12 of 21 September)

Still getting the same message - Data Cleansing not carried out: cache too small

Still getting the same message - Error: ('ConnectionException', 'evc.py', 123)

Still getting these errors repeatedly. Every 30 seconds.

2024-09-28 19:46:25,448 - read - [DEBUG] - Power_Factor is Zero so using old value
2024-09-28 19:46:25,448 - read - [DEBUG] - Data cleansed for: TD2343G049
2024-09-28 19:46:25,449 - read - [DEBUG] - Frequency is Zero so using old value
2024-09-28 19:46:25,449 - read - [DEBUG] - Phase_1_Power_Factor is Zero so using old value
2024-09-28 19:46:25,449 - read - [DEBUG] - Phase_2_Power_Factor is Zero so using old value
2024-09-28 19:46:25,450 - read - [DEBUG] - Phase_3_Power_Factor is Zero so using old value
2024-09-28 19:46:25,450 - read - [DEBUG] - Data cleansed for: Meter_ID1
2024-09-28 19:46:25,450 - read - [DEBUG] - Data cleansed for: Meter_Details
2024-09-28 19:46:25,451 - read - [DEBUG] - Export2_Energy_Today_kWh is Zero so using old value
2024-09-28 19:46:25,451 - read - [DEBUG] - Data cleansed for: Today
2024-09-28 19:46:25,451 - read - [DEBUG] - Export2_Energy_Total_kWh is Zero so using old value
2024-09-28 19:46:25,452 - read - [DEBUG] - Data cleansed for: Total
2024-09-28 19:46:25,452 - read - [DEBUG] - Data cleansed for: Energy
2024-09-28 19:46:25,452 - read - [DEBUG] - Data cleansed for: Timeslots
2024-09-28 19:46:25,453 - read - [DEBUG] - Battery_Power_Cutoff is Zero so using old value
2024-09-28 19:46:25,454 - read - [DEBUG] - Data cleansed for: Control
2024-09-28 19:46:25,454 - read - [DEBUG] - Data cleansed for: Stats

Still getting same fatal error in predbat -
Info: record_status Error: Unable to fetch history from sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh
so I have no predbat plan.

gcoan, 2024-09-28:
> Still getting same fatal error in predbat -
> Info: record_status Error: Unable to fetch history from sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh
> so I have no predbat plan.

I've seen this on AIO's with givtcp v3. The regular expression that determines geserial is picking up the wrong sensor with v3

What are your apps.yaml lines that set geserial, did you change them at all?
Second, can you list what sensors you have with the suffix '\_soc\_kwh'. I think the regular expression needs changing for 3 phase as it needs to for AIO (with givtcp v3)

LeeFarm11, 2024-09-28:
Hi @gcoan 
In apps.yaml I don't think I have changed geserial. I have this -

```
  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.givtcp_(.+)_soc_kwh'
```
Here are all my Entities with _soc_kwh
[image]

gcoan, 2024-09-28:
> Hi @gcoan
>
> In apps.yaml I don't think I have changed geserial. I have this -
>
>
>
> ```
>
>   # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
>
>   # [...]

Ok great thanks. the predbat fix is to change apps.yaml to:

geserial: 'sensor.givtcp_td2343g049_soc_kwh'

LeeFarm11, 2024-09-28:
Thanks @gcoan 
That has cleared the first error, but the subsequent errors remain. And still no predbat plan.

```
  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 'sensor.givtcp_td2343g049_soc_kwh'
  #geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  #geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'
```

```
Startup
Predbat files are installed correctly for version v8.4.9
**** Starting Standalone Predbat ****
2024-09-28 23:24:07.980319: Loading apps.yaml
2024-09-28 23:24:08.235872: Info: Connected to Home Assistant at http://supervisor/core
2024-09-28 23:24:08.244367: Info: Add-on slug is 6adb4f0d_predbat
2024-09-28 23:24:08.244505: Creating task: <coroutine object HAInterface.socketLoop at 0x7fb667bbbe60>
2024-09-28 23:24:08.245957: Info: Start socket for url http://supervisor/core/api/websocket
2024-09-28 23:24:08.248080: Info: Web Socket task started
2024-09-28 23:24:08.248318: Creating task: <coroutine object WebInterface.start at 0x7fb667bbbd80>
Web interface started
2024-09-28 23:24:08.254195: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-09-28 23:24:08.276355: Info: Web Socket active
2024-09-28 23:24:08.409849: Warn: Regular expression argument: car_charging_energy unable to match re:(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy), now will disable
2024-09-28 23:24:08.416421: Warn: Regular argument car_charging_planned expression re:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status) failed to match - disabling this item
2024-09-28 23:24:08.422146: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
2024-09-28 23:24:08.427269: Warn: Regular expression argument: octopus_saving_session unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|)), now will disable
2024-09-28 23:24:08.432911: Warn: Regular expression argument: metric_octopus_import unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate), now will disable
2024-09-28 23:24:08.438860: Warn: Regular expression argument: metric_octopus_export unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate), now will disable
Watching ['./prediction.py', './hass.py', './apps.yaml', './web.py', './config.py', './download.py', './predbat.py', './utils.py', './ha.py', './unit_test.py', './inverter.py'] for changes
2024-09-28 23:24:15.744121: Error: Exception raised 0
2024-09-28 23:24:15.750548: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 11096, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9919, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 8971, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/predbat.py", line 1036, in minute_data_load
    item = history[0][0]
KeyError: 0

2024-09-28 23:24:15.813582: Info: record_status Error: Exception raised 0
2024-09-28 23:24:15.813721: Error: 0
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 11102, in update_time_loop
    raise e
  File "/config/predbat.py", line 11096, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 9919, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 8971, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/predbat.py", line 1036, in minute_data_load
    item = history[0][0]
KeyError: 0

2024-09-28 23:25:01.298934: Error: Exception raised 0
2024-09-28 23:25:01.299253: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 11150, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 9919, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 8971, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/predbat.py", line 1036, in minute_data_load
    item = history[0][0]
KeyError: 0

2024-09-28 23:25:01.345742: Info: record_status Error: Exception raised 0
2024-09-28 23:25:01.345973: Error: 0
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 11155, in run_time_loop
    raise e
  File "/config/predbat.py", line 11150, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 9919, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 8971, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/predbat.py", line 1036, in minute_data_load
    item = history[0][0]
KeyError: 0
```

gcoan, 2024-09-28:
> geserial: 'sensor.givtcp_td2343g049_soc_kwh'

sorry stupid me

set it to:

geserial: 'td2343g049'

i.e. hard code to your inverter serial number 

the failure you see is because predbat can't find the load data because the sensor id is wrong.  This should fix it

LeeFarm11, 2024-09-29:
Thanks @gcoan 
That makes sense. Now predbat appears to be working hard doing calculations in the background relating to missing data from Historical Day 2. Since 09:43. Now it is 10:00 and still working on it. Do you think this is still something related to GivTCP or something else? Is it confused? Can you take a look below?

Bootstrap Predbat
Startup
Predbat files are installed correctly for version v8.4.9
**** Starting Standalone Predbat ****
2024-09-29 09:42:55.340468: Loading apps.yaml
2024-09-29 09:42:55.631631: Info: Connected to Home Assistant at http://supervisor/core
2024-09-29 09:42:55.648681: Info: Add-on slug is 6adb4f0d_predbat
2024-09-29 09:42:55.653899: Creating task: <coroutine object HAInterface.socketLoop at 0x7f4b645f3e60>
2024-09-29 09:42:55.654843: Info: Start socket for url http://supervisor/core/api/websocket
2024-09-29 09:42:55.656398: Info: Web Socket task started
2024-09-29 09:42:55.656792: Creating task: <coroutine object WebInterface.start at 0x7f4b645f3d80>
Web interface started
2024-09-29 09:42:55.668084: Warn: unable to find /addon_configs/6adb4f0d_predbat/appdaemon.yaml skipping checks as Predbat maybe running outside of AppDaemon
2024-09-29 09:42:55.687317: Info: Web Socket active
2024-09-29 09:42:55.824952: Warn: Regular expression argument: car_charging_energy unable to match re:(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy), now will disable
2024-09-29 09:42:55.831496: Warn: Regular argument car_charging_planned expression re:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status) failed to match - disabling this item
2024-09-29 09:42:55.837177: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
2024-09-29 09:42:55.841835: Warn: Regular expression argument: octopus_saving_session unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|)), now will disable
2024-09-29 09:42:55.847414: Warn: Regular expression argument: metric_octopus_import unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate), now will disable
2024-09-29 09:42:55.852523: Warn: Regular expression argument: metric_octopus_export unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate), now will disable
Watching ['./prediction.py', './hass.py', './apps.yaml', './web.py', './config.py', './download.py', './predbat.py', './utils.py', './ha.py', './unit_test.py', './inverter.py'] for changes
2024-09-29 09:43:07.902200: Warn: Historical day 2 has 1425 minutes of gap in the data, filled from 1.3 kWh to make new average 125.12 kWh (percent 1%)
2024-09-29 09:45:03.345688: Info: record_status Read-Only
2024-09-29 09:45:09.054404: Warn: Historical day 2 has 1420 minutes of gap in the data, filled from 1.41 kWh to make new average 101.4 kWh (percent 1%)
2024-09-29 09:45:16.025750: Info: record_status Read-Only
2024-09-29 09:50:05.369076: Warn: Historical day 2 has 1415 minutes of gap in the data, filled from 1.52 kWh to make new average 87.74 kWh (percent 2%)
2024-09-29 09:51:57.403337: Info: record_status Read-Only
2024-09-29 09:55:06.009469: Warn: Historical day 2 has 1410 minutes of gap in the data, filled from 1.68 kWh to make new average 80.8 kWh (percent 2%)
2024-09-29 09:55:12.251511: Info: record_status Read-Only

LeeFarm11, 2024-09-29:
Thanks @gcoan 

Plan now being produced. So yes, it seems the apps.yaml logic might need reviewing to allow for 3 phase Inverters (and AIO?) or some change needed at GivTCP end?

Now back to the problem of not being able to charge or discharge in predbat via GivTCP on 3 phase Inverters.......

gcoan, 2024-09-29:
glad I got you sorted with Predbat, its the battery_stack_kwh sensor that throws the apps.yaml regular expression off.

I'll have to see if I can find a way to adjust the regular expression so it ignores these sensors, or just add more documentation to explain how to override it.

gcoan, 2024-09-29:
@LeeFarm11 I saw you sent a message about how long predbat was taking to run with the logfile:

```
2024-09-29 09:43:07.902200: Warn: Historical day 2 has 1425 minutes of gap in the data, filled from 1.3 kWh to make new average 125.12 kWh (percent 1%)
2024-09-29 09:45:03.345688: Info: record_status Read-Only
2024-09-29 09:45:09.054404: Warn: Historical day 2 has 1420 minutes of gap in the data, filled from 1.41 kWh to make new average 101.4 kWh (percent 1%)
```

but I can’t see that message on github any more? It looks like Predbat is struggling with the history for your load_energy_today sensor, is repeatedly trying to load day 2 and finding its not got much data in it.  This may be perfectly valid in that you’ve only just got givtcp running, or it may be another issue?  The repeated crashes on givtcp could of course be the cause of no history.

What’s days_previous set to, maybe just set it to ‘1’ for now.

LeeFarm11, 2024-09-29:
Yes, I was only watching Log from the Addon section, rather than the proper Log File. I thought there was a problem but immediately after posting I realised it coincided with the five minute updates so I deleted the post. Here below is what that log is now showing. And I also attach the actual log file. 

days_previous is set to -
  days_previous:
    - 1
    - 2
    - 3

[predbat(8).log](https://github.com/user-attachments/files/17179379/predbat.8.log)

```
2024-09-29 10:20:05.523153: Warn: Historical day 2 has 1385 minutes of gap in the data, filled from 2.87 kWh to make new average 75.05 kWh (percent 4%)
2024-09-29 10:21:55.937205: Info: record_status Read-Only
2024-09-29 10:25:08.269676: Warn: Historical day 2 has 1380 minutes of gap in the data, filled from 2.98 kWh to make new average 71.6 kWh (percent 4%)
2024-09-29 10:25:14.318340: Info: record_status Read-Only
2024-09-29 10:30:05.544384: Warn: Historical day 2 has 1375 minutes of gap in the data, filled from 3.07 kWh to make new average 67.94 kWh (percent 5%)
2024-09-29 10:32:04.855551: Info: record_status Read-Only
2024-09-29 10:35:05.316052: Warn: Historical day 2 has 1370 minutes of gap in the data, filled from 3.18 kWh to make new average 65.49 kWh (percent 5%)
2024-09-29 10:35:10.113792: Info: record_status Read-Only
2024-09-29 10:40:06.098153: Warn: Historical day 2 has 1365 minutes of gap in the data, filled from 3.28 kWh to make new average 63.04 kWh (percent 5%)
2024-09-29 10:41:47.241241: Info: record_status Read-Only
2024-09-29 10:45:05.354462: Warn: Historical day 2 has 1360 minutes of gap in the data, filled from 3.37 kWh to make new average 60.6 kWh (percent 6%)
2024-09-29 10:45:10.050185: Info: record_status Read-Only
2024-09-29 10:50:05.373540: Warn: Historical day 2 has 1355 minutes of gap in the data, filled from 3.45 kWh to make new average 58.45 kWh (percent 6%)
2024-09-29 10:51:53.274296: Info: record_status Read-Only
2024-09-29 10:55:05.276981: Warn: Historical day 2 has 1350 minutes of gap in the data, filled from 3.54 kWh to make new average 56.69 kWh (percent 6%)
2024-09-29 10:55:10.697190: Info: record_status Read-Only
2024-09-29 11:00:05.034510: Warn: Historical day 2 has 1345 minutes of gap in the data, filled from 3.68 kWh to make new average 55.83 kWh (percent 7%)
2024-09-29 11:01:55.082990: Info: record_status Read-Only
2024-09-29 11:05:05.008481: Warn: Historical day 2 has 1340 minutes of gap in the data, filled from 3.85 kWh to make new average 55.44 kWh (percent 7%)
2024-09-29 11:05:09.685605: Info: record_status Read-Only
2024-09-29 11:10:05.454677: Warn: Historical day 2 has 1335 minutes of gap in the data, filled from 3.96 kWh to make new average 54.35 kWh (percent 7%)
2024-09-29 11:11:49.518227: Info: record_status Read-Only
2024-09-29 11:15:05.532327: Warn: Historical day 2 has 1330 minutes of gap in the data, filled from 4.22 kWh to make new average 55.2 kWh (percent 8%)
2024-09-29 11:15:10.030621: Info: record_status Read-Only
2024-09-29 11:20:05.731228: Warn: Historical day 2 has 1325 minutes of gap in the data, filled from 4.48 kWh to make new average 56.14 kWh (percent 8%)
2024-09-29 11:21:48.650413: Info: record_status Read-Only
2024-09-29 11:25:06.300614: Warn: Historical day 2 has 1320 minutes of gap in the data, filled from 4.78 kWh to make new average 57.4 kWh (percent 8%)
2024-09-29 11:25:13.973498: Info: record_status Read-Only
2024-09-29 11:30:05.097872: Warn: Historical day 2 has 1315 minutes of gap in the data, filled from 4.88 kWh to make new average 56.26 kWh (percent 9%)
2024-09-29 11:31:51.860161: Info: record_status Read-Only
2024-09-29 11:35:04.972137: Warn: Historical day 2 has 1310 minutes of gap in the data, filled from 4.98 kWh to make new average 55.2 kWh (percent 9%)
2024-09-29 11:35:09.550005: Info: record_status Read-Only
2024-09-29 11:40:05.373900: Warn: Historical day 2 has 1305 minutes of gap in the data, filled from 5.08 kWh to make new average 54.22 kWh (percent 9%)
2024-09-29 11:41:31.111268: Info: record_status Read-Only
2024-09-29 11:45:05.147341: Warn: Historical day 2 has 1300 minutes of gap in the data, filled from 5.2 kWh to make new average 53.52 kWh (percent 10%)
2024-09-29 11:45:09.831398: Info: record_status Read-Only
2024-09-29 11:50:05.507301: Warn: Historical day 2 has 1295 minutes of gap in the data, filled from 5.3 kWh to make new average 52.67 kWh (percent 10%)
2024-09-29 11:51:42.739599: Info: record_status Read-Only
2024-09-29 11:55:04.741183: Warn: Historical day 2 has 1290 minutes of gap in the data, filled from 5.4 kWh to make new average 51.84 kWh (percent 10%)
2024-09-29 11:55:09.446464: Info: record_status Read-Only
2024-09-29 12:00:05.680316: Warn: Historical day 2 has 1285 minutes of gap in the data, filled from 5.48 kWh to make new average 50.94 kWh (percent 11%)
2024-09-29 12:01:49.956470: Info: record_status Read-Only
2024-09-29 12:05:05.383478: Warn: Historical day 2 has 1280 minutes of gap in the data, filled from 5.58 kWh to make new average 50.25 kWh (percent 11%)
2024-09-29 12:05:11.607432: Info: record_status Read-Only
2024-09-29 12:10:04.859724: Warn: Historical day 2 has 1275 minutes of gap in the data, filled from 5.68 kWh to make new average 49.6 kWh (percent 11%)
2024-09-29 12:11:51.217065: Info: record_status Read-Only
2024-09-29 12:15:08.967444: Warn: Historical day 2 has 1270 minutes of gap in the data, filled from 5.98 kWh to make new average 50.68 kWh (percent 12%)
[... cut ...]
2024-09-29 14:20:05.679449: Warn: Historical day 2 has 1145 minutes of gap in the data, filled from 12.53 kWh to make new average 61.18 kWh (percent 20%)
2024-09-29 14:21:35.909773: Info: record_status Read-Only
2024-09-29 14:25:05.230701: Warn: Historical day 2 has 1140 minutes of gap in the data, filled from 12.53 kWh to make new average 60.16 kWh (percent 21%)
2024-09-29 14:25:11.369398: Info: record_status Read-Only
2024-09-29 14:30:05.535750: Warn: Historical day 2 has 1135 minutes of gap in the data, filled from 12.53 kWh to make new average 59.17 kWh (percent 21%)
```

gcoan, 2024-09-29:
It looks like the history is building up, but to stop the warnings you could reduce the days previous. Anyway it's just a warning , doesn't stop predbat

LeeFarm11, 2024-09-29:
I have updated  now to 1 day and the 'Warn: Historical day 2 .... ' records have disappeared. Will add 2 and 3 back in later. Thanks.

gcoan, 2024-09-29:
> I have updated  now to 1 day and the 'Warn: Historical day 2 .... ' records have disappeared. Will add 2 and 3 back in later. Thanks.

The amount of missing day 2 data was reducing every 5 minute Predbat run so it is building up OK, but yes, better to stop the warnings until you've got the history in place

LeeFarm11, 2024-09-29:
Back to testing Enable Charge Schedule. 
Dev 3.0.12

Force Charge Enable and Force AC Charge Enable switches appear to be working. EXCEPT there is something very wrong in the log file responses. See below how the response is becoming more congested each time. Same as reported back with Beta 3.0.0c few weeks ago.

And then tried to switch Enable Charge Schedule to ON. Unfortunately it failed. Again see results in table below.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:////Users/leeandnessahome/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip.htm">
<link rel=File-List
href="file:////Users/leeandnessahome/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip_filelist.xml">
<style>

</style>
</head>

<body link="#467886" vlink="#96607D">

Switch | Current State | Changed State | Successful | Log entries
-- | -- | -- | -- | --
Force Charge Enable | ON | OFF | Yes | 2024-09-29 15:49:19,805 - mqtt -   [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setForceCharge=   disable     2024-09-29 15:49:23,363 - read - [DEBUG] - Command: setForceCharge was   sent: {'state': 'disable'}     2024-09-29 15:49:23,510 - read - [DEBUG] - responses is: [{'id':   'setForceCharge', 'result': '{"result": {}}'}]
Force AC Charge Enable | ON | OFF | Yes | 2024-09-29 15:49:53,243 - mqtt -   [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setACCharge=   disable     2024-09-29 15:49:56,757 - read - [DEBUG] - Command: setACCharge was sent:   {'state': 'disable'}     2024-09-29 15:49:56,896 - read - [DEBUG] - responses is: [{'id':   'setForceCharge', 'result': '{"result": {}}'}]
Force Charge Enable | OFF | ON | Yes | 2024-09-29 15:57:10,314 - mqtt -   [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setForceCharge=   enable     2024-09-29 15:57:21,692 - read - [DEBUG] - Command: setForceCharge was   sent: {'state': 'enable'}     2024-09-29 15:57:21,827 - read - [DEBUG] - responses is: [{'id':   'setForceCharge', 'result': '{"result": {}}'}, {'id':   'setACCharge', 'result': '{"result": {}}'}]
Force AC Charge Enable | OFF | ON | Yes | 2024-09-29 15:57:30,529 - mqtt -   [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setACCharge=   enable     2024-09-29 15:57:30,560 - read - [DEBUG] - Command: setACCharge was sent:   {'state': 'enable'}     2024-09-29 15:57:30,697 - read - [DEBUG] - responses is: [{'id':   'setForceCharge', 'result': '{"result": {}}'}, {'id':   'setACCharge', 'result': '{"result": {}}'}, {'id':   'setForceCharge', 'result': '{"result": {}}'}]
Force Charge Enable | ON | OFF | Yes | 2024-09-29 16:01:11,986 - mqtt -   [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setForceCharge=   disable     2024-09-29 16:01:20,345 - read - [DEBUG] - Command: setForceCharge was   sent: {'state': 'disable'}     2024-09-29 16:01:23,980 - read - [DEBUG] - responses is: [{'id':   'setForceCharge', 'result': '{"result": {}}'}, {'id':   'setACCharge', 'result': '{"result": {}}'}, {'id':   'setForceCharge', 'result': '{"result": {}}'}, {'id':   'setACCharge', 'result': '{"result": {}}'}]
Force AC Charge Enable | ON | OFF | Yes | 2024-09-29 16:01:40,767 - mqtt -   [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setACCharge=   disable     2024-09-29 16:01:53,145 - read - [DEBUG] - Command: setACCharge was sent:   {'state': 'disable'}     2024-09-29 16:01:53,285 - read - [DEBUG] - responses is: [{'id':   'setForceCharge', 'result': '{"result": {}}'}, {'id':   'setACCharge', 'result': '{"result": {}}'}, {'id':   'setForceCharge', 'result': '{"result": {}}'}, {'id':   'setACCharge', 'result': '{"result": {}}'}, {'id':   'setForceCharge', 'result': '{"result": {}}'}]
Enable Charge Schedule | OFF | ON | No | 2024-09-29 16:05:53,397 - mqtt -   [DEBUG] - MQTT Message Recieved:   GivEnergy/control/TD2343G049/enableChargeSchedule= enable     2024-09-29 16:05:53,425 - read - [DEBUG] - Write Request recieved     2024-09-29 16:05:53,426 - read - [DEBUG] - Command: enableChargeSchedule   was sent: {'state': 'enable'}     2024-09-29 16:05:53,430 - read - [DEBUG] - responses is: [{'id':   'setForceCharge', 'result': '{"result": {}}'}, {'id':   'setACCharge', 'result': '{"result": {}}'}, {'id': 'setForceCharge',   'result': '{"result": {}}'}, {'id': 'setACCharge', 'result':   '{"result": {}}'}, {'id': 'setForceCharge', 'result':   '{"result": {}}'}, {'id': 'setACCharge', 'result':   '{"result": {}}'}]

</body>

</html>

LeeFarm11, 2024-09-29:
I tried turning on predbat to see if gives any additional information. Here is one complete 5 minute cycle.
I do see this - 

2024-09-29 21:58:06.843500: Warn: Inverter 0 Trying to write scheduled_charge_enable to True didn't complete got off
2024-09-29 21:58:06.897224: Info: record_status Warn: Inverter 0 write to scheduled_charge_enable failed

```
2024-09-29 21:56:55.507723: --------------- PredBat - update at 2024-09-29 21:55:00+01:00 with clock skew 0 minutes, minutes now 1315
2024-09-29 21:56:55.509284: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2024-09-29 21:56:55.563036: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 51.9 minutes
2024-09-29 21:56:55.563162: Predbat /config/predbat.py version v8.4.9 currently running, latest version is v8.4.9 latest beta v8.4.9
2024-09-29 21:56:55.563234: Autoupdate: Currently up to date
2024-09-29 21:56:55.659661: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-09-29 21:56:55.660525: Predbat mode is set to Control charge & discharge
2024-09-29 21:56:56.267862: Found 2881 load_today datapoints going back 2 days
2024-09-29 21:56:56.960176: Car charging hold True threshold 6.0
2024-09-29 21:56:56.960307: Current data so far today: load 34.5 kWh import 6.2 kWh export 0.4 kWh pv 31.1 kWh
2024-09-29 21:56:56.960358: Fetching futurerate data from None
2024-09-29 21:56:56.961875: Adding rate import: {'start': '01:00:00', 'end': '06:00:00', 'rate': 11.84} => 09-29 01:00:00 to 09-29 06:00:00 @ 11.84 date None increment False
2024-09-29 21:56:56.963169: Adding rate import: {'start': '06:00:00', 'end': '14:00:00', 'rate': 26.55} => 09-29 06:00:00 to 09-29 14:00:00 @ 26.55 date None increment False
2024-09-29 21:56:56.964686: Adding rate import: {'start': '14:00:00', 'end': '17:00:00', 'rate': 11.84} => 09-29 14:00:00 to 09-29 17:00:00 @ 11.84 date None increment False
2024-09-29 21:56:56.965513: Adding rate import: {'start': '17:00:00', 'end': '21:00:00', 'rate': 26.55} => 09-29 17:00:00 to 09-29 21:00:00 @ 26.55 date None increment False
2024-09-29 21:56:56.966437: Adding rate import: {'start': '21:00:00', 'end': '23:00:00', 'rate': 11.84} => 09-29 21:00:00 to 09-29 23:00:00 @ 11.84 date None increment False
2024-09-29 21:56:56.967282: Adding rate import: {'start': '23:00:00', 'end': '01:00:00', 'rate': 26.55} => 09-29 23:00:00 to 09-29 01:00:00 @ 26.55 date None increment False
2024-09-29 21:56:56.975963: Adding rate export: {'rate': 12} => 09-29 00:00:00 to 09-29 00:00:00 @ 12.0 date None increment False
2024-09-29 21:56:56.983310: Standing charge is set to 53.66 p
2024-09-29 21:56:57.048945: Adding rate import: {'date': '2024-09-10', 'start': '13:00:00', 'end': '14:00:00', 'rate': 0} => 09-29 13:00:00 to 09-29 14:00:00 @ 0.0 date 2024-09-10 00:00:00 increment False
2024-09-29 21:56:59.874137: Rate min forward looking: now 11.84 at end of forecast 11.84
2024-09-29 21:56:59.874729: Import rates min 11.84 max 26.55 average 20.42
2024-09-29 21:56:59.956307: Export rates min 12.0 max 12.0 average 12.0
2024-09-29 21:56:59.956411: Rate thresholds (for charge/discharge) are import 26.05p (0.0) export 12.0p (0.0)
2024-09-29 21:56:59.979283: High export rate found rates in range 12.0 to 12.0
2024-09-29 21:56:59.994121: Low Import rate found rates in range 11.84 to 11.84
2024-09-29 21:57:00.208149: Today's energy import 6.2 kWh export 0.4 kWh cost 127.26 p import 132.06 p export -4.8 p carbon 0.0 kg
2024-09-29 21:57:00.208235: Using Solcast integration from inside HA for solar forecast
2024-09-29 21:57:00.208370: PV Data for pv_forecast_today total 33.3 kWh
2024-09-29 21:57:00.208483: PV Data for pv_forecast_tomorrow total 13.03 kWh
2024-09-29 21:57:00.208592: PV Data for pv_forecast_d3 total 37.56 kWh
2024-09-29 21:57:00.208746: PV Data for pv_forecast_d4 total 47.16 kWh
2024-09-29 21:57:00.232304: PV Forecast for today is 33.3 (23.86 10% 48.07 90%) kWh and left today is 0.0 (0.0 10% 0.0 90%) kWh
2024-09-29 21:57:00.303699: PV Forecast for day tomorrow is 13.03 (8.03 10% 18.66 90%) kWh
2024-09-29 21:57:00.339166: PV Forecast for day d2 is 37.56 (10.15 10% 77.48 90%) kWh
2024-09-29 21:57:00.375348: PV Forecast for day d3 is 47.16 (13.38 10% 82.29 90%) kWh
2024-09-29 21:57:00.554471: Historical data totals for days [1] are [34.48] - min 34.48
2024-09-29 21:57:00.555184: Historical days now [1] weight [1]
2024-09-29 21:57:00.673575: Today's load divergence 118.6 % in-day adjustment 117.67 % damping 0.95x
2024-09-29 21:57:00.673686: Today's predicted so far 25.05 kWh with 3.32 kWh car/iBoost excluded and 1.2 kWh import ignored and 0 forecast extra.
2024-09-29 21:57:00.673748: Today's actual load so far 31.39 kWh with 1.72 kWh Car/iBoost excluded and 1.36 kWh import ignored.
2024-09-29 21:57:00.873686: Inverter 0: Type GE GivEnergy
2024-09-29 21:57:00.873949: Inverter 0 does not have timed pause support enabled
2024-09-29 21:57:00.874335: Invertor time 2024-09-29 20:56:09+00:00, Predbat computer time 2024-09-29 21:57:00.874247+01:00, difference -0.87 minutes
2024-09-29 21:57:00.874436: Reserve min: 4.0% Battery_min:4.0%
2024-09-29 21:57:00.874534: Inverter 0 with soc_max 21.17 kWh nominal_capacity 21.17 kWh battery rate raw 11000.0 w charge rate 11.0 kW discharge rate 11.0 kW battery_rate_min 200.0 w ac limit 11.5 kW export limit 11.5 kW reserve 4.0 % current_reserve 4.0 %
[... cut ...]
2024-09-29 21:59:55,659 - mqtt - [DEBUG] - Prepping HY2343G302 for publishing
2024-09-29 21:59:55,660 - mqtt - [DEBUG] - Prepping HY2343G446 for publishing
2024-09-29 21:59:55,660 - mqtt - [DEBUG] - Prepping Stack_0 for publishing
2024-09-29 21:59:55,661 - mqtt - [DEBUG] - Prepping HV_Battery_Stacks for publishing
2024-09-29 21:59:55,662 - mqtt - [DEBUG] - Prepping invertor for publishing
2024-09-29 21:59:55,662 - mqtt - [DEBUG] - Prepping Meter_ID_1 for publishing
2024-09-29 21:59:55,663 - mqtt - [DEBUG] - Prepping meters for publishing
```

gcoan, 2024-09-29:
@LeeFarm11 I have updated the predbat documentation to explain how geserial has to be manually set with GivTCP v3 for AIO's and 3 phase inverters

Draft here https://github.com/gcoan/batpred/blob/main/docs/apps-yaml.md#geserial

LeeFarm11, 2024-09-29:
Thanks @gcoan 

Those instructions as written would suggest (misleadingly) to a new user that predbat works on 3 phase Inverter if you install GivTCP v3. I would suggest a note to explain that predbat does not yet work with 3 phase Inverter.

gcoan, 2024-09-29:
True enough. I'll add a note 

Works fine in monitor mode 🥹

gcoan, 2024-09-29:
Added a note to the inverter setup doc https://github.com/gcoan/batpred/blob/main/docs/inverter-setup.md#givenergy-with-givtcp

LeeFarm11, 2024-10-01:
Hi @britkat1980 
Hi Mark. I guess you're busy with settling down v3. But do you think you will be able to find some time soon to progress with the issues I have detailed above? I'd love to get predbat up and running to save me having to mess about with settings in the Giv Portal on a daily basis. I hoped the solar installation would be fit, set up and forget, but without predbat it is a pain in the backside to prevent the Inverter importing at expensive times and to prevent wasting solar on really sunny days by letting the battery get to 100% SOC.
Thanks for all your efforts.

UtXtiQArZE, 2025-09-16:
is this being worked on? im getting a  Giv-HY8.0 V3 and would love to use predbatt on it

closed 2026-10-11 by britkat1980