britkat1980/giv_tcp #227: giv_tcp-ma Broken 2.4.8

ProximusAl, 2024-09-09:
Traceback (most recent call last):
  File "/app/startup.py", line 150, in <module>
    with open('/config/GivTCP/v2env.pkl', 'wb') as outp:
FileNotFoundError: [Errno 2] No such file or directory: '/config/GivTCP/v2env.pkl'

Docker will not start

ProximusAl, 2024-09-09:
Rolled back to 2.4.7 and I'm back up and running again.

britkat1980, 2024-09-09:
ah, are you using Addon or docker container?

ProximusAl, 2024-09-09:
> ah, are you using Addon or docker container?

Docker

britkat1980, 2024-09-09:
Ah ok, think I need to tweak where the data is saved… Addon uses /config/GivTCP. This will become the new mandatory config folder that you’ll want to map in v3

ProximusAl, 2024-09-09:
“ This will become the new mandatory config folder that you’ll want to map in v3”

Even with docker?  
At the moment, my docker command doesn’t map anything, should I add it now on V2?

This is what I’ve used for ages:

docker run -d \
--name GivTCP \
--net=host \
--restart=always \
-e TZ=Europe/London \
-e NUMINVERTORS=1 \
-e INVERTOR_IP_1=192.168.76.150 \
-e NUMBATTERIES_1=2 \
-e MQTT_OUTPUT=True \
-e MQTT_ADDRESS=192.168.76.2 \
-e MQTT_USERNAME=mqtt \
-e MQTT_PASSWORD=somethingsecret \
britkat/giv_tcp-ma:latest

ProximusAl, 2024-09-10:
This seems to be resolved in 2.4.9. Closing...

closed 2024-09-10 by ProximusAl