britkat1980/giv_tcp #228: Lost contact with one battery, but now two???

tonyvfield, 2024-09-11:
Hi,
I am hoping this is the right place to get some help. I know very little about HA so please make your answers and help in simple terms.

I have a plant setup with 2 inverters and 2 batteries. The system was running fine for many months but since august HA cant see one of the batteries. The inverters are on static IP, I have done a hard reset on the inverters, power cycled the router but to no avail. I have check the Static IP address and they match what HA finds.
Initial as you can see from the log HA find both inverters but then the subsequent actions fail on inv2. INV1 is still working.

2024-09-11 09:18:51,374 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-09-11 09:18:51,386 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-09-11 09:18:53,401 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-09-11 09:18:57,269 - startup - [CRITICAL] - Inverter CE2232G271 which is a Gen 1 - AC has been found at: 192.168.1.29
2024-09-11 09:18:57,475 - startup - [CRITICAL] - Inverter CE2232G255 which is a Gen 1 - AC has been found at: 192.168.1.32
2024-09-11 09:18:57,477 - startup - [INFO] - Searching for Inverters again
2024-09-11 09:18:57,488 - startup - [CRITICAL] - Running Redis
2024-09-11 09:18:57,499 - startup - [CRITICAL] - Setting up invertor: 1 of 2
2024/09/11 09:18:57 [notice] 69#69: using the "epoll" event method
2024/09/11 09:18:57 [notice] 69#69: nginx/1.20.2
2024/09/11 09:18:57 [notice] 69#69: OS: Linux 6.6.31-haos-raspi
2024/09/11 09:18:57 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/09/11 09:18:57 [notice] 69#69: start worker processes
2024/09/11 09:18:57 [notice] 69#69: start worker process 74
2024/09/11 09:18:57 [notice] 69#69: start worker process 75
2024/09/11 09:18:57 [notice] 69#69: start worker process 76
2024/09/11 09:18:57 [notice] 69#69: start worker process 77
2024-09-11 09:18:57,603 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-09-11 09:18:57,603 - startup - [CRITICAL] - Running Invertor (192.168.1.32) read loop every 15s
2024-09-11 09:18:57,612 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-09-11 09:18:57,621 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-09-11 09:18:57,634 - startup - [CRITICAL] - Serving Web Dashboard from port 3042
2024-09-11 09:18:57,661 - startup - [CRITICAL] - Setting up invertor: 2 of 2
2024-09-11 09:18:57,793 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-09-11 09:18:57,794 - startup - [CRITICAL] - Running Invertor (192.168.1.29) read loop every 15s
2024-09-11 09:18:57,817 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-09-11 09:18:57,839 - startup - [CRITICAL] - Starting Gunicorn on port 6346
2024-09-11 09:18:57,878 - startup - [CRITICAL] - Serving Web Dashboard from port 3043
[2024-09-11 09:18:58 +0100] [86] [INFO] Starting gunicorn 23.0.0
[2024-09-11 09:18:58 +0100] [86] [INFO] Listening at: http://0.0.0.0:6346 (86)
[2024-09-11 09:18:58 +0100] [86] [INFO] Using worker: sync
[2024-09-11 09:18:58 +0100] [100] [INFO] Booting worker with pid: 100
[2024-09-11 09:18:58 +0100] [101] [INFO] Booting worker with pid: 101
[2024-09-11 09:18:58 +0100] [102] [INFO] Booting worker with pid: 102
[2024-09-11 09:18:59 +0100] [81] [INFO] Starting gunicorn 23.0.0
[2024-09-11 09:18:59 +0100] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2024-09-11 09:18:59 +0100] [81] [INFO] Using worker: sync
[2024-09-11 09:18:59 +0100] [111] [INFO] Booting worker with pid: 111
[2024-09-11 09:18:59 +0100] [112] [INFO] Booting worker with pid: 112
[2024-09-11 09:18:59 +0100] [113] [INFO] Booting worker with pid: 113
2024-09-11 09:19:04,261 - Inv2 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2024-09-11 09:19:05,058 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 UPDATE  The latest version of `serve` is 14.2.3
 UPDATE  The latest version of `serve` is 14.2.3
 INFO  Accepting connections at http://localhost:3042
 INFO  Accepting connections at http://localhost:3043
2024-09-11 09:19:50,650 - Inv2 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-09-11 09:20:15,860 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...

gcoan, 2024-09-11:
Likely this is the inverter and GivTCP not talking to each other.

On the inverter that GivTCP is not able to read, try doing a 'reset to defaults' on the GivEnergy portal, this can often cure such 'not talking' errors.

An alternative fix is to delete all the PKL files from /config/GivTCP and restart the GivTCP add-on

tonyvfield, 2024-09-11:
> Likely this is the inverter and GivTCP not talking to each other.
>
> On the inverter that GivTCP is not able to read, try doing a 'reset to defaults' on the GivEnergy portal, this can often cure such [...]

Hi,
I have reset the one inverter that was playing up by using the portal. No change. Please next idea

yavinmaster, 2024-09-11:
Just a silly question - have you confirmed that the GivEnergy App and web portal can see both batteries OK?  I only ask as I had the issue that batteries were loosing communication with the Inverter after overvoltage alarms that they weren't properly recovering from.

tonyvfield, 2024-09-11:
Sorry, I should have put that in. Yes the App and the web portal can see and manage both inverters.
I have checked with GE and both inverters are on the same software and firmware. They dont see any difference between the two.

gcoan, 2024-09-11:
Try deleting the PKL files and restarting GivTCP?

What version of GivTCP are you using ?

tonyvfield, 2024-09-11:
I have deleted two pkl files. it was all I could see.
I have restarted Givtcp
I get the following trace
2024-09-11 19:57:06,543 - Inv2 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...
2024-09-11 19:58:35,942 - Inv2 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
2024-09-11 19:58:36,139 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_2/mqtt_client.py", line 402, in <module>
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
    rc = self._loop(timeout)
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1686, in _loop
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1686, in _loop
    rc = self.loop_read()
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2100, in loop_read
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2100, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
    rc = self._packet_handle()
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3814, in _packet_handle
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3814, in _packet_handle
    return self._handle_connack()
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3934, in _handle_connack
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3934, in _handle_connack
    on_connect(
    on_connect(
  File "/app/GivTCP_2/mqtt_client.py", line 386, in on_connect
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-09-11 19:58:57,746 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...

Still can see one inverter but not the other

[image]

tonyvfield, 2024-09-11:
Hold on for 1 hour please. The last set of mods seem to have kicked it into life. I also switched the 'old solftware' on and off and I think things are starting to work. But I want to see if it stays working

tonyvfield, 2024-09-11:
Yes its working. Many thanks to everyone.
I think the thing that sorted it was deleting the PKL files. 

What a star you all are.

gcoan, 2024-09-11:
Glad to hear you got it sorted 👍

closed 2024-09-16 by britkat1980