britkat1980/giv_tcp #229: 3 Phase inverter - impossible negative value

Valex1p, 2024-09-12:
3 phase inverter. Noticed it recently with Beta 2.4.741 but is still there in 3.0.0g

'sensor.td2343g090_inverter_power_out' generally runs positive and is true reflection of the inverter power output but every now and then flips negative for a period before returning positive. I've not noticed the same for other sensors but I only monitor history for a small number.

[image]

Valex1p, 2024-09-13:
I don't think I believe these readings either....

[image]
[image]
[image]

britkat1980, 2024-09-16:
Thanks, the load power is taken direct from the registers, so the anomolous data seems to be coming from there... I see similar on my setup. v3.1 will look to introduce a different datasmoothing algorithm to try and remove these odd readings.

Valex1p, 2024-09-16:
👍