britkat1980/giv_tcp #230: Crash on start

Hansibal, 2024-09-14:
Hi all,

I have tried running the addon, both version 2 and now the beta version 3, but every time I start the addon, it eventually crashes my Home Assistant.
I am running a Raspberry Pi 3/1gb with Home Assistant. 

After installing the addon, and starting it, it initially seems to start fine, the new config page is accessible, and my AIO and the Gateway are found. Shortly after (30-60 seconds), the home assistant becomes unresponsive, and the only way to get it to work again is by a hard reset.

Is this a know issue, what could it be (is a Pi 3 to old to run this?)

Thanks for any help.

Hans

Hansibal, 2024-09-14:
To add, this what is get from the log before the connection is lost:

2024-09-14 12:35:09,586 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-14 12:35:11,688 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-14 12:35:11,689 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-14 12:35:24,034 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.12:8899
2024-09-14 12:35:24,036 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-14 12:35:25,200 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-14 12:35:30,246 - Startup - startup     -  [INFO    ] - Inverter CH2344G241 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.12
2024-09-14 12:35:30,254 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.20:8899
2024-09-14 12:35:30,255 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-14 12:35:31,420 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-14 12:35:31,425 - Startup - startup     -  [INFO    ] - Inverter GW2335G398 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.20
2024-09-14 12:35:31,433 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.33:8899
2024-09-14 12:35:31,434 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-14 12:35:32,600 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-14 12:35:32,605 - Startup - startup     -  [INFO    ] - Inverter GW2335G398 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.33
2024-09-14 12:35:32,764 - Startup - startup     -  [INFO    ] - Inverter GW2335G398 IP Address is different, updating: 192.168.1.33 -> 192.168.1.20
2024-09-14 12:35:32,765 - Startup - startup     -  [INFO    ] - Inverter GW2335G398 IP Address is different, updating: 192.168.1.20 -> 192.168.1.33
2024-09-14 12:35:32,774 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-14 12:35:33,113 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-14 12:35:33,113 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-14 12:35:33,115 - Startup - startup     -  [INFO    ] - ====     http://192.168.1.25:8099/config.html             ====
2024-09-14 12:35:33,116 - Startup - startup     -  [INFO    ] - ====  Self Run is off, so no data collection is happening ====
2024-09-14 12:35:33,116 - Startup - startup     -  [INFO    ] - ====     Log into Web Gui and complete startup settings   ====
2024-09-14 12:35:33,117 - Startup - startup     -  [INFO    ] - ==============================================================
[2024-09-14 12:35:34 +0100] [79] [INFO] Starting gunicorn 23.0.0
[2024-09-14 12:35:34 +0100] [79] [INFO] Listening at: http://0.0.0.0:6345 (79)
[2024-09-14 12:35:34 +0100] [79] [INFO] Using worker: sync
[2024-09-14 12:35:34 +0100] [80] [INFO] Booting worker with pid: 80
[2024-09-14 12:35:34 +0100] [81] [INFO] Booting worker with pid: 81
[2024-09-14 12:35:34 +0100] [82] [INFO] Booting worker with pid: 82

gcoan, 2024-09-14:
If you have a single AIO then you should configure GivTCP to ONLY talk to the AIO, not the gateway.  And I think you also have to set num batteries to 0.

Multiple AIO's you configure GivTCP to the gateway and not the AIO's themselves, but it's the other way round for single AIO.

britkat1980, 2024-09-16:
> If you have a single AIO then you should configure GivTCP to ONLY talk to the AIO, not the gateway. And I think you also have to set num batteries to 0.

with v3 there is no setting for number of batteries so that is no longer needed.

britkat1980, 2024-09-16:
Crashing HA shouldn't happen. From your log above, you aren't actually running GivTCP yet, as you haven't configured the self-run setting on the web page. Can you enable that and see if it runs properly (no idea why that would be the case)

Hansibal, 2024-09-17:
Hi,

Thanks for your help so far.  I tried again to get this to run and get to the setup. The setup finds the AIO fine in the setup page and, I think I managed to save and restart it, but somewhere between starting the addon and trying to save the settings the entire HA crashes. HA usually comes back after a few minutes (rebooted).

I have got a little bit more information in the log. It does say SIGKILL somewhere and indicating it ran out of memory. I run this on a PI3 with 1gb RAM, have I just got the best excuse to update to a Pi5, or should this work on my system?

Thanks

2024-09-17 14:59:47,718 - Startup - startup     -  [INFO    ] - ====  Self Run is off, so no data collection is happening ====
2024-09-17 14:59:47,720 - Startup - startup     -  [INFO    ] - ====     Log into Web Gui and complete startup settings   ====
2024-09-17 14:59:47,720 - Startup - startup     -  [INFO    ] - ==============================================================
[2024-09-17 15:00:04 +0100] [79] [INFO] Starting gunicorn 23.0.0
[2024-09-17 15:00:04 +0100] [79] [INFO] Listening at: http://0.0.0.0:6345 (79)
[2024-09-17 15:00:04 +0100] [79] [INFO] Using worker: sync
[2024-09-17 15:00:05 +0100] [80] [INFO] Booting worker with pid: 80
[2024-09-17 15:00:05 +0100] [81] [INFO] Booting worker with pid: 81
[2024-09-17 15:00:05 +0100] [82] [INFO] Booting worker with pid: 82
2024-09-17 15:13:21,669 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-17 15:13:23,808 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-17 15:13:23,809 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-17 15:13:34,437 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.12:8899
2024-09-17 15:13:34,438 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-17 15:13:35,605 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-17 15:13:41,379 - Startup - startup     -  [INFO    ] - Inverter CH2344G241 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.12
2024-09-17 15:13:41,389 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.20:8899
2024-09-17 15:13:41,389 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-17 15:13:42,556 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-17 15:13:42,560 - Startup - startup     -  [INFO    ] - Inverter GW2335G398 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.20
2024-09-17 15:13:42,569 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.33:8899
2024-09-17 15:13:42,570 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-17 15:13:43,747 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-17 15:13:43,751 - Startup - startup     -  [INFO    ] - Inverter GW2335G398 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.33
2024-09-17 15:13:44,063 - Startup - startup     -  [INFO    ] - Inverter GW2335G398 IP Address is different, updating: 192.168.1.33 -> 192.168.1.20
2024-09-17 15:13:44,065 - Startup - startup     -  [INFO    ] - Inverter GW2335G398 IP Address is different, updating: 192.168.1.20 -> 192.168.1.33
2024-09-17 15:13:44,072 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-17 15:13:44,769 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-17 15:13:44,770 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-17 15:13:44,772 - Startup - startup     -  [INFO    ] - ====     http://192.168.1.25:8099/config.html             ====
2024-09-17 15:13:44,773 - Startup - startup     -  [INFO    ] - ====  Self Run is off, so no data collection is happening ====
2024-09-17 15:13:44,774 - Startup - startup     -  [INFO    ] - ====     Log into Web Gui and complete startup settings   ====
2024-09-17 15:13:44,775 - Startup - startup     -  [INFO    ] - ==============================================================
[2024-09-17 15:13:46 +0100] [79] [INFO] Starting gunicorn 23.0.0
[2024-09-17 15:13:46 +0100] [79] [INFO] Listening at: http://0.0.0.0:6345 (79)
[2024-09-17 15:13:46 +0100] [79] [INFO] Using worker: sync
[2024-09-17 15:13:47 +0100] [80] [INFO] Booting worker with pid: 80
[2024-09-17 15:13:47 +0100] [81] [INFO] Booting worker with pid: 81
[2024-09-17 15:13:47 +0100] [82] [INFO] Booting worker with pid: 82
2024-09-17 15:14:00,838 - Inv1 - write       -  [CRITICAL] - Restarting the GivTCP Addon in 2s...
2024-09-17 15:14:25,114 - Inv1 - write       -  [CRITICAL] - Restarting the GivTCP Addon in 2s...
2024-09-17 15:14:44,366 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-17 15:14:46,525 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-17 15:14:46,527 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-17 15:14:58,936 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.12:8899
2024-09-17 15:14:58,937 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-17 15:15:00,103 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-17 15:15:08,397 - Startup - startup     -  [INFO    ] - Inverter CH2344G241 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.12
[... cut ...]
    output=wr.rebootAddon()
           ^^^^^^^^^^^^^^^^
  File "/app/GivTCP_1/write.py", line 1475, in rebootAddon
    return json.dumps(result)
                      ^^^^^^
UnboundLocalError: cannot access local variable 'result' where it is not associated with a value
[2024-09-17 15:15:54 +0100] [82] [INFO] Worker exiting (pid: 82)
[2024-09-17 15:15:54 +0100] [79] [ERROR] Worker (pid:82) was sent SIGKILL! Perhaps out of memory?
[2024-09-17 15:15:54 +0100] [85] [INFO] Booting worker with pid: 85

gcoan, 2024-09-17:
Do you know if you have a Pi 3A or 3B?

The Home Assistant documentation says a 3A doesn't have enough RAM https://www.home-assistant.io/installation/raspberrypi and a 3B is "enough to get started"

It may be that running GivTCP on a Pi 3 is just too much for it especially if you only have 1Gb of RAM.  I run my Home Assistant as a VM on a windows machine and it says minimum recommended memory is 2Gb (I have mine set to 3Gb)

Hansibal, 2024-09-17:
It's a 3B. I think I have just found the excuse needed to upgrade ;)

Hansibal, 2024-09-19:
Just to update,, it seems to have indeed been a memory issue for the PI 3B, as it runs smooth (so far, after 5 minutes testing) on a Pi 5. Thanks!

gcoan, 2024-09-19:
Brilliant news. 

As an aside, could you share where you bought the pi and ssd from, I'm thinking about moving my VM to a PI 4 or 5. 

Cheers Geoffrey

closed 2024-10-01 by britkat1980