britkat1980/giv_tcp #231: Appears to keep restarting

yavinmaster, 2024-09-14:
I've been running giv_tcp for over a year with very few issues.  Recently I have been having the occasional issue of it not updating properly.  I added an automation in Home Assistant to warn me if the "time since last update" was too long.  Today, I noticed the data wasn't updating but the automation warning hadn't run.  On investigation, I found my inverter had changed IP address at around 8:15am (according to when the log files started to show errors)????  I change the settings and all appeared to be working again.  This is all background in case it is important.

What I have however noted, is that watching the DEBUG logs it appears to keep restarting itself with something like below happening every 25 seconds.  My self-run timer is set to 20 seconds.  Is this normal?  I just didn't know if "-----Starting-----" means the whole system is starting or the fetch is starting.  I've looked back at older logs and if I look at the start of this month, it wasn't doing this.  If I look at yesterday it was.....I'm wondering if this is an indication of the other issues I've been seeing.

My installation is on docker and I've just updated to the latest version (2.4.9?) today and am seeing the same results.

`2024-09-14 14:34:06,045 - mqtt - [DEBUG] - Publishing: GivEnergy/ED2248G541/raw
2024-09-14 14:34:06,046 - mqtt - [DEBUG] - Prepping invertor for publishing
2024-09-14 14:34:26,071 - read - [DEBUG] - ----------------------------Starting----------------------------
2024-09-14 14:34:26,071 - read - [DEBUG] - Getting All Registers
2024-09-14 14:34:26,072 - read - [DEBUG] - Connecting to: 192.168.30.45
2024-09-14 14:34:26,072 - queue - [DEBUG] - Pushed job 5fc64b83-86d4-49a8-b4c2-0c33f1937bee into GivTCP_1
2024-09-14 14:34:26,232 - sync - [DEBUG] - Connection to Modbus server established. Socket ('172.30.0.9', 58309)`

britkat1980, 2024-09-16:
The starting message is there in the debug logging to signify the start of the read loop. The reason its at 25s is that the read process takes 5s and then it waits for 20s for the next one.
GivTCP is not always restarting this is expected behaviour. My guess is that you didn't have debug logs on before so you won't have seen them previously.
I don't recommend keeping debug logs on if not needed.

yavinmaster, 2024-09-16:
Thanks for the clarification.

closed 2024-09-16 by yavinmaster