britkat1980/giv_tcp #233: Docker 3.0.0

ProximusAl, 2024-09-14:
On the just released dockerhub on 3.0.0 I now get this:

2024-09-14 19:24:23,116 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)

Constant looping. Will rollback

ProximusAl, 2024-09-14:
Hmmm,  I ran through the config again, and it now seems to work. 

Took a few attempts to get MQTT server settings to stick on Enable. 

Will monitor…

ProximusAl, 2024-09-14:
So I rolled back to 2.4.9 and started again from scratch, then re-upgraded to 3.0.0. 

This is giv_tcp-ma running under docker. 

I got the same issue as before, but it seems to be the EV charger config that was the issue. I don’t have one, and it looks like it’s enabled by default.  Went to config page and Disabled and it now seems to be fine. 

One thing I did notice, is in V3, all the values I use on my dashboard now have .0 after everything. 
Inverter SoC on 2.4.9 would say 84%, now on 3.0.0 it says 84.0% which looks weird in a gauge. 

Is this going to be new going forward? If so, I’ll have to format it on my existing dash to remove the .0 from everything.

ProximusAl, 2024-09-16:
OK, so I've got around the .0 issue on all values by formatting custom sensors for my dashboard.

    dashboard_soc:
      friendly_name: "Battery Charge Level"
      unit_of_measurement: "%"
      value_template: >
        {% if is_state('sensor.givtcp_edXXXXgyyy_soc', 'unavailable') or is_state('sensor.givtcp_edXXXXgyyy_soc', 'unknown') %}
          0
        {% else %}
          {{ states('sensor.givtcp_edXXXXgyyy_soc')|float(0)|int }}
        {% endif %}

closed 2024-09-16 by ProximusAl

gcoan, 2024-09-16:
@ProximusAl personally I don't think this should be closed unless @britkat1980 says this is working to specification/design.

You've put a cludge in to get it to look the same as GivTCP v2.4, that's not a fix

ProximusAl, 2024-09-16:
All right then :)

gcoan, 2024-09-16:
> All right then :)

You know it's the right thing to do 😉

britkat1980, 2024-09-16:
I'll put a catch in the startup script to avoid the NameError issue.
The addition of the .0 is a mistake, I think I was trying to track down an error and was forcing values to a float, which I guess has done this. I'll revert as I don't think that was the issue anyway.

pjnewcombe, 2024-09-18:
I have the same issue. What is the syntax please to add to a docker compose file to disable the EVC? Thanks.

ProximusAl, 2024-09-25:
Resolved with 3.0.1.