britkat1980/giv_tcp #234: Possible Version 3 issue

ProximusAl, 2024-09-17:
Since upgrading to V3.0.0 of Docker, HA is spewing this a lot.

2024-09-17 09:35:10.482 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.givtcp_edXXXXg111_self_consumption_energy_today_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 2.3 (2.4) with last_updated set to 2024-09-17T08:32:16.029948+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

2024-09-17 09:40:10.482 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.givtcp_edXXXXg111_load_energy_today_kwh from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 9.4 (9.5) with last_updated set to 2024-09-17T08:38:54.810465+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

It's asking to create a bug in HA Core, but I'm not convinced it's there where the issue lies, as I didnt get this in previous GivTCP V2.4.9.

gcoan, 2024-09-17:
@ProximusAl I (and others) get this frequently on GivTCP 2.4.3, I'm surprised you don't.

Previously reported and was due to be fixed in v3, so if its occurring for you then clearly it must still be awry 

https://github.com/britkat1980/giv_tcp/issues/213

ProximusAl, 2024-09-17:
:(  Yeah, I've never really seen it before, and I keep a decent eye on the HA logs....

britkat1980, 2024-09-18:
missed this in v3, will come in next point release

gcoan, 2024-09-19:
Possibly related, I'm seeing this error in the HA system log for my two gen 1 hybrid inverters:

Entity sensor.givtcp_xxxxx_soc_kwh (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

britkat1980, 2024-09-19:
You running a dev version for that? I changed to measurement, but will make it “total”

gcoan, 2024-09-19:
> You running a dev version for that? I changed to measurement, but will make it “total”

No, I upgraded to v3.0.0 prod version last night.  Not running the dev version at all

ProximusAl, 2024-09-25:
This also seems resolved in 3.0.1.  Thanks

closed 2024-09-25 by ProximusAl