britkat1980/giv_tcp #235: V3 issue

SheriffOfNotts, 2024-09-18:
Configured according to documentation, but keep getting this in the logs?

2024-09-17 19:57:35,513 - Startup - startup     -  [ERROR   ] - Self Run loop process stuck. Killing and restarting...
2024-09-17 19:57:35,513 - Startup - startup     -  [INFO    ] - Restarting Invertor read loop every 30s
2024-09-17 19:57:35,519 - Startup - startup     -  [ERROR   ] - Self Run loop process stuck. Killing and restarting...
2024-09-17 19:57:35,521 - Startup - startup     -  [INFO    ] - Restarting Invertor read loop every 30s
2024-09-17 19:57:37,065 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-17 19:57:37,071 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-17 19:57:37,122 - Inv2 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-17 19:57:37,126 - Inv2 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-17 19:57:44,312 - Inv1 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2024-09-17 19:57:52,400 - Inv2 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2024-09-17 19:57:55,339 - Inv1 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:57:57,343 - Inv1 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:01,348 - Inv1 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:03,504 - Inv2 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:05,508 - Inv2 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:09,353 - Inv1 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:09,512 - Inv2 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:17,517 - Inv2 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:25,357 - Inv1 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:33,522 - Inv2 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:58:57,362 - Inv1 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized
2024-09-17 19:59:05,528 - Inv2 - HA_Discovery -  [ERROR   ] - Bad connection Returned code= Not authorized

britkat1980, 2024-09-18:
In your web-gui config you should have an auto configured username and password for the MQTT broker, if you are using the HA addon. This will usually be a very long string for the password. You shouldn't change this. This is a different approach to v2. Did you make any changes to that?

britkat1980, 2024-09-18:
If you delete the username and password from the config and restart, it should auto detect the mqtt details and update them

SheriffOfNotts, 2024-09-18:
Thanks. I currently have v2 and v3 installed (using v2 due to issue). Do I need to uninstall v2 first or can it co-exist with v3, but disabled?

britkat1980, 2024-09-21:
No need to uninstall. I suggest stopping v2, deleting the allsettings.json file from your config folder, then restarting v3. This should then work. Noting you will need to enable self_run in the file, once it has started for the first time.

SheriffOfNotts, 2024-09-22:
Thank you. I've done as you suggested, but it's showing a lot of errors.. I also tried clearing out the MQTT GivTCP topics as you suggested on FB.

[2024-09-22 11:55:59 +0100] [87] [INFO] Booting worker with pid: 87
[2024-09-22 11:55:59 +0100] [88] [INFO] Booting worker with pid: 88
2024-09-22 11:56:04,325 - Inv2 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-22 11:56:04,371 - Inv2 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-22 11:56:07,157 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-22 11:56:07,197 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-22 11:56:14,160 - Inv1 - read        -  [INFO    ] - cache len= 0
2024-09-22 11:56:14,178 - Inv1 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2024-09-22 11:56:19,899 - Inv2 - read        -  [INFO    ] - cache len= 0
2024-09-22 11:56:19,939 - Inv2 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2024-09-22 11:57:08,967 - Inv1 - read        -  [INFO    ] - cache len= 1
2024-09-22 11:57:13,804 - Inv2 - read        -  [INFO    ] - cache len= 1
2024-09-22 11:57:42,284 - Inv1 - read        -  [INFO    ] - cache len= 2
2024-09-22 11:57:44,746 - Inv2 - read        -  [INFO    ] - cache len= 2
2024-09-22 11:58:15,602 - Inv1 - read        -  [INFO    ] - cache len= 3
2024-09-22 11:58:15,758 - Inv2 - read        -  [INFO    ] - cache len= 3
2024-09-22 11:58:47,929 - Inv2 - read        -  [INFO    ] - cache len= 4
2024-09-22 11:58:48,986 - Inv1 - read        -  [INFO    ] - cache len= 4
2024-09-22 11:59:19,053 - Inv2 - read        -  [INFO    ] - cache len= 5
2024-09-22 11:59:22,510 - Inv1 - read        -  [INFO    ] - cache len= 5
2024-09-22 11:59:50,197 - Inv2 - read        -  [INFO    ] - cache len= 6
2024-09-22 11:59:55,755 - Inv1 - read        -  [INFO    ] - cache len= 6
[2024-09-22 12:00:28 +0100] [80] [CRITICAL] WORKER TIMEOUT (pid:86)
[2024-09-22 12:00:28 +0100] [80] [CRITICAL] WORKER TIMEOUT (pid:87)
[2024-09-22 12:00:28 +0100] [80] [CRITICAL] WORKER TIMEOUT (pid:88)
[2024-09-22 12:00:31 +0100] [80] [ERROR] Worker (pid:86) was sent SIGKILL! Perhaps out of memory?
[2024-09-22 12:00:31 +0100] [80] [ERROR] Worker (pid:88) was sent SIGKILL! Perhaps out of memory?
--- Logging error ---
2024-09-22 12:00:31,243 - Inv1 - read        -  [INFO    ] - cache len= 7
2024-09-22 12:00:31,337 - Inv2 - read        -  [INFO    ] - cache len= 7
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1113, in emit
    stream.write(msg + self.terminator)
RuntimeError: reentrant call inside <_io.BufferedWriter name='<stderr>'>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1113, in emit
    stream.write(msg + self.terminator)
  File "/usr/local/lib/python3.11/site-packages/gunicorn/arbiter.py", line 241, in handle_chld
    self.reap_workers()
  File "/usr/local/lib/python3.11/site-packages/gunicorn/arbiter.py", line 553, in reap_workers
    self.log.error(msg)
  File "/usr/local/lib/python3.11/site-packages/gunicorn/glogging.py", line 271, in error
    self.error_log.error(msg, *args, **kwargs)
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1518, in error
    self._log(ERROR, msg, args, **kwargs)
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1634, in _log
    self.handle(record)
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1644, in handle
    self.callHandlers(record)
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1706, in callHandlers
    hdlr.handle(record)
  File "/usr/local/lib/python3.11/logging/__init__.py", line 978, in handle
    self.emit(record)
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1118, in emit
    self.handleError(record)
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1031, in handleError
    sys.stderr.write('--- Logging error ---\n')
RuntimeError: reentrant call inside <_io.BufferedWriter name='<stderr>'>
Call stack:
  File "/usr/local/bin/gunicorn", line 8, in <module>
    sys.exit(run())
  File "/usr/local/lib/python3.11/site-packages/gunicorn/app/wsgiapp.py", line 66, in run
    WSGIApplication("%(prog)s [OPTIONS] [APP_MODULE]", prog=prog).run()
  File "/usr/local/lib/python3.11/site-packages/gunicorn/app/base.py", line 235, in run
    super().run()
[... 8 lines cut ...]
  File "/usr/local/lib/python3.11/site-packages/gunicorn/arbiter.py", line 588, in spawn_worker
    worker = self.worker_class(self.worker_age, self.pid, self.LISTENERS,
  File "/usr/local/lib/python3.11/site-packages/gunicorn/workers/base.py", line 65, in __init__
    self.tmp = WorkerTmp(cfg)
  File "/usr/local/lib/python3.11/site-packages/gunicorn/workers/workertmp.py", line 23, in __init__
    fd, name = tempfile.mkstemp(prefix="wgunicorn-", dir=fdir)
  File "/usr/local/lib/python3.11/tempfile.py", line 341, in mkstemp
    return _mkstemp_inner(dir, prefix, suffix, flags, output_type)
  File "/usr/local/lib/python3.11/tempfile.py", line 256, in _mkstemp_inner
    fd = _os.open(file, flags, 0o600)
  File "/usr/local/lib/python3.11/site-packages/gunicorn/arbiter.py", line 241, in handle_chld
    self.reap_workers()
  File "/usr/local/lib/python3.11/site-packages/gunicorn/arbiter.py", line 553, in reap_workers
    self.log.error(msg)
[... cut ...]
[2024-09-22 12:00:31 +0100] [99] [INFO] Booting worker with pid: 99
[2024-09-22 12:00:31 +0100] [100] [INFO] Booting worker with pid: 100
[2024-09-22 12:00:43 +0100] [101] [INFO] Booting worker with pid: 101
2024-09-22 12:00:53,769 - Inv2 - read        -  [INFO    ] - cache len= 8
2024-09-22 12:01:02,415 - Inv1 - read        -  [INFO    ] - cache len= 8
2024-09-22 12:01:24,942 - Inv2 - read        -  [INFO    ] - cache len= 9
2024-09-22 12:01:35,641 - Inv1 - read        -  [INFO    ] - cache len= 9

britkat1980, 2024-09-22:
Looks like you’re running Dev there? Thats always risky as it’s often broken (it’s what I use to try code changes).
Perhaps run Beta

SheriffOfNotts, 2024-09-22:
Ah fat fingers when I was trying a re-install. I'll try again..

SheriffOfNotts, 2024-09-23:
Hi, I was still having problems so I uninstalled v2 & v3, deleted the GivTVP directory, deleted the related MQTT entries, fully restarted HA and re-installed v3. It's now showing the following repeated entries? I appreciate all your help..

 HTTP  9/23/2024 9:20:11 AM 192.168.0.20 Returned 304 in 14 ms
 HTTP  9/23/2024 9:20:11 AM 192.168.0.20 Returned 304 in 14 ms
 HTTP  9/23/2024 9:20:11 AM 192.168.0.20 Returned 304 in 13 ms
 HTTP  9/23/2024 9:20:11 AM 192.168.0.20 Returned 304 in 13 ms
 HTTP  9/23/2024 9:20:11 AM 192.168.0.20 Returned 304 in 13 ms
 HTTP  9/23/2024 9:20:11 AM 192.168.0.20 GET /app.json
 HTTP  9/23/2024 9:20:11 AM 192.168.0.20 Returned 304 in 2 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 200 in 4 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /css/styles.css
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 5 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /js/app.js
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /js/libraries/jquery.js
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 6 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 6 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /js/enums.js
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /js/sensors.js
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /js/formatters.js
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 7 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 8 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 10 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /js/helpers.js
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /js/converters.js
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 5 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 5 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /app.json
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 3 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 200 in 6 ms
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 GET /css/styles.css
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 GET /js/libraries/jquery.js
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 GET /js/app.js
 HTTP  9/23/2024 9:32:41 AM 192.168.0.20 Returned 304 in 43 ms
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 Returned 304 in 40 ms
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 Returned 304 in 37 ms
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 GET /js/formatters.js
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 GET /js/sensors.js
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 GET /js/enums.js
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 GET /js/helpers.js
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 GET /js/converters.js
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 Returned 304 in 43 ms
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 Returned 304 in 41 ms
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 Returned 304 in 45 ms
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 Returned 304 in 48 ms
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 Returned 304 in 45 ms
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 GET /app.json
 HTTP  9/23/2024 9:32:42 AM 192.168.0.20 Returned 200 in 7 ms

britkat1980, 2024-09-23:
These logs are there when you visit the web dashboard. Should really be suppressed, but I haven’t been able to do that properly

SheriffOfNotts, 2024-09-23:
No problem. My concern is that it doesn't seem to be working - errors in predbat and nothing showing on the GivTCP dashboard? I've re-started and current logs are (nothing since 10:57, logs from 11:33):
2024-09-23 10:14:00,279 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-23 10:14:02,326 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-23 10:14:02,327 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-23 10:14:12,843 - Startup - client      -  [INFO    ] - Connection established to 192.168.0.46:8899
2024-09-23 10:14:12,844 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-23 10:14:14,014 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-23 10:14:14,017 - Startup - startup     -  [INFO    ] - Inverter GW2315G079 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.0.46
2024-09-23 10:14:14,029 - Startup - client      -  [INFO    ] - Connection established to 192.168.0.119:8899
2024-09-23 10:14:14,032 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-23 10:14:15,193 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-23 10:14:23,737 - Startup - startup     -  [INFO    ] - Inverter CH2314G037 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.0.119
2024-09-23 10:14:23,764 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-23 10:14:23,867 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-23 10:14:23,867 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-23 10:14:23,867 - Startup - startup     -  [INFO    ] - ====     http://192.168.0.129:8099/config.html             ====
2024-09-23 10:14:23,868 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-23 10:14:23,871 - Startup - startup     -  [INFO    ] - Running Invertor 1 (GW2315G079) read loop every 30/120s
2024-09-23 10:14:23,880 - Startup - startup     -  [INFO    ] - Setting up invertor: 2
2024-09-23 10:14:24,130 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-23 10:14:24,133 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-23 10:14:24,135 - Startup - startup     -  [INFO    ] - ====     http://192.168.0.129:8099/config.html             ====
2024-09-23 10:14:24,138 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-23 10:14:24,146 - Startup - startup     -  [INFO    ] - Running Invertor 2 (CH2314G037) read loop every 30/120s
2024-09-23 10:14:24,159 - Startup - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2024-09-23 10:14:25 +0100] [77] [INFO] Starting gunicorn 23.0.0
[2024-09-23 10:14:25 +0100] [77] [INFO] Listening at: http://0.0.0.0:6345 (77)
[2024-09-23 10:14:25 +0100] [77] [INFO] Using worker: sync
[2024-09-23 10:14:25 +0100] [92] [INFO] Booting worker with pid: 92
[2024-09-23 10:14:25 +0100] [93] [INFO] Booting worker with pid: 93
[2024-09-23 10:14:25 +0100] [94] [INFO] Booting worker with pid: 94
[2024-09-23 10:14:25 +0100] [80] [INFO] Starting gunicorn 23.0.0
[2024-09-23 10:14:25 +0100] [80] [INFO] Listening at: http://0.0.0.0:6346 (80)
[2024-09-23 10:14:25 +0100] [80] [INFO] Using worker: sync
[2024-09-23 10:14:26 +0100] [95] [INFO] Booting worker with pid: 95
[2024-09-23 10:14:26 +0100] [96] [INFO] Booting worker with pid: 96
[2024-09-23 10:14:26 +0100] [97] [INFO] Booting worker with pid: 97
2024-09-23 10:14:30,704 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-23 10:14:30,717 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-23 10:14:31,419 - Inv2 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-23 10:14:31,439 - Inv2 - read        -  [CRITICAL] - Detecting inverter characteristics...
 INFO  Accepting connections at http://localhost:3000
2024-09-23 10:14:38,042 - Inv1 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2024-09-23 10:14:51,290 - Inv2 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
 HTTP  9/23/2024 10:56:59 AM 192.168.0.20 GET /
 HTTP  9/23/2024 10:56:59 AM 192.168.0.20 Returned 304 in 147 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /css/styles.css
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /js/enums.js
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /js/app.js
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /js/sensors.js
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /js/libraries/jquery.js
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /js/helpers.js
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 111 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 107 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 105 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /js/formatters.js
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /js/converters.js
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 129 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 121 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 50 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 49 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 152 ms
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 GET /app.json
 HTTP  9/23/2024 10:57:00 AM 192.168.0.20 Returned 304 in 10 ms

[image]

britkat1980, 2024-10-01:
Is this still unresolved? Does data come through properly to HA?

SheriffOfNotts, 2024-10-01:
Hi, tried everything and couldn’t resolve it. In the end I restored to 2.4.9 and it’s worked flawlessly. Any idea what caused it?

closed 2025-10-23 by britkat1980