britkat1980/giv_tcp #236: How to migrate docker config from V2 -> V3

pjnewcombe, 2024-09-18:
Thank you for the amazing work that has gone into this project. We have derived great value for nearly 2 years.

I run giv_tcp as a stand alone docker container. I had to rebuild my home server yesterday and it apparently pulled the new V3 docker image. Apparently this is no longer compatible with my pre-existing docker compose config, which I've pasted below. 

Any pointers would be very welcome

**Works in 2.4.9**

```
NUMINVERTORS=1
INVERTOR_IP_1=1
NUMBATTERIES_1=1
TZ="Europe/London"
MQTT_OUTPUT=True
MQTT_ADDRESS=[My server IP]
HA_AUTO_D=False
DAYRATE=0.3103
DAYRATESTART="05:30"
NIGHTRATE="23:30"
SMARTTARGET=False
```

**Attempted edit for 3.0.0 (but does not work)**

```
NUMINVERTORS=1
INVERTOR_IP_1=1
NUMBATTERIES_1=1
EVC_ENABLE=False
TZ="Europe/London"
MQTT_OUTPUT=True
MQTT_ADDRESS=[My server IP]
HA_AUTO_D=False
DAYRATE=0.3103
DAYRATESTART="05:30"
NIGHTRATE="23:30"
SMARTTARGET=False
```

Which results in the following error output with the V3  docker container:

```
Sep 18 19:12:56 server givtcp[254610]: 2024-09-18 19:12:56,825 - Startup - startup     -  [INFO    ] - Searching for Inverters
Sep 18 19:12:56 server givtcp[254610]: 2024-09-18 19:12:56,826 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
Sep 18 19:13:23 server givtcp[254610]: 2024-09-18 19:13:23,274 - Startup - startup     -  [INFO    ] - No inverters found...
Sep 18 19:13:23 server givtcp[254610]: 2024-09-18 19:13:23,274 - Startup - startup     -  [ERROR   ] - =============================================================
Sep 18 19:13:23 server givtcp[254610]: 2024-09-18 19:13:23,274 - Startup - startup     -  [ERROR   ] - ====               NO INVERTERS FOUND                    ====
Sep 18 19:13:23 server givtcp[254610]: 2024-09-18 19:13:23,274 - Startup - startup     -  [ERROR   ] - ====      add manually using a file editor to modify     ====
Sep 18 19:13:23 server givtcp[254610]: 2024-09-18 19:13:23,274 - Startup - startup     -  [ERROR   ] - ====          /config/GivTCP/allsettings.json            ====
Sep 18 19:13:23 server givtcp[254610]: 2024-09-18 19:13:23,274 - Startup - startup     -  [ERROR   ] - =============================================================
Sep 18 19:13:23 server givtcp[254610]: 2024-09-18 19:13:23,292 - Startup - startup     -  [INFO    ] - EVC IP is missing from config. Please update and restart GivTCP
Sep 18 19:13:23 server givtcp[254610]: 2024-09-18 19:13:23,293 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
```

Am  I missing something obvious I need to change in the docker compose?

ProximusAl, 2024-09-19:
Go to here: http://<<HA-IP>>:8099/config.html and re-configure your config, ensure you disable the EVC if you dont have one.

That will get it running.

ProximusAl, 2024-09-19:
Also, if using Docker, you need to map a new volume to store the settings in:
sudo docker run -d \
--name GivTCP \
--net=host \
--restart=always \
-e TZ=Europe/London \
-e NUMINVERTORS=1 \
-e INVERTOR_IP_1=192.168.76.150 \
-e NUMBATTERIES_1=2 \
-e MQTT_OUTPUT=True \
-e MQTT_ADDRESS=192.168.76.2 \
-e MQTT_USERNAME=meh \
-e MQTT_PASSWORD=meh \
**-v /home/alan/docker/givtcp/config:/config/GivTCP/ \**
britkat/giv_tcp-ma:latest

pjnewcombe, 2024-09-19:
Thanks @ProximusAl. I should have been clearer, I didn't include the docker options in my paste above. I do set the volume (and have been wiping it to try to get this working). Thanks for the tip re web interface, though I am trying to do this declaratively (it is part of a NixOS installation), so I will revert back to 2.49 for now. 

A small update is I discovered I can get 3.0 working if I manually edit the "allsettings.json" that appears in the /config/GivTCP folder, and restart.

britkat1980, 2024-09-21:
v3 does not require any ENV to be set, it is all auto discovered and.-/or modified through the allsettings.json file in the mapped volume or via the Web GUI on port 8099

pjnewcombe, 2024-09-22:
> v3 does not require any ENV to be set, it is all auto discovered and.-/or modified through the allsettings.json file in the mapped volume or via the Web GUI on port 8099

Thank you. That explains it, and I did have it working after editing allsettings.json.

I'm sure it's on your radar but the Docker Hub main page still describes setup using environmental variables despite now serving the V3 image: https://hub.docker.com/r/britkat/giv_tcp-ma

I'll "close" then.

Thanks again.

closed 2024-09-22 by pjnewcombe