britkat1980/giv_tcp #239: Won't start due to CCTV camera on network

spangingpan, 2024-09-22:
Just moved from givtcp 2.4.9 add-on to v3.
GivTCP detects a CCTV camera as being an inverter and gets stuck. I've tried to specify the correct IP in the settings but this does not seem to work.

CCTV camera: 192.168.1.210
Inverter: 192.168.168.12
HA runs on the 192.168.1.x/24 subnet.

Log message:
2024-09-22 09:20:27,005 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-22 09:20:29,027 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-22 09:20:29,027 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-22 09:20:42,154 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.210:8899
2024-09-22 09:20:42,155 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-22 09:21:03,175 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,176 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,176 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,176 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,176 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,176 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,176 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,176 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,177 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,177 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,177 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:03,177 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:05,428 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.210:8899
2024-09-22 09:21:05,428 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-22 09:21:26,450 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,451 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,452 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:26,452 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-22 09:21:28,717 - Startup - startup     -  [INFO    ] - EVC IP is missing from config. Please update and restart GivTCP
2024-09-22 09:21:28,717 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)
2024-09-22 09:22:28,718 - Startup - startup     -  [ERROR   ] - Error in watchdog loop: ('NameError', 'startup.py', 688)

I've moved back to 2.4.9 for now, on 2.4.9 it at least gives up on the CCTV camera and reads the IP I set in the config:
2024-09-22 09:24:08,835 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-09-22 09:24:14,465 - startup - [ERROR] - Gathering inverter details for 192.168.1.210 failed.
2024-09-22 09:24:14,465 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-09-22 09:24:16,469 - startup - [ERROR] - Gathering inverter details for 192.168.1.210 failed.
2024-09-22 09:24:16,469 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-09-22 09:24:16,470 - startup - [CRITICAL] - Running Redis
2024-09-22 09:24:16,471 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/09/22 09:24:16 [notice] 189#189: using the "epoll" event method
2024/09/22 09:24:16 [notice] 189#189: nginx/1.20.2
2024/09/22 09:24:16 [notice] 189#189: OS: Linux 6.6.46-haos
2024/09/22 09:24:16 [notice] 189#189: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/09/22 09:24:16 [notice] 189#189: start worker processes
2024/09/22 09:24:16 [notice] 189#189: start worker process 193
2024/09/22 09:24:16 [notice] 189#189: start worker process 194
2024-09-22 09:24:16,605 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-09-22 09:24:16,605 - startup - [CRITICAL] - Running Invertor (192.168.168.12) read loop every 5s
2024-09-22 09:24:16,608 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-09-22 09:24:16,610 - startup - [CRITICAL] - Starting Gunicorn on port 6345

britkat1980, 2024-09-24:
Thats an interesting error. I'm wondering if the reason its getting stuck is because of the NameError, which is associated with the EVC IP missing and EVC-Enable being on. If you open up the allsettings.json file and change the evc_enable setting to false and restart, do you get any further? (this is fixed in the upcoming 3.0.1 release

spangingpan, 2024-09-24:
Thanks for the pointer, I edited the allsettings.json file:
Set EVC enabled to false.
Set inverter1 to enable and also added the IP address.
Fixed the mqtt details, I'm using the mqtt addon with a defined user/pass.

It's discovered the inverter and appears to be pulling data correctly, the logs stopped at 10:01 so I assume it is working in the background.
I'll monitor it today and see hot it gets on :)

2024-09-24 10:00:38,295 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-24 10:00:40,309 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-24 10:00:40,310 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-24 10:00:53,420 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.210:8899
2024-09-24 10:00:53,420 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-24 10:01:14,443 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,444 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,444 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,444 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,444 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,444 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,444 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,445 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,445 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,445 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,445 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:14,445 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:16,695 - Startup - client      -  [INFO    ] - Connection established to 192.168.1.210:8899
2024-09-24 10:01:16,695 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-24 10:01:37,718 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,719 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,719 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,719 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,719 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,719 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,719 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,719 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,719 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,720 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,720 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:37,720 - Startup - framer      -  [INFO    ] - No frame header found, await more data
2024-09-24 10:01:39,975 - Startup - startup     -  [INFO    ] - Setting up invertor: 1
2024-09-24 10:01:40,009 - Startup - client      -  [INFO    ] - Connection established to 192.168.168.12:8899
2024-09-24 10:01:40,010 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-24 10:01:41,246 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-24 10:01:47,256 - Startup - startup     -  [INFO    ] - Inverter xxxxxxx which is a Gen2 - Hybrid with 1 batteries has been found at: 192.168.168.12
2024-09-24 10:01:47,259 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-24 10:01:47,259 - Startup - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2024-09-24 10:01:47,259 - Startup - startup     -  [INFO    ] - ====     http://192.168.1.16:8099/config.html             ====
2024-09-24 10:01:47,259 - Startup - startup     -  [INFO    ] - ==============================================================
2024-09-24 10:01:47,260 - Startup - startup     -  [INFO    ] - Running Invertor 1 () read loop every 30/120s
[2024-09-24 10:01:47 +0100] [77] [INFO] Starting gunicorn 23.0.0
[2024-09-24 10:01:47 +0100] [77] [INFO] Listening at: http://0.0.0.0:6345 (77)
[2024-09-24 10:01:47 +0100] [77] [INFO] Using worker: sync
[2024-09-24 10:01:47 +0100] [80] [INFO] Booting worker with pid: 80
[2024-09-24 10:01:47 +0100] [81] [INFO] Booting worker with pid: 81
[2024-09-24 10:01:47 +0100] [82] [INFO] Booting worker with pid: 82
2024-09-24 10:01:48,384 - Inv1 - read        -  [INFO    ] - Starting watch_plant loop...
2024-09-24 10:01:48,388 - Inv1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-09-24 10:01:58,677 - Inv1 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages

britkat1980, 2024-09-24:
OK great.

fyi, you don't need to change the MQTT username/password. v3 addon can auto discover the MQTT connection details which is what you will have seen. Either way will work though

closed 2024-09-24 by britkat1980