britkat1980/giv_tcp #240: V3: Error Processing Inverter Data

joealden, 2024-09-24:
Data from my AIO and Gateway is flowing perfectly in V3, but data from my Hybrid Inverter (for my solar panels) isn't.

Here's the error shown in the add-on logs:

```
Inv3 - read -  [ERROR] - Error processing Inverter data: ((<class 'TypeError'>, TypeError('list indices must be integers or slices, not str'), <traceback object at 0xffffac6bd5c0>), 871)
Inv3 - read -  [ERROR] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0xffffac6a3a40>), 1587)
Inv3 - read -  [ERROR] - runAll2 Error processing registers: ('KeyError', 'read.py', 1766)
Inv3 - read -  [ERROR] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
```

Any ideas? Let me know if you need any more info!

britkat1980, 2024-09-25:
If you have no batteries associated with the Gen3 then this might be the cause and is fixed in 3.0.1. please update and report back.

joealden, 2024-09-25:
Yep 3.0.1 has fixed it - thanks!

closed 2024-09-25 by joealden