britkat1980/giv_tcp #241: GivTCP v3.0.0g-beta Timeout Errors

drt102, 2024-09-25:
I've just redeployed my GivTCP Docker container which pulled down the latest version. I didn't have any issues with v2 but with this new one I am getting "5 consecutive timeout errors in watch loop" errors every 10 minutes or so:

2024-09-25 11:11:42,252 - read - [ERROR] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2024-09-25 11:21:19,601 - read - [ERROR] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2024-09-25 11:31:53,716 - read - [ERROR] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:

I also seem to get an erroneous spikes in the Battery_Charge_Energy_Today_kWh. Current value should be 6.5 but every so often I get a spike to 21.6 which throws out all of the energy stats.

britkat1980, 2024-09-25:
This Beta is now superceded, use Production 3.0.1 and report back if there are still issues

drt102, 2024-09-26:
I have updated and MQTT is reporting v3.0.1 although Docker Hub says the latest version is 3.0.3. I think it is actually running v3.0.3 from the list of images. It is now more stable than with the beta bus is still occasionally reporting 5 consecutive timeouts:

2024-09-26 09:11:06,866 - read - [INFO] - Starting watch_plant loop...
2024-09-26 09:11:06,877 - read - [CRITICAL] - Detecting inverter characteristics...
2024-09-26 09:11:54,487 - read - [INFO] - Publishing Home Assistant Discovery messages
2024-09-26 10:15:16,645 - read - [INFO] - Starting watch_plant loop...
2024-09-26 10:15:16,655 - read - [CRITICAL] - Detecting inverter characteristics...
2024-09-26 11:26:47,925 - read - [ERROR] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2024-09-26 14:17:43,548 - read - [ERROR] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:

SwiftRR, 2024-09-26:
I am also getting 5 timeout errors with modbus restart using v3.0.0 on HAOS on pi:
`2024-09-26 11:26:47,925 - read - [ERROR] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:`

Happens every 3-4 hours.

I have tried production v3.0.1 but this seems more unstable than v3.0.0

willholley, 2024-10-15:
I captured a WireShark trace whilst giv_tcp was reporting timeouts. Curiously, the trace looks okay to my eyes - the inverter (192.168.1.147) is responding but the streamreader in giv_tcp isn't returning anything from the socket from what I can tell.

[givtcp-trace.zip](https://github.com/user-attachments/files/17382534/givtcp-trace.zip)

willholley, 2024-10-16:
I think I found the issue - described at https://github.com/britkat1980/giv_tcp/issues/273#issuecomment-2417247754 and should be a simple fix.

britkat1980, 2024-10-24:
Fixed in 3.0.4

closed 2024-10-24 by britkat1980