britkat1980/giv_tcp #244: GivTCP 3.0.1 not fully starting

creednmd, 2024-09-25:
Have installed version 3.0.1 and Add-on says it has started, but unable to get to the web UI.

Logs say:

2024-09-25 14:18:39,738 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
Traceback (most recent call last):
  File "/app/startup.py", line 315, in <module>
    mask=str(interface['ipv4']['address']).split('/')[1][:-2]
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range

Have no idea what this means, and how to fix it. HA beginner, and unsure what to do now. Did have v2 installed, but not set up, and fully uninstalled (I think) before installing v3.

britkat1980, 2024-09-25:
There is an error in 3.0.1. Roll back until I can fix it

SwiftRR, 2024-09-25:
Same here. I updated about an hour ago and I finished up on v3.0.3.
I couldn't then open the Web UI at all.
I have restored v3.0.0 from backup but still the Web UI refused to open, instead creating more instances of the UI config screen. So I have now lost the Web UI. givTCP is at least running and I presume that the config is fine.

sblee84, 2024-09-26:
I had the same issue on 3.0.1 and continue to have it on 3.0.3 - I also rolled back to HA's backup of 3.0.0. Not sure if it's relevant but my HA uses DHCP with a reservation (though I tried static IP and it didn't help), and my Eero network uses a /22 netmask rather than /24.

britkat1980, 2024-10-24:
Fixed in 3.0.4

closed 2024-10-24 by britkat1980