britkat1980/giv_tcp #247: Set value for "GivTCP Control GivTCP Target SOC" using a parameter

BobB001, 2024-09-26:
Hi folks,
<TL;DR>
Is it possible to pass the value of a variable to GivTCP SAnnnnGnnn Control GivTCP Target SOC?

Background:
I'm using Home Assistant and GivTCP v2.4.9.
My inverter does not always accept a control command the first time it is issued, so I end up surrounding all commands with a while loop, something like:
```
While the setting isn't what you want
  Request the new setting
  Wait a bit
End While
```
This is a bit tedious to copy everywhere, so I'd like to set up a script to do the loop and call it with a parameter of the actual value.
Something like this:
```
alias: Set Givenergy Battery target SOC
sequence:
  - repeat:
      sequence:
        - delay:
            hours: 0
            minutes: 0
            seconds: 30
            milliseconds: 0
        - device_id: xxx
          domain: number
          entity_id: xxx
          type: set_value
          value: {{ target_soc }}
      while:
        - condition: not
          conditions:
            - condition: template
              value_template: >-
                {{states("number.givtcp_sa2237g056_target_soc") | int ==
                target_soc}}
```

...where target_soc is a field passed into the script.
But I can't get this to work. It's as if the Control GivTCP Target SOC device action is checking specifically for a number.

Do any of you brainboxes know how I might do this? Or is it a lost cause? My second option is to write several scripts each with a hard-coded value, so have a Set_SOC_50, Set_SOC_60, Set_SOC_80 etc.

Thanks in advance.

gcoan, 2024-09-26:
why does your inverter not accept commands first time, what's in the givtcp logfile?

This suggests an underlying network problem in the first instance if commands are timing out or not being completed.  I see occasional timeouts (and I think givtcp v3 is better at auto retrying), but they are occasional, most commands work first time.

Maybe you need to improve the wifi connection strength to your inverter, eg use a wifi repeater?

BobB001, 2024-09-26:
Thanks Geoffrey, You're right. I'm the proud owner of _ye olde worlde_ Gen 1 inverter with a Wi-Fi dongle. It sits in the garage and while the signal is reasonable it's not perfect. Even the GivEnergy app fails to command it sometimes. I do have Wi-Fi repeaters around the house already, and it doesn't feel necessary to improve the signal. It's "good enough" (quite often the command is accepted, occasionally I need to send it twice, I have seen a repeat of 6 times before acceptance.)

To answer your question, I see in the log a few lines similar to:
2024-09-25 20:08:11,828 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.xx.xx, 8899) failed: timed out

Your reply is interesting because it hints at GivTCP v3 auto retrying... Of course, if that was the case my problem goes away because I don't need to do the loop logic. I've seen my loops trigger in v2, so if there is retry logic in v2, it doesn't always work in my case.

Maybe I need to take a brave pill and update to v3...

gcoan, 2024-09-26:
I have two Gen 1 hybrid inverters, with wifi dongles.

Originally my inverters (in the garage) were configured to talk to an existing powerline wifi repeater in the lounge, so only about 5m away through one wall as they couldn't see my home wifi router at all.

I was probably as you are, it worked, but not always consistently.  If you have existing wifi repeaters it might be worth you going through the inverter wifi setup process (instructions on the kb.givenergy site or there is a good givenergy wifi setup video on youtube) and checking that your inverter is connected to the correct strongest strength wifi node.

I eventually gave up with my powerline repeaters, they were just too unreliable and bought a Mercusys AX3000 mesh network with 3 nodes, and then I added a further 3 including one in the garage where the inverters are.  I've been very pleased by it, very solid wifi all round the house and looking just now there was only 1 error today on the givtcp log.

As for GivTCP v3, I have at the moment reverted back to 2.4.9 as v3 didn't report my battery information.  There is I believe a fix for this but I've decided to hold off for another week.
I think I saw auto retries occurring in the v3 logfile but can't be 100% sure as I'm not running it at the moment

BobB001, 2024-10-01:
I'm not convinced this is answerable. Geoffrey is right that my network isn't perfect, but the question of parameterisation of a command does not appear to be solvable. I'll stick with hard coding...
Marking as closed to keep the issue tally clean.
Thanks all.

closed 2024-10-01 by BobB001