britkat1980/giv_tcp #248: Lost access to battery with 3.0.3?

mprowe, 2024-09-27:
Following the bummpy transfer from 2.4.9 through 3.0.0 > 3.0.1 > 3.0.2 and now 3.0.3, I now see that I now no longer am able to "see" my battery? Not sure how to pass the information in text-form, so it will have to be screenshot (sorry).
[image]
Do I go back to 2.4.9? Or leave it alone until you advise on a solution?
Regards, Martin

gcoan, 2024-09-27:
I had this happen on v3.0.0.  There is a fix @britkat1980 advised on deleting entities from MQTT and I believe it was going to be fixed in v3.0.1 however I have at present reverted back to v2.4.9 as that was working and stable for me.

It *might* work correctly if you downgrade to 2.4.9 and then upgrade directly to the latest patch to v3,  going via the interim releases may be why the fixes introduced in 3.0.1 didn't work for you.  That's a suggestion rather than evidence based though !!

mprowe, 2024-09-28:
> however I have at present reverted back to v2.4.9 as that was working and stable for me.

Okay. I'm going to do the same. I'm afraid my skills are not up to poking around in the MQTT Server!

britkat1980, 2024-09-28:
> Following the bummpy transfer from 2.4.9 through 3.0.0 > 3.0.1 > 3.0.2 and now 3.0.3, I now see that I now no longer am able to "see" my battery? Not sure how to pass the information in text-form, [...]

The entity friendly names look odd, can you confirm via the MQTT integration pages that there aren't any other battery devices. You can delete them from that page, to save mucking around with an MQTT broker.

closed 2024-10-24 by britkat1980