britkat1980/giv_tcp #250: Automation not able to activate "Enable Charge Control"?

mprowe, 2024-09-29:
I know everone says this sort of thing... But I know that this used to work befor the aborted upgrade to 3.0.3.
This is my automation, in full:

```
alias: __1st-Turn_On_Charger
description: Wait until soc <80% & Off Peak is On
trigger:
  - type: battery_level
    platform: device
    device_id: 28c27bfc66e17ae87f3b72ca9f5b0413
    entity_id: 8cd465c35b6da647f27bb1704aef7e49
    domain: sensor
    below: 55
condition:
  - condition: state
    entity_id: binary_sensor.octopus_energy_electricity_21e5233878_2200021723572_off_peak
    state: "on"
    enabled: false
action:
  - action: switch.turn_on
    metadata: {}
    data: {}
    target:
      device_id: 061f35b7ae7f52f0cfd86f6efcf95161
      entity_id: switch.givtcp_sa2219g385_enable_charge_schedule
mode: single

```
The trigger (SOC < x%) and the condition (off-peak=on) work fine. The problem is that the **action** does not turn on the Charge Enable.
I can operate the switch perfectly well from the dashboard. Aditionaly, the Developer Tools | actions does not work either. Which is not supprising as I suspect that code via the two routes is the same?

Is there any way to start with a clean sheet - either 2.4.9 or 3.0.3?

Regards, M.

gcoan, 2024-09-29:
Try removing the device id, it's not necessary in the automation and it could have changed with the new givtcp

mprowe, 2024-09-30:
Thanks Geoffrey,
Good call. Yes, removing the device_is alowed me to manualy run the Action. But... Now the Trigger does not work!!
Just in case I'm doing more harm than good by all my attempts to correct, can I reask my question. Is it possible to start again?
What happens if I uninstall both 2.4.9 & 3.0.3. Then delete the MQTT tree (553 topics) with MQTExporer .
Then install 3.0.3 and start again?

Regards, M.

gcoan, 2024-09-30:
> Thanks Geoffrey,
>
> Good call. Yes, removing the device_is alowed me to manualy run the Action. But... Now the Trigger does not work!!
>
> Just in case I'm doing more harm than good by all my attempts to [...]

Your problem with the trigger is exactly the same as the problem with your action, you are trying to "do something" with a coded device id that has almost certainly changed with the new version of GivTCP.

I've only been using Home Assistant for just over a year and so I don't know what the guidance for automations used to be, but for me, I have **never** used device id in **any** of my automations.  Its both unreadable and as you're finding out, not portable across versions.

My automations look like this:

```
  - platform: numeric_state
    entity_id: sensor.givtcp_sd2237g395_invertor_temperature
    for:
      minutes: 15
    below: 10
```

I'm triggering on the numeric value of a specific sensor, temperature in this case.  Nothing to do with device's or coded entity id's at all.  GivTCP v3 retains the same naming convention as v2 so it should all just work the same.

With regards to your other question, no you shouldn't need to uninstall v2, v3 and delete all the MQTT topics.  Just shutting down v2 and starting v3 is all that I did to swap from one to the other.  I went backwards and forwards a few times between the two versions.   v3.0.0 had a problem with the battery sensor names so I am at present back running 2.4.9, but I believe this is fixed now in v3.0.1, I just haven't tested it again yet

britkat1980, 2024-09-30:
> Thanks Geoffrey, Good call. Yes, removing the device_is alowed me to manualy run the Action. But... Now the Trigger does not work!! Just in case I'm doing more harm than good by all my attempts to [...]

For a clean slate:
- Stop all GivTCP containers
- Delete all files in the config/GivTCP folder
- Delete all related topics in MQTT broker
- Start 3.0.3 and update config

closed 2024-10-01 by britkat1980