britkat1980/giv_tcp #251: unable to add on GivTCP repository

gc01dr, 2024-09-29:
Hi 

I'm trying to install GIvTCP add on. I've cut and pasted the address "https://github.com/britkat1980/giv_tcp" to Add repository on HA  but I keep getting this error message.  Any idea about how to resolve this please. 

 Cmd('git') failed due to: exit code(128) cmdline: git clone -v --recursive --depth=1 --shallow-submodules -- https://github.com/britkat1980/giv_tcp /data/addons/git/a6a2857d stderr: 'Cloning into '/data/addons/git/a6a2857d'... POST git-upload-pack (175 bytes) POST git-upload-pack (244 bytes) Submodule 'WebDashboard' (https://github.com/DanielGallo/GivEnergy-Smart-Home-Display.git) registered for path 'WebDashboard' Cloning into '/data/addons/git/a6a2857d/WebDashboard'... fatal: remote error: upload-pack: not our ref 4dd6de564b77b0e29430446387185f0b548961d9 fatal: Fetched in submodule path 'WebDashboard', but it did not contain 4dd6de564b77b0e29430446387185f0b548961d9. Direct fetching of that commit failed. '
 onto the Add Respos

gcoan, 2024-09-29:
> I'm trying to install GIvTCP add on. I've cut and pasted the address "https://github.com/britkat1980/giv_tcp" to Add repository on HA but I keep getting this error message. Any idea about how to resolve this please.

Try using https://github.com/britkat1980/ha-addons which is the correct directory for Givtcp v3

gc01dr, 2024-09-29:
Thank you. That worked! Regards Geoff

closed 2024-09-29 by gc01dr