britkat1980/giv_tcp #252: Energy Usage values not resetting before new day

lewiswatt, 2024-09-29:
Hi all,

Since v3.0.3 on the start of a new day my energy values for all GivTCP values don't seem to clear at midnight, resulting in the cumulative of all import/export/generation being factored into the following day's results.

[image]

On the surface, I'm not sure what would cause this issue, as the values in question seemingly reset within 30 seconds of midnight passing. The entity behaviour before and after the update has not changed.

Apologies if this an unrelated config error on my part, as GivTCP seems to be the only thing I've updated in the past week or so when this started happening.

Alibloke, 2024-09-29:
This has also started happening to me after upgrading to v3.

gcoan, 2024-09-29:
There has been a discussion about spikes and dips in the today energy usage values in the energy dashboard, not specifically v3 but is maybe more often happening with v3 https://github.com/britkat1980/ha-addons/issues/20

One solution is to swap to using the total sensors rather than the today sensors in the energy dashboard

lewiswatt, 2024-09-29:
I've looked at my history for all the relevant sensors, and I do see that behaviour on the "Charge Today", with all spikes climbing to exactly 21.6kWh. "Discharge Today" only seems to have had two erroneous spikes a few weeks ago.

I'm not sure if the behaviour in that thread is exactly the problem I'm seeing here, as why would those spikes cause the following day's data to be carried forward?

britkat1980, 2024-09-30:
I see this same happening if I use Today stats for the Energy Dashboard. I actually use Total figures, which is far more reliable.
Some of this is a function of read-loop timings and synchronisation of Inverter and HA times.

I suggest moving to Total energy stats and HA will work out the midnight reset

gcoan, 2024-09-30:
The energy dashboard runs off the long term statistics in home assistant so as long as you haven't disabled the total sensors (so they have no history) you should be able to swap over and see no impact

e10kstarfire, 2024-09-30:
> If the total energy stats are more reliable, then I'll happily move over to them. Is there any way to preserve the previous history though?

I moved to total energy and it preserved the stats, I had to also change battery entity from today to total as it caused weird issues without.

cbrinks17, 2025-01-02:
> > If the total energy stats are more reliable, then I'll happily move over to them. Is there any way to preserve the previous history though?
> 
> I moved to total energy and it preserved the stats, I had to also change battery entity from today to total as it caused weird issues without.

What entity did you use for the battery charge/discharge totals? I do not seem to have any entities carried over from v2.4.9 that store the totals for the battery. In v3, it created a new MQTT entity called "{Friendly Name} Battery Stack 1". This seems to work fine for the data created since I upgraded to v3, but it didn't exist in v2.4.9, so last year's data is now all wrong.

gcoan, 2025-01-02:
I use sensor.givtcp_xxxx_battery_discharge_energy_total_kwh and similar charge sensor.

They all existed on GivTCP 2 and 3 and I didn't have any issues with the energy dashboard when I upgraded.

I only have battery_stack sensors for voltage and temperature, nothing for charge/discharge rate or SoC.

cbrinks17, 2025-01-03:
> I use sensor.givtcp_xxxx_battery_discharge_energy_total_kwh and similar charge sensor.
>
> They all existed on GivTCP 2 and 3 and I didn't have any issues with the energy dashboard when I upgraded.
>
> I [...]

I have an AIO, and I have inverter 1 set up as the battery and Inverter 2 set up as the gateway. I have toggled the "only report battery data" for the gateway inverter set. I do not seem to have the entity you describe for the charge total on my battery. It seems to be there for the gateway, but I don't think it is provided anymore due to that toggle. Should that be on for the gateway or the battery?

gcoan, 2025-01-03:
If you have only a single AIO then the gateway doesn't really perform any function, only in multi AIO setups does the gateway control the individual AIO's.

In terms of GivTCP data I'd suggest you turn the toggle back on for the AIO and use the AIO battery total info not the gateway, this sounds like why you are not seeing the same sensor data that I am.

Could possibly raise the sensor consistency as a GivTCP question, whether the battery total's could be provided for the gateway in single AIO setups

cbrinks17, 2025-01-03:
I tried turning that on/off for the battery, and it seems to have no effect. I still don't see those battery statistics. Annoyingly the link in the git hub page to the settings returns a 404. I don't really know what that setting does.

The values that I do have on the Stack 1 device don't line up with what I see on the GiveEnergy cloud for total charge/discharge either. They seem to be equal on the GE dashboard, but the charge total is much higher on the Stack 1 device in MQTT.

[image]

[image]

[image]

iainfogg, 2025-04-27:
For some reason I cannot see the Total entities in the dropdowns in the various categories on the Energy dashboard.

Various of you have moved to using them, I'm not sure how though!

Has anyone else struggled to get the Total entities to show?

They definitely exist as I have them in my Settings, Entities section as well as my MQTT Explorer view, and the entities are being updated.

gcoan, 2025-04-27:
> Has anyone else struggled to get the Total entities to show? The only Total entity I see in the energy dashboard dropdowns is battery throughput.

Check developer tools/statistics to see if there is any issues with the statistics being collected for the total entities.  That's usually the problem that there has been an issue and stats are not being collected

Maybe the unit of measurement isn't set to something that the Energy dashboard recognises.

iainfogg, 2025-04-28:
Thanks @gcoan that was really helpful, I didn't know Dev Tools / Statistics was a thing.

Turns out I had a wildcard exclude in the recorder config to avoid storing tonnes of stats I don't care about for some energy monitoring plugs, and unfortunately that wildcard also coincided with the GivTCP 'energy_total' entities :(

So I guess I'm back to v2 until the stats thing gets worked out.

gcoan, 2025-04-28:
Every day a school day @iainfogg !

I have a potential way out for you to load the missing statistics from the givenergy portal

iainfogg, 2025-04-28:
> I have a potential way out for you to load the missing statistics from the givenergy portal 

@gcoan I'm definitely interested in that!

gcoan, 2025-04-28:
> > I have a potential way out for you to load the missing statistics from the givenergy portal
> 
> [@gcoan](https://github.com/gcoan) I'm definitely interested in that!

@iainfogg Do you have an account on the givenergy beta customer forum?
Here's an invite link if you don't https://community-beta.givenergy.cloud/invites/MCPtJv4fuq

I can PM you on there my current draft of the procedures

steve-daedilus, 2025-05-18:
I switched from GivTCP 2 to 3 a couple of days ago, and since then I have seen this same issue.

I tried to switch to totals, however when I did so I couldn't get pricing to operate any more.

Using daily stats I could get pricing but was seeing costs as negative for import and positive for export.

To correct the state on daily I've had to go through multiple sensor values and zero them out at the midnight switch.

I'm also using the Octopus data feed. I'm sure I'm missing something silly here, but at this point I've no idea what.

gcoan, 2025-05-18:
> I tried to switch to totals, however when I did so I couldn't get pricing to operate any more.
> 
> Using daily stats I could get pricing but was seeing costs as negative for import and positive for export.

I'm exclusively using totals on the energy dashboard and it all works fine for me.

I use givtcp_xxx_import_energy_total_kwh for the import and the octopus integration sensor 'current rate electricity' for the "use an entity with the current price"

Likewise givtcp_xxx_export_energy_total_kwh and 'current rate export electricity' for the return to grid.

When you add a new sensor to the energy dashboard like this it doesn't create all the history data, so you have to back-load the history,  or just use the energy dashboard going forward

steve-daedilus, 2025-05-19:
> When you add a new sensor to the energy dashboard like this it doesn't create all the history data, so you have to back-load the history, or just use the energy dashboard going forward

How would I go about back-loading the history? Is there a way to do this easily, or is it a manual exercise?

gcoan, 2025-05-19:
> 
> > When you add a new sensor to the energy dashboard like this it doesn't create all the history data, so you have to back-load the history, or just use the energy dashboard going forward
> 
> How would I go about back-loading the history? Is there a way to do this easily, or is it a manual exercise?

It's not totally straight forward as there isn't a standard HA API for this.

I have however developed a set of procedures to load history from various sources including the GivEnergy portal and the Octopus data download. This is what I used to back-load the history when I changed from today to total sensors myself.

There are a few people beta-testing the process before I share it more widely.  If you want to be included then join the GivEnergy beta community forum https://community-beta.givenergy.cloud/invites/MCPtJv4fuq and I'll add you

steve-daedilus, 2025-05-19:
> There are a few people beta-testing the process before I share it more widely. If you want to be included then join the GivEnergy beta community forum https://community-beta.givenergy.cloud/invites/MCPtJv4fuq and I'll add you

I've done this now, thanks!

closed 2025-10-23 by britkat1980