britkat1980/giv_tcp #254: TimeZone not being used properly

BrianUK6, 2024-09-30:
With the v3 the time in the Docker Container was 1 hour behind (UK).   
The allsettings.json has field "TZ"  defaulted to Europe/London.

When I used docker exec bash and manually set the timezone it immediately corrected to the UK datetime.

New code is not using the allsettings.   Several places do timezone stuff and difficult to follow the logic, but GivLUT.py looks for "timezone" item not "TZ" item.

Looks like "IsAddOn" changes have broken things for NOT "IsAddOn".

I'm NOT using HA.

britkat1980, 2024-10-01:
Thanks for the feedback. I've tweaked the code and hopefully the hierarchy of Timezone is as follows:

1. If isAddon use supervisor reported timezone
2. If notAddon use the "TZ" container ENV
3. If all else fails use "Europe\London" as the timezone

So to change Timezone change the ENV at runtime (if using docker direct)

This will come from Dev 3.0.21 and above and in Prod 3.0.4 and above once ready to push

britkat1980, 2024-10-24:
Fixed in 3.0.4

closed 2024-10-24 by britkat1980