britkat1980/giv_tcp #255: setChargeRate and setDischargeRate

BrianUK6, 2024-09-30:
**MQTT Control setChargeRate and setDischargeRate**
In v2 these were Watts and the inverter battery charge/discharge levels could be directly controlled, and as Inverter level it applies to all attached batteries.
These map to 72 & 73 via the Giv API and control the inverter charging level.
GivTCP v2 did the same.  So everything was in agreement.

v3 docs (https://github.com/britkat1980/giv_tcp/blob/main/SETTINGS-GUIDE.md) and code now say percentage, but that is wrong.  They should remain as Watts and the v2 code restored.
v3 converts % to watts by looking at a battery max power, but how does it know what battery to use it a mixed battery setup.
Let alone the inverter has the max ch/disch power control - not the battery.  Logic doesn't work.

If you want to add 2 new fields to pass in a %, then add them as new - plus you'll need to identify which battery you are referring to.

For ref Palm.py directly uses reg 72/73 to set 0/3000 (AC3)     [completely out of date for additional equipment]
Plus it sets % via  
id": 77,"name": "AC Charge Upper % Limit",

britkat1980, 2024-10-01:
v3 Uses the same charge rate controls and logic using watts as v2. there is no change there. There is an additional control which has been added which uses the AC Rate Limit registers which provides a complementary 0-100% control. This has greater granularity than the traditional derived watts control. you can choose which you wish to use.

If you think this is not the case can you provide more details on the system you are using and what the errors you are seeing.

BrianUK6, 2024-10-03:
ok - I'll do code compares etc.  My code didn't change, but on switching over to v3 I received errors about setting the Charge/Discharge rate.

I have a verification loop that is triggered by my Zappi starting to charge meaning and I have cheap IOG. I then change the battery charge/dischard accordingly.  I wait for the MQTT to change to give verification and also do a read via the GE API to get the charge/discharge rate (it does not cache that).  When all agree then I can guarantee everything is what it should be and then switch other large house items on/off as needed.    If that is not done the changes that don't work are missed.  Costs money to miss them :-) :-) 

The foolproof checking immediately did not work with v3 - it was giving mismatches.   Tracked down to the using 0/3000 (which has worked for eons) and finding it needs to be a %.   

I'll get to the bottom of it and report back.

BrianUK6, 2024-10-08:
Looks like my sys was using older code that I had cleaned up and at the same time made the Charge/Disharge  agree with the docs at that time.  Have found various versions of old givtcp that do things differently.

Anyways, I'll bring my code up to GivTCP v3.

Also, looking through all the G.3 code (for this purpose) highlighted how there are so many different ways to acheive the same thing and get from A to B...  must be nightmare to verify them all....

APoller1, 2024-10-31:
I have a similar issue.
I can change the charge Rate, but it reverts in about 2 seconds to 2600. Wherever I change it, it is changed back to 2600. HA, Giv App or portal. Any suggestions?

APoller1, 2024-10-31:
It worked in v2.

polsign, 2025-06-23:
I have an AIO and notice that setting the Charge Rate in Watts translates to setting Charge Rate AC (Inverter %). On the next read the Charge Rate Watts reverts to whatever value was set on the portal previously. eg if 2000W and you send 3000W you end up with 50% Inverter setting and 2000W Charge Rate which = 1000W. Discharge appears to be similar behaviour except when starting from 6000W/100% if you drop to 3000W you get 50% instead and the next read you revert to 6000W which then changes the 50% back to 100%. Is there any fix in the pipeline please?

APoller1, 2025-06-24:
GivEnergy think it's related to my installed EVC which changes the battery charge rate, even when on grid only.
Only way I've found to fix it is to set the EVC to standalone, which is a bit pointless as it then discharges the battery.

polsign, 2025-06-24:
Can you see any entries in the portal log that indicate where the change is coming from? In my scenarios I can see that the rates in watts are not changed by givTCP, only the % registers are modified. Changing the watts from the remote control section of the portal  or the GE app works for me

polsign, 2025-06-24:
with a web browser go to givenergy.cloud/dashboard and then from the left hand menu choose My Inverter(s). If you have multiple entries then you can chose remote control (cog icon) from the list, if only 1 inverter then the cog should be under actions at top right. might be off a 3 dot menu.  From there you will see all the underlying registers and at the bottom you can get the logs of activity. For each register you need to use the refresh icon next to the name to make sure it shows any external changes from apps or GivTCP/HA

APoller1, 2025-06-25:
Thanks for the instructions it is the "EV Charger Inverter Control" that resets the battery charge rate...

polsign, 2025-06-25:
I dont have an EV so not played with that. I have gone back to v2 for now until this is resolved as its just not usable for my need to balance export and charge :(