britkat1980/giv_tcp #256: EVC causes startup failure

ChrisMontford, 2024-09-30:
I had v3.0.3 upgraded and running without the EVC setup.

I changed the setup to enable EVC and added in the ip address of the EVC. Now GiveTCP won't start and gives this error on start up:

2024-09-30 23:27:26,151 - Startup - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2024-09-30 23:27:28,212 - Startup - startup     -  [INFO    ] - Searching for Inverters
2024-09-30 23:27:28,213 - Startup - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-09-30 23:27:32,486 - Startup - startup     -  [INFO    ] - GivEVC *************** found at: ***.***.***.***
2024-09-30 23:27:44,539 - Startup - client      -  [INFO    ] - Connection established to ***.***.***.***:8899
2024-09-30 23:27:44,547 - Startup - client      -  [INFO    ] - Detecting plant
2024-09-30 23:27:45,709 - Startup - client      -  [INFO    ] - Plant Detected
2024-09-30 23:27:48,472 - Startup - startup     -  [INFO    ] - Inverter FD2325G254 which is a Gen3 - Hybrid with 1 batteries has been found at: ***.***.***.***
Traceback (most recent call last):
  File "/app/startup.py", line 446, in <module>
    setts["serial_number_evc"]=evcList[1][1]
                               ~~~~~~~^^^
KeyError: 1

peter1rhodes, 2024-10-01:
I had this issue when I first switched to v3. As I recall, restarting the add-on, or possibly Home Assistant, fixed it.

britkat1980, 2024-10-24:
Fixed in 3.0.4

closed 2024-10-24 by britkat1980