britkat1980/giv_tcp #258: [ERROR   ] - Timed out waiting for regcache

BrianUK6, 2024-10-01:
V3.0.3
Lots of:
 Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache

That's a 5 sec wait loop.  Had a look in Docker and found /GivTCP_1/.regcache_lockfile_1  has been in existing for over 2 hours.

Code has Try/Except, so there's something else going on.

Manually deleted file.

britkat1980, 2024-10-01:
changes made in current dev version to try and remove stale lockfiles. hoping a fix is available in 3.0.4. If you want to test early try dev3.0.20+

BrianUK6, 2024-10-02:
Now I know what causes it it's easy to manage until in mainstream.  Thanks.

although a belts-n-braces startup clearout is a good idea as that deals with any crashes/bugs etc...    lets the code start with a clean plate each time.

BrianUK6, 2024-10-03:
I gave the download zip to AI and told it to look for regcache and filelocking and potential errors.   It created new code.
This is your proj, so just pasted it's comments below.
If it assists, great. If not then to the bin :-) 

======================

From ChatGPT:
I have updated the GivLUT.py file to include robust lockfile handling using the FileLock library. Key changes include:

Importing FileLock:
Added: from filelock import FileLock at the top of the file.

Lockfile Implementation:
Replaced the lockfile definition with a FileLock object:
python
Copy code
cachelockfile = ".regcache_lockfile_" + str(GiV_Settings.givtcp_instance)
cachelock = FileLock(cachelockfile)
File Access with Locking:

Wrapped all open() operations for reading/writing regcache inside a context that uses cachelock, ensuring that access to the cache is properly synchronized:
python
Copy code
with cachelock, open(GivLUT.regcache, 'rb') as inp:
    # Reading from regcache

with cachelock, open(GivLUT.regcache, 'wb') as outp:
    # Writing to regcache
This update will help ensure that the regcache lockfile is properly managed, preventing issues like leaving the lockfile in an unintended state.

Would you like me to apply similar changes to other parts of the project where regcache is used, or is this sufficient for now?

willholley, 2024-10-08:
My GivTCP addon got into this state as well today. The regcache file was last updated at 11:54. It looks like there were lots of write errors manifesting around that time and I can only recover by deleting the regcache and restarting GivTCP.

If useful, the write log is at https://gist.github.com/willholley/5ef7f11b1c802658e094cc1afcfa5e80

willholley, 2024-10-08:
a sample from the combined log:

```
2024-10-08 17:19:48,548 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/CE2249G756/Stats
2024-10-08 17:19:48,582 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/CE2249G756/Timeslots
2024-10-08 17:19:48,621 - Inv1 - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/CE2249G756/raw
2024-10-08 17:19:48,634 - Inv1 - mqtt        -  [DEBUG   ] - Prepping DX2408G039 for publishing
2024-10-08 17:19:48,639 - Inv1 - mqtt        -  [DEBUG   ] - Prepping batteries for publishing
2024-10-08 17:19:48,647 - Inv1 - mqtt        -  [DEBUG   ] - Prepping invertor for publishing
2024-10-08 17:19:49,607 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:19:49,638 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
2024-10-08 17:19:54,654 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
2024-10-08 17:19:54,656 - Inv1 - read        -  [ERROR   ] - Error in Watch Loop: ('TypeError', 'read.py', 123)
2024-10-08 17:19:54,674 - Inv1 - read        -  [DEBUG   ] - Opening Modbus Connecion to: 192.168.1.147
2024-10-08 17:19:54,689 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:19:54,704 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
2024-10-08 17:19:59,719 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
2024-10-08 17:19:59,721 - Inv1 - read        -  [ERROR   ] - Error in Watch Loop: ('TypeError', 'read.py', 123)
2024-10-08 17:19:59,739 - Inv1 - read        -  [DEBUG   ] - Opening Modbus Connecion to: 192.168.1.147
2024-10-08 17:19:59,757 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:19:59,766 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
2024-10-08 17:20:04,781 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
2024-10-08 17:20:04,784 - Inv1 - read        -  [ERROR   ] - Error in Watch Loop: ('TypeError', 'read.py', 123)
2024-10-08 17:20:04,790 - Inv1 - read        -  [DEBUG   ] - Opening Modbus Connecion to: 192.168.1.147
2024-10-08 17:20:04,804 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:20:04,813 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
2024-10-08 17:20:09,831 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
2024-10-08 17:20:09,834 - Inv1 - read        -  [ERROR   ] - Error in Watch Loop: ('TypeError', 'read.py', 123)
2024-10-08 17:20:09,840 - Inv1 - read        -  [DEBUG   ] - Opening Modbus Connecion to: 192.168.1.147
2024-10-08 17:20:09,854 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:20:09,863 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
2024-10-08 17:20:14,882 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
2024-10-08 17:20:14,884 - Inv1 - read        -  [ERROR   ] - Error in Watch Loop: ('TypeError', 'read.py', 123)
2024-10-08 17:20:14,901 - Inv1 - read        -  [DEBUG   ] - Opening Modbus Connecion to: 192.168.1.147
2024-10-08 17:20:14,928 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:20:14,942 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
2024-10-08 17:20:19,960 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
2024-10-08 17:20:19,963 - Inv1 - read        -  [ERROR   ] - Error in Watch Loop: ('TypeError', 'read.py', 123)
2024-10-08 17:20:19,973 - Inv1 - read        -  [DEBUG   ] - Opening Modbus Connecion to: 192.168.1.147
2024-10-08 17:20:19,993 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:20:20,002 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
2024-10-08 17:20:25,020 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
2024-10-08 17:20:25,023 - Inv1 - read        -  [ERROR   ] - Error in Watch Loop: ('TypeError', 'read.py', 123)
2024-10-08 17:20:25,031 - Inv1 - read        -  [DEBUG   ] - Opening Modbus Connecion to: 192.168.1.147
2024-10-08 17:20:25,046 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:20:25,057 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
2024-10-08 17:20:30,073 - Inv1 - GivLUT      -  [ERROR   ] - Timed out waiting for regcache
2024-10-08 17:20:30,079 - Inv1 - read        -  [ERROR   ] - Error in Watch Loop: ('TypeError', 'read.py', 123)
2024-10-08 17:20:30,095 - Inv1 - read        -  [DEBUG   ] - Opening Modbus Connecion to: 192.168.1.147
2024-10-08 17:20:30,112 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
2024-10-08 17:20:30,121 - Inv1 - read        -  [DEBUG   ] - Command: setChargeRate was sent: {'chargeRate': '0'}
```

britkat1980, 2024-10-08:
I’ve changed lockfile handling in latest dev version, which seems to fix this. Could you run that version and see if it stops this happening?

willholley, 2024-10-08:
Thanks @britkat1980 . I'm running that now, but seeing a lot of:

```
2024-10-08 17:46:37,311 - Inv1 - mqtt        -  [DEBUG   ] - Prepping DX2408G039 for publishing
2024-10-08 17:46:37,318 - Inv1 - mqtt        -  [DEBUG   ] - Prepping batteries for publishing
2024-10-08 17:46:37,342 - Inv1 - mqtt        -  [DEBUG   ] - Prepping invertor for publishing
2024-10-08 17:47:29,589 - Inv1 - REST        -  [DEBUG   ] - Response in file is: {'id': 'setDischargeSlot', 'result': '{"result": "Setting Discharge Slot 1 failed: (\'Exception\', \'write.py\', 719)"}'}
2024-10-08 17:47:29,593 - Inv1 - REST        -  [DEBUG   ] - Response in file is: {'id': 'setBatteryReserve', 'result': '{"result": "Setting Battery Reserve failed: (\'Exception\', \'write.py\', 367)"}'}
2024-10-08 17:47:29,596 - Inv1 - REST        -  [DEBUG   ] - found REST response
2024-10-08 17:47:29,896 - Inv1 - REST        -  [DEBUG   ] - Response in file is: {'id': 'setDischargeSlot', 'result': '{"result": "Setting Discharge Slot 1 failed: (\'Exception\', \'write.py\', 719)"}'}
2024-10-08 17:47:30,003 - Inv1 - REST        -  [DEBUG   ] - Response in file is: {'id': 'setDischargeSlot', 'result': '{"result": "Setting Discharge Slot 1 failed: (\'Exception\', \'write.py\', 719)"}'}
2024-10-08 17:47:30,110 - Inv1 - REST        -  [DEBUG   ] - Response in file is: {'id': 'setDischargeSlot', 'result': '{"result": "Setting Discharge Slot 1 failed: (\'Exception\', \'write.py\', 719)"}'}
2024-10-08 17:47:30,217 - Inv1 - REST        -  [DEBUG   ] - Response in file is: {'id': 'setDischargeSlot', 'result': '{"result": "Setting Discharge Slot 1 failed: (\'Exception\', \'write.py\', 719)"}'}
2024-10-08 17:47:30,285 - Inv1 - read        -  [DEBUG   ] - Write Request recieved
```

perhaps there's some more state to reset?

willholley, 2024-10-11:
The dev build has definitely fixed the regcache error. I'll investigate the timeout errors in a different issue (I have a dev build working locally now so will try and debug it).

sparks1372, 2024-10-17:
is it clear which version the issue was introduced in?