britkat1980/giv_tcp #260: givTCP v3.0.0 Weird log reading: [CRITICAL] - Function code 86 received. Gracefully handled

SwiftRR, 2024-10-02:
givTCP v3.0.0 running on HAOS (on pi4) using HAOS Core: 2024.9.3.
I run Predbat v8.4.9.
I have a GE AC3.0 inverter with updated firmware which can run Pause controls.

I am receiving the log line below, about every half hour. The CRITICAL line is always followed by Pause mode being set which can be 'disabled'. 'PauseCharge' or 'PauseDischarge'. 
[image]

I am also receiving regular time-out errors:
[image]

.... and self run errors

[image]

givTCP seems to recover from each of these glitches, as you can see from the log extracts above

Any idea what is causing these? I am ignoring but something isn't right. Are there bugs?

I have tried running v3.0.1 and v3.0.3 but these are unstable on my setup.

Rob

PhillWalker, 2024-10-20:
I am seeing this error regularly too

britkat1980, 2024-10-24:
The critical message is not an error and can be ignored. It’s tracking a fw bug in the inverters which GE are aware of

closed 2025-10-23 by britkat1980