britkat1980/giv_tcp #268: Reporting incorrect battery size

dieswarte, 2024-10-05:
I have a Gen 2 HY 5.0 inverter with 2 x 9.5 batteries. On v3 it reverts back to reporting the battery size as only being 13.16 for some reason and then it messes up the calculations within Predbat. I have check the GivEnergy portal and there it still shows the capacity correctly as 19.05 kWh. First drop shown on the graph is just after the upgrade to v3. Then when I restarted GivTCP it would read it correctly at 19.05 for a short while before dropping down to 13.16 again - last increase on the graph is after I downgraded back to v2.

[image]

britkat1980, 2024-10-24:
Fixed in 3.0.4

closed 2024-10-24 by britkat1980

IWR69, 2025-01-21:
I still have this issue with 3.0.4 but it seems to affect lots of other sensors too which in turn causes havoc with my sensors.
When the Portal decides to show me the Remote Control History, rather than always timing out, these spikes do not appear.

[image]

IWR69, 2025-01-23:
At the same time these spikes occur HA Logbook records GivTCP retrieving some invalid, possibly default, values. Here's some examples.

For example when Power SOC kWh incorrectly changes to 13.16, I have 9.52, Control Mode changes to Unknown (should be Eco) and Battery Type changes to Lead_acid (should be Lithium). When Power SOC kWh incorrect changed to 0 Control Mode changed to Eco (Paused).

For Battery Capacity kWh I noticed Control Mode change to Unknown, Battery Type to Lead Acid and Meter Type to Ct_or_em418 (should be Em115).

For Inverter Max Bat Rate I noticed Meter Type change to Ct_or_em418, Inverter Type to Gen1 Hybrid (I have Gen3 Hybrid) and Inverter Firmware to D0.0-A0.0 (I have D0.311-A0.311)