britkat1980/giv_tcp #270: Calibration state can cause GivTCP V3 to fail to start

kenbell, 2024-10-06:
I had a spurious calibration occur on my battery (cause unknown).  When it did so, the soc_force_adjust register ended up stuck, with the value '6'.

This breaks GivTCP which only anticipates 4 values: (https://github.com/britkat1980/giv_tcp/blob/main/GivTCP/GivLUT.py#L726)
```['Off','Start','','Charge Only']```

It looks like the full set of values is 8: (https://github.com/britkat1980/giv_tcp/blob/main/GivTCP/givenergy_modbus_async/model/register.py#L719)
```
class BatteryCalibrationStage(IntEnum):
    """Battery calibration stages."""

    OFF = 0
    DISCHARGE = 1
    SET_LOWER_LIMIT = 2
    CHARGE = 3
    SET_UPPER_LIMIT = 4
    BALANCE = 5
    SET_FULL_CAPACITY = 6
    FINISH = 7
```
The error thrown by GivTCP is:
```
2024-10-06 15:35:24,283 - Inv1 - read        -  [DEBUG   ] - Calculating Mode...
2024-10-06 15:35:24,283 - Inv1 - read        -  [DEBUG   ] - Mode is: Eco
2024-10-06 15:35:24,283 - Inv1 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7fa935a880>), 893)
2024-10-06 15:35:24,284 - Inv1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa935b200>), 1633)
2024-10-06 15:35:24,285 - Inv1 - read        -  [DEBUG   ] - processData result: processData Error processing registers: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa935b200>), 1633)
2024-10-06 15:35:24,285 - Inv1 - read        -  [DEBUG   ] - Running pubFromPickle
2024-10-06 15:35:24,286 - Inv1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1809)

```

britkat1980, 2024-10-24:
Ok, thanks I’ll add in other options for calibration stages, but only those ones already there can be set

ungrim97, 2025-03-12:
Is there any update on this (or a workaround). I tried to install GivTCP v3 to try and fix a stuck callibration register, but I can't run it due to the above issue

closed 2025-10-23 by britkat1980