britkat1980/giv_tcp #273: read.py errors in 3.0.4

willholley, 2024-10-08:
Version: 3.0.4
Environment: HAOS, HA 2024.10.1
Inverter: AC 3.0

After updating to GivTCP v3. I'm seeing lots of timeouts and errors. I've cleared out all the pkl files, tried reinstalling, etc. The debug
log file is available at https://gist.github.com/willholley/dc33b004ec12b95c661a2f8e02bc8ca3.

Unfortunately, the main error in the logs doesn't contain much detail:
```
2024-10-08 09:06:39,291 - Inv1 - read        -  [DEBUG   ] - Error num 1 in watch loop execute command: ('Exception', 'read.py', 175)
```

Reverting back to 2.4.1 seems fine, and I don't see the timeout or read errors.

willholley, 2024-10-11:
I've played with this on a local build. The first set of operations after startup always works but the subsequent partial / full refreshes always timeout, regardless of the timeout value or refresh timeouts. In reality, this "works" because after 5 timeouts the whole process restarts.

The exceptions don't provide any useful detail so next thing I can try is Wireshark...

willholley, 2024-10-16:
I made a bit of progress. Attaching a debugger, I can see that during a partial refresh, we start hitting a `NotImplementedError`  at https://github.com/britkat1980/giv_tcp/blob/fe07cd56e01607bf97b360b3a5a23b544219c820/GivTCP/givenergy_modbus_async/pdu/transparent.py#L154 with function code 57. This is confirmed in the Wireshark trace (frame 3) linked in https://github.com/britkat1980/giv_tcp/issues/241#issuecomment-2414646674.

This is not converted to an `InvalidFrame` error at https://github.com/britkat1980/giv_tcp/blame/fe07cd56e01607bf97b360b3a5a23b544219c820/GivTCP/givenergy_modbus_async/pdu/base.py#L98 and so surfaces as a `TimeoutError` at https://github.com/britkat1980/giv_tcp/blob/fe07cd56e01607bf97b360b3a5a23b544219c820/GivTCP/givenergy_modbus_async/client/client.py#L453.

Presumably giv_tcp should explicitly handle function code 57, but uncommenting the conversion to `InvalidFrame` error also makes this more obvious.

britkat1980, 2024-10-24:
Fixed in 3.0.4

closed 2024-10-24 by britkat1980