britkat1980/giv_tcp #275: SetCharge and Discharge - did change..

BrianUK6, 2024-10-08:
[2.1.3](https://github.com/britkat1980/giv_tcp/releases/tag/2.1.3)
BREAKING CHANGES:
Charge and Discharge rates now use absolute power values (2600W) not percentage (100%) to align with Cloud portal and to give correct operation

Found it : This is what I was using.   That has gone away in v.3

BrianUK6, 2024-10-09:
Had a more detailed look through the code.
These 2 variants are different fields in the Inverter, and from FB the lower net value of the 2 is used.

SetDischargeRate & SetChargeRate  -  used to be Watts, and is now % --  implemented in all units.
SetDischargeRateAC & SetChargeRateAC  -- has always been % -- only implemented in MQTT

The Portal does both .  In the Portal, if you change the AC % the MQTT reflects the change.
[image]
So the "AC" variant above needs to be fully implemented in:  api.json, GivLUT.py, REST.py, write.py.

The code change for SetDischargeRate to % confuses things as you never see that % change as a percentage value recorded in MQTT or the portal. The % change wasn't necessary when the "AC" variant exists already.

I cannot use the AC field variant because it's only partially implemented in GivTCP, so going to change code for Set..Rate to allow a "w" or "%" to be appended (default %) and react accordingly.

britkat1980, 2024-10-21:
If fixed in v3 I'll close this now

closed 2024-10-21 by britkat1980