britkat1980/giv_tcp #277: increased disk write on 3.0.3

Tau512, 2024-10-12:
I upgraded from the GivTCP v2 to v3 this wednesday and since then i'm seeing elevated disk write IO thats related to the GivTCP addon.

HAOS is ran in proxmox VM.
Proxmox disk stats shows the following for 1 hour period. the drop at the end was me stopping the GivTCP addon:
[image]

1 week period, with the rise on 9th Oct related to v3.0.3 installation.
[image]

note that the increase is not immediate, and slowly increases over a period of around 10hours between low and ~650kB where it levels out at.

while i run this on enterprise level SSD storage so not concerned with the write wear, this does seem to be excess writes in the region of 12x increase (compared to v2 releases)  and could be a problem for mSD rPi users.

Tau512, 2024-10-13:
This is a 1 day period to show the issue a bit clearer.
[image]
this one shows the point when i stopped the GivTCP addon as i reported yesterday (13:00 on 12/oct/24). It increased disk usage till 01:00 on 13/oct/24 which then leveled out.

britkat1980, 2024-10-21:
Thanks for diagnostics on this. I think its due to a much larger cache stack (history of 24hrs read data). This was in an effort to implement a more robust outlier detection mechanism, but it wasn't sufficiently effective, so this is being rolled back in upcoming Beta 3.0.6. Hopefully this will revert disk writes to previous lower level

britkat1980, 2024-10-24:
Fixed in 3.0.4

closed 2024-10-24 by britkat1980