britkat1980/giv_tcp #278: 3.0.3 'invalid number' errors writing to Influx

cdrajb, 2024-10-14:
Talking to my AC3, 3.0.3 seems to be getting info and putting it out to mqtt
It discovers the inverter as "Gen1 - Ac with 1 batteries", which sounds OK.
But writes to Influx are getting error 400:
2024-10-14 09:12:35,223 - Inv1 - rest        -  [DEBUG   ] - <<< Body: {"code":"invalid","message":"unable to parse 'CE2051xxxx,tagKey=GivReal ac_charge_power=0,battery_current=None,battery_power=420,battery_voltage=52.29,charge_power=0,charge_time_remaining=0,combined_generation_power=None,discharge_completion_time=\"2024-10-14T20:07:35.104420+01:00\",discharge_power=420,discharge_time_remaining=655,eps_power=1,export_power=0,grid_current=2.02,grid_frequency=50.09,grid_power=-15,grid_voltage=246.0,import_power=15,inverter_output_frequency=50.1,invertor_power=417,load_power=525,pv_current_string_1=3.0,pv_current_string_2=0.0,pv_power=93,pv_power_string_1=93,pv_power_string_2=0,pv_voltage_string_1=243.3,pv_voltage_string_2=0.0,soc=60,soc_kwh=4.91,self_consumption_power=510,battery_to_grid=0,battery_to_house=420,grid_to_battery=0,grid_to_house=15,solar_to_battery=0,solar_to_grid=0,solar_to_house=93,ac_charge_energy_today_kwh=6.8,battery_charge_energy_today_kwh=6.8,battery_discharge_energy_today_kwh=2.5,battery_throughput_today_kwh=9.3,export_energy_today_kwh=0.0,import_energy_today_kwh=10.5,invertor_energy_today_kwh=2.5,load_energy_today_kwh=6.2,pv_energy_today_kwh=0.0,self_consumption_energy_today_kwh=0,ac_charge_energy_total_kwh=11579.1,battery_charge_energy_total_kwh=4919.1,battery_discharge_energy_total_kwh=3795.9,battery_throughput_total_kwh=21822.2,export_energy_total_kwh=437.1,import_energy_total_kwh=24897.6,invertor_energy_total_kwh=9592.6,load_energy_total_kwh=28891.5,pv_energy_total_kwh=6417.5,self_consumption_energy_total_kwh=5980.4': invalid number"}

I'm guessing  it's the "combined_generation_power=None" and "battery_current=None" values causing this.  Have I missed a step?
(I just did docker pull as for previous versions - 2.4.3 is working fine)

cdrajb, 2024-10-20:
I commented out the following lines in read.py:  lines 974 and 1512 and 955
I then restarted the container, and it seems to be working now (it is writing to Influx).  
I'd say this confirms my suspicion that it's the values for those two variables that were causing the problem. 
(I'm now struggling to turn the log level back to Info: something keeps overwriting the settings.py file!)

britkat1980, 2024-10-21:
I've added a check for None values to avoid this from Beta 3.0.6 (coming soon). Also you should modify the allsettings.json file rather than the settings.py. That is recreated at each reboot

cdrajb, 2024-10-26:
> Fixed in 3.0.4

Thanks.   Though I should report that I've tested 3.0.4, but am still
getting errors writing to Influx.   It's a KeyError:

2024-10-26 11:42:42,807 - read - [DEBUG] - Pushing output to Influx
2024-10-26 11:42:42,808 - read - [ERROR] - Error publishing data from
pickle: ('KeyError', 'read.py', 1843)

By adding a try:except block around the GivInflux.publish command in
read.py, it looks like it might be from Battery_to_Grid:

/config/GivTCP/logs # tail log_inv_1.log
2024-10-26 21:35:31,341 - mqtt - [DEBUG] - Prepping Flows for publishing
2024-10-26 21:35:31,341 - mqtt - [DEBUG] - Prepping Power for publishing
2024-10-26 21:35:31,344 - mqtt - [DEBUG] - Publishing:
GivEnergy/CE2051G156/Stats
2024-10-26 21:35:31,345 - mqtt - [DEBUG] - Publishing:
GivEnergy/CE2051G156/Timeslots
2024-10-26 21:35:31,347 - mqtt - [DEBUG] - Publishing:
GivEnergy/CE2051G156/raw
2024-10-26 21:35:31,347 - mqtt - [DEBUG] - Prepping BG2047G077 for
publishing
2024-10-26 21:35:31,347 - mqtt - [DEBUG] - Prepping batteries for publishing
2024-10-26 21:35:31,348 - mqtt - [DEBUG] - Prepping invertor for publishing
2024-10-26 21:35:32,159 - read - [DEBUG] - Pushing output to Influx
2024-10-26 21:35:32,160 - read - [ERROR] - KeyError publishing data to
Influx: 'Battery_to_Grid'