britkat1980/giv_tcp #279: Britkat1980 Appreciation

poolski, 2024-10-16:
Not strictly-speaking an Issue™, but I'd like us to take a moment to appreciate 
- How responsive and patient @britkat1980 is, even to "RTFM"-type issues
- How much work they've put into this project
- [add your own]

peter1rhodes, 2024-10-16:
Hear, hear

yavinmaster, 2024-10-16:
Totally agree - I saw he had a ["Buy me a coffee"](https://www.buymeacoffee.com/givtcp) link in the latest readme so last week showed my appreciation with this.  After a year of using the software and putting up with us all asking for support, I thought it only right.

plandregan, 2024-12-14:
Yes, he is a nice guy and perodically visits GivHQ for some internal testing and discussions of future work