britkat1980/giv_tcp #286: Development of 3 Phase Inverter Compatability

LeeFarm11, 2024-10-21:
Hi @britkat1980  Using Dev 3.0.32

First issue is here -

```
2024-10-21 19:41:26,798 - read - [INFO] - Starting watch_plant loop...
2024-10-21 19:41:26,815 - read - [CRITICAL] - Detecting inverter characteristics...
2024-10-21 19:41:47,835 - read - [DEBUG] - Running full refresh
2024-10-21 19:41:47,835 - read - [DEBUG] - Running processData
2024-10-21 19:41:47,835 - read - [DEBUG] - Beginning parsing of Inverter data
2024-10-21 19:41:47,852 - read - [DEBUG] - Getting Battery Details
2024-10-21 19:41:47,855 - read - [DEBUG] - Building battery output: 
2024-10-21 19:41:47,856 - read - [DEBUG] - Battery HY2343G302 added
2024-10-21 19:41:47,857 - read - [DEBUG] - Building battery output: 
2024-10-21 19:41:47,858 - read - [DEBUG] - Battery HY2343G257 added
2024-10-21 19:41:47,859 - read - [DEBUG] - Building battery output: 
2024-10-21 19:41:47,860 - read - [DEBUG] - Battery HY2343G446 added
2024-10-21 19:41:47,860 - read - [DEBUG] - Building battery output: 
2024-10-21 19:41:47,861 - read - [DEBUG] - Battery HY2343G025 added
2024-10-21 19:41:47,862 - read - [DEBUG] - Building battery output: 
2024-10-21 19:41:47,863 - read - [DEBUG] - Battery HY2343G259 added
2024-10-21 19:41:47,864 - read - [DEBUG] - Building battery output: 
2024-10-21 19:41:47,865 - read - [DEBUG] - Battery HY2343G144 added
2024-10-21 19:41:47,866 - read - [DEBUG] - Getting mode control figures
2024-10-21 19:41:47,867 - read - [DEBUG] - Saving the battery reserve percentage for later: 4
2024-10-21 19:41:47,868 - read - [DEBUG] - Calculating Mode...
2024-10-21 19:41:47,868 - read - [DEBUG] - Mode is: Eco (Paused)
2024-10-21 19:41:47,869 - read - [DEBUG] - Force Export is not Running
2024-10-21 19:41:47,870 - read - [DEBUG] - Getting TimeSlot data
2024-10-21 19:41:47,871 - read - [DEBUG] - Getting TimeSlot data
2024-10-21 19:41:47,874 - read - [DEBUG] - cache len= 0
2024-10-21 19:41:47,875 - read - [ERROR] - inverter Update failed so using last known good data from cache: ((<class 'KeyError'>, KeyError('Grid_Power'), <traceback object at 0x7eff9ec1cc80>), 1681)
2024-10-21 19:41:47,876 - read - [DEBUG] - processData result: processData Error processing registers: ((<class 'KeyError'>, KeyError('Grid_Power'), <traceback object at 0x7eff9ec1cc80>), 1681)
2024-10-21 19:41:47,877 - read - [DEBUG] - Running pubFromPickle
2024-10-21 19:41:47,877 - read - [ERROR] - runAll2 Error processing registers: ('KeyError', 'read.py', 1837)
2024-10-21 19:42:17,923 - read - [DEBUG] - Running partial refresh
2024-10-21 19:42:23,864 - read - [DEBUG] - Data get was successful, now running handler if needed: 
2024-10-21 19:42:23,864 - read - [DEBUG] - Running processData
2024-10-21 19:42:23,864 - read - [DEBUG] - Beginning parsing of Inverter data
2024-10-21 19:42:23,880 - read - [DEBUG] - Getting Battery Details
2024-10-21 19:42:23,883 - read - [DEBUG] - Building battery output: 
2024-10-21 19:42:23,885 - read - [DEBUG] - Battery HY2343G302 added
2024-10-21 19:42:23,885 - read - [DEBUG] - Building battery output: 
2024-10-21 19:42:23,886 - read - [DEBUG] - Battery HY2343G257 added
2024-10-21 19:42:23,887 - read - [DEBUG] - Building battery output: 
2024-10-21 19:42:23,888 - read - [DEBUG] - Battery HY2343G446 added
2024-10-21 19:42:23,889 - read - [DEBUG] - Building battery output: 
2024-10-21 19:42:23,890 - read - [DEBUG] - Battery HY2343G025 added
2024-10-21 19:42:23,891 - read - [DEBUG] - Building battery output: 
2024-10-21 19:42:23,892 - read - [DEBUG] - Battery HY2343G259 added
2024-10-21 19:42:23,892 - read - [DEBUG] - Building battery output: 
2024-10-21 19:42:23,894 - read - [DEBUG] - Battery HY2343G144 added
2024-10-21 19:42:23,895 - read - [DEBUG] - Getting mode control figures
2024-10-21 19:42:23,896 - read - [DEBUG] - Calculating Mode...
2024-10-21 19:42:23,897 - read - [DEBUG] - Mode is: Eco (Paused)
2024-10-21 19:42:23,897 - read - [DEBUG] - Force Export is not Running
2024-10-21 19:42:23,899 - read - [DEBUG] - Getting TimeSlot data
2024-10-21 19:42:23,899 - read - [DEBUG] - Getting TimeSlot data
2024-10-21 19:42:23,902 - read - [DEBUG] - cache len= 0
2024-10-21 19:42:23,904 - read - [ERROR] - inverter Update failed so using last known good data from cache: ((<class 'KeyError'>, KeyError('Grid_Power'), <traceback object at 0x7eff9ec41a40>), 1681)
2024-10-21 19:42:23,904 - read - [DEBUG] - processData result: processData Error processing registers: ((<class 'KeyError'>, KeyError('Grid_Power'), <traceback object at 0x7eff9ec41a40>), 1681)
2024-10-21 19:42:23,905 - read - [DEBUG] - Running pubFromPickle
2024-10-21 19:42:23,906 - read - [ERROR] - runAll2 Error processing registers: ('KeyError', 'read.py', 1837)
2024-10-21 19:42:53,940 - read - [DEBUG] - Running partial refresh
2024-10-21 19:42:58,374 - read - [DEBUG] - Data get was successful, now running handler if needed: 
2024-10-21 19:42:58,375 - read - [DEBUG] - Running processData
2024-10-21 19:42:58,376 - read - [DEBUG] - Beginning parsing of Inverter data
```

Approximately every 7 minutes I do get attempt to remove cache (pkl) files -

```
2024-10-21 19:58:08,383 - read - [DEBUG] - Building battery output: 
2024-10-21 19:58:08,383 - read - [DEBUG] - Battery HY2343G025 added
2024-10-21 19:58:08,384 - read - [DEBUG] - Building battery output: 
2024-10-21 19:58:08,385 - read - [DEBUG] - Battery HY2343G259 added
2024-10-21 19:58:08,385 - read - [DEBUG] - Building battery output: 
2024-10-21 19:58:08,386 - read - [DEBUG] - Battery HY2343G144 added
2024-10-21 19:58:08,387 - read - [DEBUG] - Getting mode control figures
2024-10-21 19:58:08,388 - read - [DEBUG] - Calculating Mode...
2024-10-21 19:58:08,388 - read - [DEBUG] - Mode is: Eco (Paused)
2024-10-21 19:58:08,389 - read - [DEBUG] - Force Export is not Running
2024-10-21 19:58:08,389 - read - [DEBUG] - Getting TimeSlot data
2024-10-21 19:58:08,390 - read - [DEBUG] - Getting TimeSlot data
2024-10-21 19:58:08,392 - read - [DEBUG] - cache len= 0
2024-10-21 19:58:08,393 - read - [ERROR] - 10 failed inverter reads in a row so removing Cache (pkl) files to force update...
2024-10-21 19:58:08,394 - read - [ERROR] - inverter Update failed so using last known good data from cache: ((<class 'KeyError'>, KeyError('Grid_Power'), <traceback object at 0x7fbb69601f80>), 1681)
2024-10-21 19:58:08,394 - read - [DEBUG] - processData result: processData Error processing registers: ((<class 'KeyError'>, KeyError('Grid_Power'), <traceback object at 0x7fbb69601f80>), 1681)
2024-10-21 19:58:08,394 - read - [DEBUG] - Running pubFromPickle
2024-10-21 19:58:08,395 - read - [ERROR] - runAll2 Error processing registers: ('KeyError', 'read.py', 1837)
2024-10-21 19:58:38,450 - read - [DEBUG] - Running partial refresh
2024-10-21 19:58:43,384 - read - [DEBUG] - Data get was successful, now running handler if needed: 
2024-10-21 19:58:43,384 - read - [DEBUG] - Running processData
2024-10-21 19:58:43,385 - read - [DEBUG] - Beginning parsing of Inverter data
2024-10-21 19:58:43,401 - read - [DEBUG] - Getting Battery Details
2024-10-21 19:58:43,403 - read - [DEBUG] - Building battery output: 
2024-10-21 19:58:43,404 - read - [DEBUG] - Battery HY2343G302 added
2024-10-21 19:58:43,404 - read - [DEBUG] - Building battery output: 

```

LeeFarm11, 2024-10-21:
By the way, some background info. Giv have made some changes recently. In the Portal we now see some changes -

1. The numbering of the Charge and Discharge Slots has been messed up between Remote Control and the App/ Portal Settings. I raised in on FaceBook a while ago and now it has been mentioned in the Known Issues section in the Portal -

3PH Charge Slots - Between Remote Control and Inverter Card/App

We have identified that there is a small error and therefore mismatch of charge slots, 9 and 10 on the inverter card/app, pair with 1 and 2 on the remote control. This will be resolved imminently.

In the short term - remote control can be utilized if this is an easier way to manage your charge slots.
PORTAL
WORKING ON IT
Created at 2024-10-16 09:31:21 by Adam R. (GivEnergy) 

2. Remote Control has an issue with Charge Slot 2 -

There are 2 copies of Start Time and End Time. They seem to work in unison, so I **guess** only a UI issue.

[image]

3. New items -

[image]

For discharge lower limit we now have -

Power Factor, Battery Reserve %
Charge Settings - Discharge Down to %
Charge/ Discharge Schedules - DC Discharge (1 to 10) Lower SOC % Limit

I wonder in what order they have priority.

4. I attach latest set of settings available by REST and my current settings.

[Inv_Values.csv](https://github.com/user-attachments/files/17467081/Inv_Values.csv)

LeeFarm11, 2024-10-21:
Now updated to Dev 3.0.36

Various strange looking messages -

2024-10-21 22:47:33,432 - HA_Discovery - [ERROR] - Error connecting to MQTT Broker: (<class 'RuntimeError'>, RuntimeError('dictionary changed size during iteration'), <traceback object at 0x7f3f881fdd80>)

2024-10-21 22:48:52,924 - read - [DEBUG] - Inverter_Power_Out is outside of allowable bounds so using old value. Out of bounds value is: 429490069.0. Min limit: -30000. Max limit: 30000

2024-10-21 22:48:52,924 - read - [DEBUG] - Data cleansed for: Battery_Details
2024-10-21 22:48:52,924 - read - [DEBUG] - Inverter_Power_Out is outside of allowable bounds so using old value. Out of bounds value is: 429490069.0. Min limit: -30000. Max limit: 30000
2024-10-21 22:48:52,925 - read - [DEBUG] - Data cleansed for: Power
2024-10-21 22:48:52,926 - read - [DEBUG] - Data cleansed for: Power
2024-10-21 22:48:52,926 - read - [DEBUG] - Power_Factor is Zero so using old value
2024-10-21 22:48:52,927 - read - [DEBUG] - Data cleansed for: TD2343G049
2024-10-21 22:48:52,927 - read - [DEBUG] - Frequency is Zero so using old value
2024-10-21 22:48:52,927 - read - [DEBUG] - Phase_1_Power_Factor is Zero so using old value
2024-10-21 22:48:52,928 - read - [DEBUG] - Phase_2_Power_Factor is Zero so using old value
2024-10-21 22:48:52,928 - read - [DEBUG] - Phase_3_Power_Factor is Zero so using old value
2024-10-21 22:48:52,928 - read - [DEBUG] - Data cleansed for: Meter_ID1
2024-10-21 22:48:52,928 - read - [DEBUG] - Data cleansed for: Meter_Details
2024-10-21 22:48:52,929 - read - [DEBUG] - Export2_Energy_Today_kWh is Zero so using old value
2024-10-21 22:48:52,929 - read - [DEBUG] - Data cleansed for: Today
2024-10-21 22:48:52,929 - read - [DEBUG] - Export2_Energy_Total_kWh is Zero so using old value
2024-10-21 22:48:52,930 - read - [DEBUG] - Data cleansed for: Total
2024-10-21 22:48:52,930 - read - [DEBUG] - Data cleansed for: Energy
2024-10-21 22:48:52,931 - read - [DEBUG] - Data cleansed for: Timeslots
2024-10-21 22:48:52,931 - read - [DEBUG] - Battery_Power_Cutoff is Zero so using old value

Same messages repeating every 30 seconds -
2024-10-21 22:51:49,475 - read - [DEBUG] - Inverter_Power_Out is outside of allowable bounds so using old value. Out of bounds value is: 429490067.0. Min limit: -30000. Max limit: 30000
2024-10-21 22:51:49,476 - read - [DEBUG] - Data cleansed for: Power
2024-10-21 22:51:49,476 - read - [DEBUG] - Data cleansed for: Power
2024-10-21 22:51:49,476 - read - [DEBUG] - Power_Factor is Zero so using old value
2024-10-21 22:51:49,479 - read - [DEBUG] - Data cleansed for: TD2343G049
2024-10-21 22:51:49,480 - read - [DEBUG] - Frequency is Zero so using old value
2024-10-21 22:51:49,480 - read - [DEBUG] - Phase_1_Power_Factor is Zero so using old value
2024-10-21 22:51:49,480 - read - [DEBUG] - Phase_2_Power_Factor is Zero so using old value
2024-10-21 22:51:49,481 - read - [DEBUG] - Phase_3_Power_Factor is Zero so using old value
2024-10-21 22:51:49,481 - read - [DEBUG] - Data cleansed for: Meter_ID1
2024-10-21 22:51:49,481 - read - [DEBUG] - Data cleansed for: Meter_Details
2024-10-21 22:51:49,482 - read - [DEBUG] - Export2_Energy_Today_kWh is Zero so using old value
2024-10-21 22:51:49,482 - read - [DEBUG] - Data cleansed for: Today
2024-10-21 22:51:49,482 - read - [DEBUG] - Export2_Energy_Total_kWh is Zero so using old value
2024-10-21 22:51:49,483 - read - [DEBUG] - Data cleansed for: Total
2024-10-21 22:51:49,483 - read - [DEBUG] - Data cleansed for: Energy
2024-10-21 22:51:49,483 - read - [DEBUG] - Data cleansed for: Timeslots
2024-10-21 22:51:49,484 - read - [DEBUG] - Battery_Power_Cutoff is Zero so using old value

Also this multiple times -
2024-10-21 22:52:59,074 - read - [INFO] - Phase_1_Current jumped too far in a single read: 2.7->10.1 so using previous value

Also this -
2024-10-21 22:56:29,413 - read - [INFO] - Inverter_Power_Out jumped too far in a single read: 429490069.0->2655.0 so using previous value
2024-10-21 22:56:29,413 - read - [INFO] - Load_Phase2_Power jumped too far in a single read: 0.0->6461.2 so using previous value
2024-10-21 22:56:29,413 - read - [INFO] - Load_Phase3_Power jumped too far in a single read: 23.6->6485.0 so using previous value
2024-10-21 22:56:29,414 - read - [INFO] - Export_Phase1_Power jumped too far in a single read: 6550.0->87.6 so using previous value
2024-10-21 22:56:29,415 - read - [INFO] - Export_Phase2_Power jumped too far in a single read: 6550.7->91.3 so using previous value
2024-10-21 22:56:29,415 - read - [INFO] - Export_Phase3_Power jumped too far in a single read: 6548.6->86.6 so using previous value
2024-10-21 22:56:29,416 - read - [INFO] - Grid_Phase1_Current jumped too far in a single read: 0.8->3.9 so using previous value
2024-10-21 22:56:29,416 - read - [INFO] - Grid_Phase2_Current jumped too far in a single read: 0.8->4.0 so using previous value
2024-10-21 22:56:29,417 - read - [INFO] - Grid_Phase3_Current jumped too far in a single read: 0.8->3.8 so using previous value
2024-10-21 22:56:29,418 - read - [DEBUG] - Data cleansed for: Power
2024-10-21 22:56:29,418 - read - [DEBUG] - Data cleansed for: Power
2024-10-21 22:56:29,419 - read - [DEBUG] - Power_Factor is Zero so using old value
2024-10-21 22:56:29,420 - read - [DEBUG] - Data cleansed for: TD2343G049
2024-10-21 22:56:29,420 - read - [INFO] - Phase_1_Current jumped too far in a single read: 2.7->10.1 so using previous value
2024-10-21 22:56:29,421 - read - [DEBUG] - Frequency is Zero so using old value
[... cut ...]
2024-10-21 22:49:29,111 - read - [DEBUG] - Running pubFromPickle
2024-10-21 22:49:29,113 - read - [DEBUG] - Dealt with HY2343G025
2024-10-21 22:49:29,113 - read - [DEBUG] - Dealt with HY2343G144
2024-10-21 22:49:29,115 - read - [DEBUG] - Dealt with HY2343G257
2024-10-21 22:49:29,116 - read - [DEBUG] - Dealt with HY2343G259
2024-10-21 22:49:29,117 - read - [DEBUG] - Dealt with HY2343G302
2024-10-21 22:49:29,118 - read - [DEBUG] - Dealt with HY2343G446
2024-10-21 22:49:29,119 - read - [DEBUG] - Dealt with Battery_Stack_1

```

britkat1980, 2024-10-27:
It looks like it's running OK now? If you run latest prod version 3.0.4 do you get those same. Any issues with the values and/or control

LeeFarm11, 2024-10-27:
I didn't think it was running OK with those above messages, so I haven't tested further. Are you suggesting that each of the above messages are ok? Especially those repeated below -

1.
2024-10-21 22:47:33,432 - HA_Discovery - [ERROR] - Error connecting to MQTT Broker: (<class 'RuntimeError'>, RuntimeError('dictionary changed size during iteration'), <traceback object at 0x7f3f881fdd80>)

2.
2024-10-21 22:48:52,924 - read - [DEBUG] - Inverter_Power_Out is outside of allowable bounds so using old value. Out of bounds value is: 429490069.0. Min limit: -30000. Max limit: 30000

3.
2024-10-21 22:48:52,924 - read - [DEBUG] - Inverter_Power_Out is outside of allowable bounds so using old value. Out of bounds value is: 429490069.0. Min limit: -30000. Max limit: 30000

4.
2024-10-21 22:48:52,929 - read - [DEBUG] - Export2_Energy_Today_kWh is Zero so using old value
2024-10-21 22:48:52,929 - read - [DEBUG] - Data cleansed for: Today
2024-10-21 22:48:52,929 - read - [DEBUG] - Export2_Energy_Total_kWh is Zero so using old value

5.
2024-10-21 22:52:59,074 - read - [INFO] - Phase_1_Current jumped too far in a single read: 2.7->10.1 so using previous value

6.
2024-10-21 22:56:29,413 - read - [INFO] - Inverter_Power_Out jumped too far in a single read: 429490069.0->2655.0 so using previous value
2024-10-21 22:56:29,413 - read - [INFO] - Load_Phase2_Power jumped too far in a single read: 0.0->6461.2 so using previous value
2024-10-21 22:56:29,413 - read - [INFO] - Load_Phase3_Power jumped too far in a single read: 23.6->6485.0 so using previous value
2024-10-21 22:56:29,414 - read - [INFO] - Export_Phase1_Power jumped too far in a single read: 6550.0->87.6 so using previous value
2024-10-21 22:56:29,415 - read - [INFO] - Export_Phase2_Power jumped too far in a single read: 6550.7->91.3 so using previous value
2024-10-21 22:56:29,415 - read - [INFO] - Export_Phase3_Power jumped too far in a single read: 6548.6->86.6 so using previous value
2024-10-21 22:56:29,416 - read - [INFO] - Grid_Phase1_Current jumped too far in a single read: 0.8->3.9 so using previous value
2024-10-21 22:56:29,416 - read - [INFO] - Grid_Phase2_Current jumped too far in a single read: 0.8->4.0 so using previous value
2024-10-21 22:56:29,417 - read - [INFO] - Grid_Phase3_Current jumped too far in a single read: 0.8->3.8 so using previous value

LeeFarm11, 2024-11-01:
Hi. Think I misread your last post. I have now installed Prod version 3.0.4.
And now running for a couple of days.

I still have this every 15-20 seconds - 

```
2024-11-01 12:11:17,407 - read - [DEBUG] - Power_Factor is Zero so using old value
2024-11-01 12:11:17,408 - read - [DEBUG] - Data cleansed for: TD2343G049
2024-11-01 12:11:17,408 - read - [DEBUG] - Frequency is Zero so using old value
2024-11-01 12:11:17,410 - read - [DEBUG] - Phase_1_Power_Factor is Zero so using old value
2024-11-01 12:11:17,413 - read - [DEBUG] - Phase_2_Power_Factor is Zero so using old value
2024-11-01 12:11:17,413 - read - [DEBUG] - Phase_3_Power_Factor is Zero so using old value
2024-11-01 12:11:17,414 - read - [DEBUG] - Data cleansed for: Meter_ID1
2024-11-01 12:11:17,414 - read - [DEBUG] - Data cleansed for: Meter_Details
2024-11-01 12:11:17,415 - read - [DEBUG] - Export2_Energy_Today_kWh is Zero so using old value
2024-11-01 12:11:17,415 - read - [DEBUG] - Data cleansed for: Today
2024-11-01 12:11:17,418 - read - [DEBUG] - Export2_Energy_Total_kWh is Zero so using old value
2024-11-01 12:11:17,419 - read - [DEBUG] - Data cleansed for: Total
2024-11-01 12:11:17,419 - read - [DEBUG] - Data cleansed for: Energy
2024-11-01 12:11:17,420 - read - [DEBUG] - Data cleansed for: Timeslots
2024-11-01 12:11:17,421 - read - [DEBUG] - Battery_Power_Cutoff is Zero so using old value
2024-11-01 12:11:17,421 - read - [DEBUG] - Data cleansed for: Control
2024-11-01 12:11:17,422 - read - [DEBUG] - Data cleansed for: Stats
```

I also have 414 'jumped too far in a single read' messages today like here -

```
2024-11-01 14:05:52,772 - read - [DEBUG] - Data cleansed for: Battery_Details
2024-11-01 14:05:52,773 - read - [DEBUG] - Battery_Charge_Power jumped too far in a single read: 0.5->11142.3 so using previous value
2024-11-01 14:05:52,774 - read - [DEBUG] - Returning new Battery_Discharge_Power data as the last read was smoothed
2024-11-01 14:05:52,774 - read - [DEBUG] - Inverter_Power_Out is outside of allowable bounds so using old value. Out of bounds value is: 429485933.6. Min limit: -30000. Max limit: 30000
2024-11-01 14:05:52,774 - read - [DEBUG] - Returning new AC_Charge_Power data as the last read was smoothed
2024-11-01 14:05:52,775 - read - [DEBUG] - Returning new Grid_Apparent_Power data as the last read was smoothed
2024-11-01 14:05:52,775 - read - [DEBUG] - Returning new Meter_Import_Power data as the last read was smoothed
2024-11-01 14:05:52,776 - read - [DEBUG] - Returning new Load_Phase2_Power data as the last read was smoothed
2024-11-01 14:05:52,776 - read - [DEBUG] - Returning new Load_Phase3_Power data as the last read was smoothed
2024-11-01 14:05:52,776 - read - [DEBUG] - Export_Phase1_Power jumped too far in a single read: 22.6->6196.7 so using previous value
2024-11-01 14:05:52,777 - read - [DEBUG] - Export_Phase2_Power jumped too far in a single read: 25.4->6192.6 so using previous value
2024-11-01 14:05:52,777 - read - [DEBUG] - Export_Phase3_Power jumped too far in a single read: 21.6->6191.9 so using previous value
2024-11-01 14:05:52,778 - read - [DEBUG] - Returning new Grid_Phase1_Current data as the last read was smoothed
2024-11-01 14:05:52,778 - read - [DEBUG] - Returning new Grid_Phase2_Current data as the last read was smoothed
2024-11-01 14:05:52,778 - read - [DEBUG] - Returning new Grid_Phase3_Current data as the last read was smoothed
2024-11-01 14:05:52,779 - read - [DEBUG] - Data cleansed for: Power
2024-11-01 14:05:52,779 - read - [DEBUG] - Data cleansed for: Power
2024-11-01 14:05:52,779 - read - [DEBUG] - Power_Factor is Zero so using old value
2024-11-01 14:05:52,780 - read - [DEBUG] - Data cleansed for: TD2343G049
2024-11-01 14:05:52,780 - read - [DEBUG] - Phase_1_Current jumped too far in a single read: 7.3->20.2 so using previous value
2024-11-01 14:05:52,780 - read - [DEBUG] - Phase_2_Current jumped too far in a single read: 1.1->12.1 so using previous value
2024-11-01 14:05:52,781 - read - [DEBUG] - Phase_3_Current jumped too far in a single read: 3.2->10.3 so using previous value
2024-11-01 14:05:52,781 - read - [DEBUG] - Phase_1_Power jumped too far in a single read: 1722->6943 so using previous value
2024-11-01 14:05:52,781 - read - [DEBUG] - Phase_2_Power jumped too far in a single read: -267->5046 so using previous value
2024-11-01 14:05:52,782 - read - [DEBUG] - Phase_3_Power jumped too far in a single read: -783->4590 so using previous value
2024-11-01 14:05:52,782 - read - [DEBUG] - Frequency is Zero so using old value
2024-11-01 14:05:52,782 - read - [DEBUG] - Phase_1_Power_Factor is Zero so using old value
2024-11-01 14:05:52,782 - read - [DEBUG] - Phase_2_Power_Factor is Zero so using old value
2024-11-01 14:05:52,783 - read - [DEBUG] - Phase_3_Power_Factor is Zero so using old value
2024-11-01 14:05:52,783 - read - [DEBUG] - Data cleansed for: Meter_ID1
2024-11-01 14:05:52,783 - read - [DEBUG] - Data cleansed for: Meter_Details
2024-11-01 14:05:52,784 - read - [DEBUG] - Export2_Energy_Today_kWh is Zero so using old value
2024-11-01 14:05:52,784 - read - [DEBUG] - Data cleansed for: Today
2024-11-01 14:05:52,784 - read - [DEBUG] - Export2_Energy_Total_kWh is Zero so using old value
2024-11-01 14:05:52,785 - read - [DEBUG] - Data cleansed for: Total
2024-11-01 14:05:52,785 - read - [DEBUG] - Data cleansed for: Energy
2024-11-01 14:05:52,785 - read - [DEBUG] - Data cleansed for: Timeslots
[... cut ...]
25 x PV_Voltage_String_1 between 07:44 and 08:26 and between 15:09 and 15:49 today
2024-11-01 08:20:47,318 - read - [DEBUG] - PV_Voltage_String_1 is outside of allowable bounds so using old value. Out of bounds value is: 745.4. Min limit: 0. Max limit: 700

374 x Inverter_Power_Out all starting with 4294 like below today
2024-11-01 04:52:34,147 - read - [DEBUG] - Inverter_Power_Out is outside of allowable bounds so using old value. Out of bounds value is: 429481472.0. Min limit: -30000. Max limit: 30000

LeeFarm11, 2024-11-01:
Also the first 56000 records in the log today, from 00:00:00 to 03:09:52 were like this -

```
2024-11-01 03:09:52,569 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
```
followed by this -
```
2024-11-01 03:09:54,476 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:09:54,481 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:10:06,138 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:10:06,142 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:10:33,042 - read - [ERROR] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 76)
2024-11-01 03:10:33,044 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:10:33,049 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:10:44,685 - read - [ERROR] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 76)
2024-11-01 03:10:44,687 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:10:44,692 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:10:54,521 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:10:54,527 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:11:06,391 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:11:06,397 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:11:33,081 - read - [ERROR] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 76)
2024-11-01 03:11:33,082 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:11:33,094 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:11:44,943 - read - [ERROR] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 76)
2024-11-01 03:11:44,944 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:11:44,948 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:11:54,414 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:11:54,424 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:12:05,510 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:12:05,514 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:12:33,042 - read - [ERROR] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 76)
2024-11-01 03:12:33,044 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:12:33,062 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:12:44,123 - read - [ERROR] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 76)
2024-11-01 03:12:44,123 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:12:44,129 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:12:54,349 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:12:54,353 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:13:05,609 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:13:05,614 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:13:16,922 - read - [ERROR] - Error in inital detect/refresh: ('KeyError', 'read.py', 76)
2024-11-01 03:13:16,923 - read - [INFO] - Starting watch_plant loop...
2024-11-01 03:13:16,927 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-01 03:13:17,580 - read - [DEBUG] - Running full refresh
2024-11-01 03:13:17,580 - read - [DEBUG] - Running processData
2024-11-01 03:13:17,581 - read - [DEBUG] - Beginning parsing of Inverter data
2024-11-01 03:13:17,621 - read - [DEBUG] - Getting Battery Details
2024-11-01 03:13:17,624 - read - [DEBUG] - Building battery output: 
2024-11-01 03:13:17,625 - read - [DEBUG] - Battery HY2343G302 added
2024-11-01 03:13:17,626 - read - [DEBUG] - Building battery output: 
2024-11-01 03:13:17,626 - read - [DEBUG] - Battery HY2343G257 added
2024-11-01 03:13:17,627 - read - [DEBUG] - Building battery output: 
2024-11-01 03:13:17,628 - read - [DEBUG] - Battery HY2343G446 added
2024-11-01 03:13:17,628 - read - [DEBUG] - Building battery output: 
2024-11-01 03:13:17,629 - read - [DEBUG] - Battery HY2343G025 added
2024-11-01 03:13:17,629 - read - [DEBUG] - Building battery output: 
2024-11-01 03:13:17,630 - read - [DEBUG] - Battery HY2343G259 added
2024-11-01 03:13:17,630 - read - [DEBUG] - Building battery output: 
2024-11-01 03:13:17,631 - read - [DEBUG] - Battery HY2343G144 added
2024-11-01 03:13:17,632 - read - [DEBUG] - Getting mode control figures
2024-11-01 03:13:17,633 - read - [DEBUG] - Calculating Mode...
2024-11-01 03:13:17,633 - read - [DEBUG] - Mode is: Eco (Paused)
2024-11-01 03:13:17,634 - read - [DEBUG] - Force Export is not Running
2024-11-01 03:13:17,634 - read - [DEBUG] - Getting TimeSlot data
2024-11-01 03:13:17,635 - read - [DEBUG] - Getting TimeSlot data
2024-11-01 03:13:17,661 - read - [DEBUG] - cache len= 11
2024-11-01 03:13:17,661 - read - [DEBUG] - Running the data cleansing process
2024-11-01 03:13:17,671 - read - [DEBUG] - Stack_Charge_Energy_Today_kWh jumped too far in a single read: 0.1->2.4 so using previous value
2024-11-01 03:13:17,673 - read - [DEBUG] - Data cleansed for: HY2343G302
2024-11-01 03:13:17,674 - read - [DEBUG] - Data cleansed for: HY2343G257
2024-11-01 03:13:17,676 - read - [DEBUG] - Data cleansed for: HY2343G446
2024-11-01 03:13:17,677 - read - [DEBUG] - Data cleansed for: HY2343G025
2024-11-01 03:13:17,678 - read - [DEBUG] - Data cleansed for: HY2343G259
2024-11-01 03:13:17,679 - read - [DEBUG] - Data cleansed for: HY2343G144
[... cut ...]
2024-11-01 03:13:17,679 - read - [DEBUG] - Data cleansed for: Battery_Details
2024-11-01 03:13:17,680 - read - [DEBUG] - Inverter_Power_Out is outside of allowable bounds so using old value. Out of bounds value is: 429485733.6. Min limit: -30000. Max limit: 30000
2024-11-01 03:13:17,681 - read - [DEBUG] - SOC jumped too far in a single read: 10->15 so using previous value
2024-11-01 03:13:17,681 - read - [DEBUG] - Data cleansed for: Power
```

I have also had many thousands more of that mqtt message during the day.

LeeFarm11, 2024-11-01:
And discrepancy between the log file in File Editor and in Addons Log. The Addons Log appears to have more data, but cannot see the additional data if more than a few minutes ago. The evc data is missing from the File Editer Log. But the 'Running full refresh' message only shows in the File Editer Log, not in the Addons Log.

File Editor Log -

```
2024-11-01 18:38:48,591 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:48,793 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:48,997 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:49,199 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:49,403 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:49,610 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:49,815 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:50,021 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:50,225 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:50,426 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:50,629 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:50,830 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:51,031 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:51,233 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:51,439 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:51,640 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:51,844 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:52,045 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:52,252 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:52,459 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:52,661 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:52,862 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:53,063 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:53,264 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:53,346 - read - [DEBUG] - Running full refresh
2024-11-01 18:38:53,469 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:53,675 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:53,884 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:54,086 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:54,287 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:54,488 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:54,689 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:54,893 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:55,098 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:55,299 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:55,500 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:55,706 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:55,908 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:56,109 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:56,311 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:56,511 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:56,713 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:56,916 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:57,117 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:57,318 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:57,519 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:57,723 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:57,932 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:58,138 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:58,339 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:58,540 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:58,741 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:58,949 - mqtt - [DEBUG] - In wait loop (multi_MQTT_publish)
```

And in Addons Log for the same time frame.
```
2024-11-01 18:38:48,591 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:48,793 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:48,997 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:49,199 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:49,403 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:49,610 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:49,815 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:50,021 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
[... 35 lines cut ...]
2024-11-01 18:38:55,098 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:55,299 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
[... cut ...]
2024-11-01 18:38:58,339 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:58,488 - EVC - evc         -  [DEBUG   ] - connected OK Returned code=Success
2024-11-01 18:38:58,489 - EVC - evc         -  [DEBUG   ] - Subscribing to GivEnergy/control/#
2024-11-01 18:38:58,540 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:58,741 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
2024-11-01 18:38:58,949 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (multi_MQTT_publish)
```

britkat1980, 2024-11-01:
Those debug messages aren't of concern, unless the rejected value is actually correct, but they don't seem to be. 
Does the data in the HA graphs seem sensible and stable?
The repeated rebooting of self-run is the OK boy one that needs looking at.

LeeFarm11, 2024-11-01:
Log completely empty (apart from In wait loop) for over 3 hours is OK. Seriously ?

This graph? 
[image]

LeeFarm11, 2024-11-02:
Looking into the graphs a bit more, this seems very weird. The GivTCP Power Load Phase 1 Power, GivTCP Power Load Phase 2 Power, GivTCP Power Load Phase 3 Power are showing really odd results. Doesn't seem to relate to my undestanding of Load. I include print from Giv Portal for comparison.

[image]
[image]
[image]
[image]
[image]

LeeFarm11, 2024-11-04:
[image]

LeeFarm11, 2024-11-04:
Enable Charge Schedule does not work. As previously discussed, on the Phase 3 Inverter it seems this setting does not exist, but it is needed for Predbat. I think it needs to switch both Force Charge Enable and Force AC Charge Enable. I tried to switch On Enable Charge Schedule while Force Charge Enable and Force AC Charge Enable were both Off, and also when they were both On.
Log file from 19:53 to 20:06 is included.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:////Users/leeandnessahome/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip.htm">
<link rel=File-List
href="file:////Users/leeandnessahome/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip_filelist.xml">
<style>

</style>
</head>

<body link="#467886" vlink="#96607D">

Time | Control | Action | Success
-- | -- | -- | --
19:54:00 | Force Charge Enable | Turn off | Yes
19:55:00 | Force AC Charge Enable | Turn off | Yes
19:56:00 | Force Charge Enable | Turn on | Yes
19:57:00 | Force AC Charge Enable | Turn on | Yes
19:58:00 | Force AC Charge Enable | Turn off | Yes
19:59:00 | Force Charge Enable | Turn off | Yes
20:00:00 | Enable Charge Schedule | Turn on | **No**
20:02:00 | Force Charge Enable | Turn on | Yes
20:03:00 | Force AC Charge Enable | Turn on | Yes
20:04:00 | Enable Charge Schedule | Turn on | **No**

</body>

</html>

[GivTCPLog.txt](https://github.com/user-attachments/files/17623325/GivTCPLog.txt)

LeeFarm11, 2024-11-04:
Adding @gcoan for info

gcoan, 2024-11-04:
> Enable Charge Schedule does not work. As previously discussed, on the Phase 3 Inverter it seems this setting does not exist, but it is needed for Predbat

Good to see you're still plugging away at this !

If Predbat didn't require Enable Charge Schedule then do you think (a) 3 phase control via GivTCP is working and (b) Predbat could control it OK?

(and the same with enable discharge schedule) ?

As a test, in your apps.yaml comment out the REST lines so that Predbat is controlling GivTCP using the home assistant controls

Then where the configuration for the enable charge and discharge schedule is in apps.yaml:

```
scheduled_charge_enable:
    - switch.givtcp_{geserial}_enable_charge_schedule
  #  - switch.givtcp2_{geserial2}_enable_charge_schedule
  scheduled_discharge_enable:
    - switch.givtcp_{geserial}_enable_discharge_schedule
  #  - switch.givtcp2_{geserial2}_enable_discharge_schedule
```

Create a dummy helper switch in Home Assistant and put the name of it in place of the givtcp enable_charge_schedule control

And see what happens.

If this is the only problem with getting Predbat to work with GivTCP then we have a workaround for the 3 phase inverters and can look to make it more elegant afterwards (i.e. allowing Predbat to ignore this control for 3 phase inverters and getting Predbat to work in REST mode)

britkat1980, 2024-11-05:
> Enable Charge Schedule does not work. As previously discussed, on the Phase 3 Inverter it seems this setting does not exist, but it is needed for Predbat. I think it needs to switch both Force Charge Enable and Force AC Charge Enable. I tried to switch On Enable Charge Schedule while Force Charge Enable and Force AC Charge Enable were both Off, and also when they were both On. Log file from 19:53 to 20:06 is included.
> 
> <style> </style>
> Time	Control	Action	Success
> 19:54:00	Force Charge Enable	Turn off	Yes
> 19:55:00	Force AC Charge Enable	Turn off	Yes
> 19:56:00	Force Charge Enable	Turn on	Yes
> 19:57:00	Force AC Charge Enable	Turn on	Yes
> 19:58:00	Force AC Charge Enable	Turn off	Yes
> 19:59:00	Force Charge Enable	Turn off	Yes
> 20:00:00	Enable Charge Schedule	Turn on	**No**
> 20:02:00	Force Charge Enable	Turn on	Yes
> 20:03:00	Force AC Charge Enable	Turn on	Yes
> 20:04:00	Enable Charge Schedule	Turn on	**No**
> [GivTCPLog.txt](https://github.com/user-attachments/files/17623325/GivTCPLog.txt)

When you Enable Charge Schedule does it change the status of the Force Charge enable and Force AC Charge Enable? I'm trying to nail down if its not changing the schedule status, or not even changing the underlying registers

britkat1980, 2024-11-05:
I think I've found something, try Dev3.0.39 to see if the charge schedule status works

LeeFarm11, 2024-11-05:
@gcoan

I don't think there is a problem with Enable_Discharge_Schedule. Only a problem with Enable_Charge_Schedule.

I created a helper toggle switch. I called it helper_predbat_enable_charge_schedule.

I then created an automation -
```
alias: predbat_force_charge_automation
description: ""
trigger:
  - platform: state
    entity_id:
      - input_boolean.helper_predbat_enable_charge_schedule
    to: null
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: input_boolean.helper_predbat_enable_charge_schedule
            state: "on"
        sequence:
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
      - conditions:
          - condition: state
            entity_id: input_boolean.helper_predbat_enable_charge_schedule
            state: "off"
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
mode: single
```

I then commented out REST in apps.yaml -
```
  # When set use the REST API rather than HA entity for control, should be more reliable/faster to control
  # Set one per inverter
  # If using Docker then change homeassistant.local to the Docker IP address
  #givtcp_rest:
  #  - 'http://192.168.1.204:6345'
```
and I set this is apps.yaml
```
  scheduled_charge_enable:
  #  - switch.givtcp_{geserial}_enable_charge_schedule
    - input_boolean.helper_predbat_enable_charge_schedule
```

I set Read Only to Off and waited. When the time approached a time to charge (20:00), the times of Charge Slot 1 were set and the enable_charge_schedule triggered correctly - setting both Force Charge Enable and Force AC Charge Enable to On.

I note that the Charge Slot SOC target was not changed (it was previously set at and remains at 4%), but the (global) Charge Up To % was changed by predbat/GivTCP (in this case to 98%). I assume that individual slots SOC targets are therefore not used by predbat, and therefore I have now set Slot 1 SOC to 100%. (I have also set Discharge Slot 1 SOC to 4% making the same assumption).

20:00 arrived. SUCCESS. The battery is charging as expected. I will monitor during the evening to check it turns off at 22:00.

Thanks very much. 

[image]

Then I saw Mark's note about Dev3.0.39, so tomorrow I will undo the changes in apps.yaml and try the Dev version.

gcoan, 2024-11-05:
Brilliant news.

Obviously if we can get a working solution in core GivTCP that works for all inverters, but if not then  this could be a viable workaround and we can document it for others.

On your question r/e charge slots, Predbat only ever uses slot 1 for charging and discharging and sets the global charge-to limit. So you need to make sure that the other slots are all cleared as otherwise the inverter will start obey those other charge/discharge instructions and Predbat won't know about them.  I think I put this in the docs, but maybe I forgot....

tonka-truck, 2024-11-05:
I’ve been watching this with interest and looking promising. 

I’ve updated my 3 phase inverter to DA0.007-DD0.007-A0.007 and not sure yet where the issues are, however I can no longer retrieve data from either HA/GivTCP or the GivEnergy iOS app. The GivEnergy.cloud appears working but haven’t looked further for now.

LeeFarm11, 2024-11-05:
> Brilliant news.
>
> Obviously if we can get a working solution in core GivTCP that works for all inverters, but if not then this could be a viable workaround and we can document it for others.
>
> On your [...]

Yes, potential conflict with the other slots I understood and I get. But I was surprised that the Charge Slot 1 Upper SOC % Limit setting is totally ignored by predbat. Obviously not an issue, just surprised. User needs to ensure that Charge Slot 1 Upper SOC% Limit is set to 100% and presumably also that Discharge Slot 1 Lower SOC % Limit is set close to 0% before engaging predbat.

LeeFarm11, 2024-11-05:
> I’ve been watching this with interest and looking promising.
>
> I’ve updated my 3 phase inverter to DA0.007-DD0.007-A0.007 and not sure yet where the issues are, however I can no longer retrieve data [...]

I also updated to DA0.007-DD0.007-A0.007, I think from DA0.004-DD0.004-A0.004, a few days ago. I had no resulting issue with HA or GivTCP.

gcoan, 2024-11-05:
> But I was surprised that the Charge Slot 1 Upper SOC % Limit setting is totally ignored by predbat. Obviously not an issue, just surprised. User needs to ensure that Charge Slot 1 Upper SOC% Limit [...]

This is a problem of different inverter firmware features.

My Gen 1 hybrid has only 2 discharge slots and 1 charge slot and has no concept of different soc limits for different charge/discharge slots, it simply has a single overall reserve limit (discharge to) and a target to charge to.

The individual soc %'s for each charge slot was only introduced with newer inverter that have more memory, like the Gen 3 hybrid's and the 3 phase inverter.  Not sure about the AIO but I doubt it as its not cropped up before.
Can you confirm what the sensor names are and I will add guidance to the documentation.

LeeFarm11, 2024-11-05:
Here are the settings shown in Portal

[image]
[image]

And here shown in HA GivTCP.
[image]

LeeFarm11, 2024-11-05:
This still with the helper set up as before. Maybe I was wrong about scheduled_discharge enable, but maybe predbat was also not playing properly?
Looks like something was trying to set Battery Mode to Timed Export, rather than trying to setting scheduled discharge enable to True

I tried to force a Discharge by setting a high export rate like this in apps.yaml -
```
  rates_export:
    -  start: "22:30:00"
       end: "22:59:00"
       rate: 25
    -  start: "23:00:00"
       end: "22:29:00"
       rate: 12
```

predbat did then want to export, but only from 22:49 to 23:00. 
It did set the Discharge Slot 1 End time to 23:00, but did not set the Discharge Slot 1 Start Time.
It did try to set scheduled_discharge_enable to True, but it did not succeed. The Discharge did not happen.

From Predbat -
10:45 calculation
```
2024-11-05 22:45:00.819804: --------------- PredBat - update at 2024-11-05 22:45:00+00:00 with clock skew 0 minutes, minutes now 1365
2024-11-05 22:45:00.820932: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2024-11-05 22:45:00.858728: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 22.8 minutes
2024-11-05 22:45:00.858836: Predbat /config/predbat.py version v8.5.9 currently running, latest version is v8.5.9 latest beta v8.5.9
2024-11-05 22:45:00.858904: Autoupdate: Currently up to date
2024-11-05 22:45:00.920088: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-11-05 22:45:00.920937: Predbat mode is set to Control charge & discharge
2024-11-05 22:45:02.150652: Found 5761 load_today datapoints going back 4 days
2024-11-05 22:45:03.173206: Car charging hold True threshold 6.0
2024-11-05 22:45:03.173469: Current data so far today: load 46.2 kWh import 38.8 kWh export 0.0 kWh pv 7.0 kWh
2024-11-05 22:45:03.174820: Adding rate rates_import: {'start': '00:00:00', 'end': '05:00:00', 'rate': 11.84} => 11-05 00:00:00 to 11-05 05:00:00 @ 11.84 date None increment False
2024-11-05 22:45:03.175825: Adding rate rates_import: {'start': '05:00:00', 'end': '13:00:00', 'rate': 26.55} => 11-05 05:00:00 to 11-05 13:00:00 @ 26.55 date None increment False
2024-11-05 22:45:03.177720: Adding rate rates_import: {'start': '13:00:00', 'end': '16:00:00', 'rate': 11.84} => 11-05 13:00:00 to 11-05 16:00:00 @ 11.84 date None increment False
2024-11-05 22:45:03.178642: Adding rate rates_import: {'start': '16:00:00', 'end': '20:00:00', 'rate': 26.55} => 11-05 16:00:00 to 11-05 20:00:00 @ 26.55 date None increment False
2024-11-05 22:45:03.179672: Adding rate rates_import: {'start': '20:00:00', 'end': '22:00:00', 'rate': 11.84} => 11-05 20:00:00 to 11-05 22:00:00 @ 11.84 date None increment False
2024-11-05 22:45:03.180345: Adding rate rates_import: {'start': '22:00:00', 'end': '00:00:00', 'rate': 26.55} => 11-05 22:00:00 to 11-05 00:00:00 @ 26.55 date None increment False
2024-11-05 22:45:03.181887: Adding rate rates_export: {'start': '22:30:00', 'end': '22:59:00', 'rate': 25} => 11-05 22:30:00 to 11-05 22:59:00 @ 25.0 date None increment False
2024-11-05 22:45:03.182173: Adding rate rates_export: {'start': '23:00:00', 'end': '22:29:00', 'rate': 12} => 11-05 23:00:00 to 11-05 22:29:00 @ 12.0 date None increment False
2024-11-05 22:45:03.186093: Standing charge is set to 53.66 p
2024-11-05 22:45:03.243869: Adding rate rates_import_override: {'date': '2024-09-10', 'start': '13:00:00', 'end': '14:00:00', 'rate': 0} => 11-05 13:00:00 to 11-05 14:00:00 @ 0.0 date 2024-09-10 00:00:00 increment False
2024-11-05 22:45:05.647030: Rate min forward looking: now 11.84 at end of forecast 11.84
2024-11-05 22:45:05.647617: Import rates min 11.84 max 26.55 average 20.42
2024-11-05 22:45:05.752389: Export rates min 0 max 25.0 average 12.25
2024-11-05 22:45:05.752506: Rate thresholds (for charge/discharge) are import 26.05p (0.0) export 0.5p (0.0)
2024-11-05 22:45:05.774873: High export rate found rates in range 12.0 to 25.0
2024-11-05 22:45:05.788425: Low Import rate found rates in range 11.84 to 11.84
2024-11-05 22:45:05.983596: Today's energy import 38.8 kWh export 0.0 kWh total 38.8 kWh cost 518.75 p import 518.75 p export 0.0 p carbon 0.0 kg
2024-11-05 22:45:05.983736: Using Solcast integration from inside HA for solar forecast
2024-11-05 22:45:05.983882: PV Data for pv_forecast_today total 10.96 kWh
2024-11-05 22:45:05.984001: PV Data for pv_forecast_tomorrow total 23.75 kWh
2024-11-05 22:45:05.984117: PV Data for pv_forecast_d3 total 14.39 kWh
2024-11-05 22:45:05.984272: PV Data for pv_forecast_d4 total 15.0 kWh
2024-11-05 22:45:06.006049: PV Forecast for today is 10.96 (6.05 10% 18.9 90%) kWh and left today is 0.0 (0.0 10% 0.0 90%) kWh
2024-11-05 22:45:06.061412: PV Forecast for day tomorrow is 23.75 (6.01 10% 50.5 90%) kWh
2024-11-05 22:45:06.091679: PV Forecast for day d2 is 14.39 (4.96 10% 38.72 90%) kWh
2024-11-05 22:45:06.123041: PV Forecast for day d3 is 15.0 (2.94 10% 47.95 90%) kWh
2024-11-05 22:45:06.276420: Historical data totals for days [1, 2, 3] are [30.68, 41.33, 34.7] - min 30.68
2024-11-05 22:45:06.276500: Model filter enabled - Discarding day 1 as it is the lowest of the 3 datapoints
2024-11-05 22:45:06.277559: Historical days now [2, 3] weight [0.75, 0.5]
2024-11-05 22:45:06.385178: Today's load divergence 105.63 % in-day adjustment 105.35 % damping 0.95x
[... cut ...]
2024-11-05 23:00:33,929 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/setBatteryReserve= 4
2024-11-05 23:00:33,930 - mqtt - [INFO] - Requesting Control Action: setBatteryReserve - {'reservePercent': '4'}
2024-11-05 23:00:34,837 - read - [DEBUG] - Write Request recieved
2024-11-05 23:00:34,838 - read - [DEBUG] - Command: setBatteryReserve was recieved: {'reservePercent': '4'}
2024-11-05 23:00:35,335 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-05 23:00:35,336 - read - [DEBUG] - No new writes, removing writerequest file

```

gcoan, 2024-11-06:
> Looks like something was trying to set Battery Mode to Timed Export, rather than trying to setting scheduled discharge enable to True
>
> I tried to force a Discharge by setting a high export rate like [...]

There is some sort of interlinking within GivTCP for the battery mode and the discharge enable switches.  I think you can set the start  and end discharge times and then set the battery mode to Timed Export, or you can set the start and end discharge times and turn the scheduled discharge on.
The controls that Predbat uses are all listed in apps.yaml, and they're the latter approach, battery mode isn't one of the explicit controls that is used by Predbat.

Did you have your rates_export setup via the Octopus integration?  Setting a rates_export in apps.yaml probably won't then be used by Predbat.
If you want to override the Octopus integration rates use rates_export_override, or you can simply use the select.predbat_manual_force_discharge control, set it to the time you want to discharge and Predbat will adjust the plan to do a discharge at that time slot.

LeeFarm11, 2024-11-06:
> Did you have your rates_export setup via the Octopus integration? Setting a rates_export in apps.yaml probably won't then be used by Predbat. If you want to override the Octopus integration rates [...]

No, I do not have Octopus integration set up. I am using rates_import and rates_export. I used that special rate of 25 only to try to force export for testing purposes.

LeeFarm11, 2024-11-06:
> There is some sort of interlinking within GivTCP for the battery mode and the discharge enable switches. I think you can set the start and end discharge times and then set the battery mode to Timed [...]

@britkat1980 @gcoan

This is in apps.yaml -
```
  scheduled_discharge_enable:
    - switch.givtcp_{geserial}_enable_discharge_schedule
```

For Charging, GivTCP did set scheduled charge enable to True. But for Discharging, GivTCP did NOT attempt to set scheduled discharge enable to True. Instead, it tried to set Battery Mode to Timed Export. And it did not work.
Are you able to explain this inconsistency?

Is this inconsistency in GivTCP? Or in predbat? Or something to do with turning off REST in predbat apps.yaml? Or something else?

LeeFarm11, 2024-11-06:
I have now -
Turned off GivTCP
Updated GivTCP Dev to 3.0.39
Changed apps.yaml -
     switched automatic restart from GivTCP to GivTCP Dev
     Turned on REST
     switched back to   
          scheduled_charge_enable:
               - switch.givtcp_{geserial}_enable_charge_schedule
Started GivTCP-Dev

The rates_import and rates_export are -
```
  rates_import:
    -  start: "00:05:00"
       end: "04:59:59"
       rate: 11.84
    -  start: "05:00:00"
       end: "13:04:59"
       rate: 26.55
    -  start: "13:05:00"
       end: "15:59:59"
       rate: 11.84
    -  start: "16:00:00"
       end: "20:04:59"
       rate: 26.55
    -  start: "20:05:00"
       end: "21:59:59"
       rate: 11.84
    -  start: "22:00:00"
       end: "00:04:59"
       rate: 26.55
  #
  rates_export:
    -  rate: 12
    
```

predbat has recalculated successfully. Let's see what happens at 13:05.....
[image]

gcoan, 2024-11-06:
> For Charging, GivTCP did set scheduled charge enable to True. But for Discharging, GivTCP did NOT attempt to set scheduled discharge enable to True. Instead, it tried to set Battery Mode to Timed [...]

For charging and discharging, Predbat is setting charge/discharge start time, end time and then turning enable scheduled charge/discharge on.

For charging as noted earlier, Enable Scheduled Charge doesn't appear for your 3 phase inverter so you've had to add a dummy switch and then code an automation to turn on Force charge enable and Force AC charge enable in GivTCP (as an aside, just turning on Force charge enable ought to be sufficient, but anyway).

For discharging when you turn on scheduled discharge enable the battery mode gets changed from Eco to Timed Export.  This is entirely consistent with older inverters.
There isn't a battery mode of 'charging' (or similar), the battery is in Eco mode and can be force charging as well.  It's inconsistent but it's the way the inverter firmware works (at least on my Gen 1).

> Instead, it tried to set Battery Mode to Timed Export. And it did not work.

'it did not work' is a GivTCP problem, sounds like Predbat was giving the right instructions to start the discharge (and export excess above house load)

> Or something to do with turning off REST in predbat apps.yaml?

No shouldn't be, Predbat has two ways of communicating to GivTCP, via the HA controls or via REST.  We turned off the REST control option so we could override the REST control to set charge enable which Predbat would otherwise be trying to do directly to GivTCP (and as noted earlier wouldn't have worked).

LeeFarm11, 2024-11-06:
> 
> (as an aside, just turning on Force charge enable ought to be sufficient, but anyway).

I have now set predbat to Read Only (will explain later). 

I am currently charging using the Portal settings with Enable AC Charge and Enable Force Charge both On. Both relevant Charge Up to limits are set to 100%. SOC currently 50%. 19kw coming from Grid. (Storage Heaters on plus charging battery at 11kw). When I turn off Enable AC Charge to Off (leaving Enable Force Charge On), the battery stops charging, grid only supplies to match load. 

If I swap it round - Enable AC Charge to On and Enable Force Charge to Off, then battery discharges to match load. Nothing (0) is imported from Grid.

If I then Set Enable Force Charge back to on, then I start to import 19kW again.

So I do need to set both settings to On to make it charge the battery from the Grid.

LeeFarm11, 2024-11-06:
> I have now - Turned off GivTCP Updated GivTCP Dev to 3.0.39 Changed apps.yaml - switched automatic restart from GivTCP to GivTCP Dev Turned on REST switched back to scheduled_charge_enable: - [...]

Unfortunately Charging did not happen.

Here is the 12:35 predbat plan
```
2024-11-06 12:35:00.307916: --------------- PredBat - update at 2024-11-06 12:35:00+00:00 with clock skew 0 minutes, minutes now 755
2024-11-06 12:35:00.309009: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2024-11-06 12:35:00.354754: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 47.1 minutes
2024-11-06 12:35:00.354876: Predbat /config/predbat.py version v8.5.9 currently running, latest version is v8.5.9 latest beta v8.5.9
2024-11-06 12:35:00.354954: Autoupdate: Currently up to date
2024-11-06 12:35:00.416402: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-11-06 12:35:00.417338: Predbat mode is set to Control charge & discharge
2024-11-06 12:35:01.748954: Found 5761 load_today datapoints going back 4 days
2024-11-06 12:35:02.773465: Car charging hold True threshold 6.0
2024-11-06 12:35:02.773579: Current data so far today: load 24.9 kWh import 13.3 kWh export 0.0 kWh pv 5.2 kWh
2024-11-06 12:35:02.774926: Adding rate rates_import: {'start': '00:05:00', 'end': '04:59:59', 'rate': 11.84} => 11-06 00:05:00 to 11-06 04:59:00 @ 11.84 date None increment False
2024-11-06 12:35:02.775937: Adding rate rates_import: {'start': '05:00:00', 'end': '13:04:59', 'rate': 26.55} => 11-06 05:00:00 to 11-06 13:04:00 @ 26.55 date None increment False
2024-11-06 12:35:02.777538: Adding rate rates_import: {'start': '13:05:00', 'end': '15:59:59', 'rate': 11.84} => 11-06 13:05:00 to 11-06 15:59:00 @ 11.84 date None increment False
2024-11-06 12:35:02.778200: Adding rate rates_import: {'start': '16:00:00', 'end': '20:04:59', 'rate': 26.55} => 11-06 16:00:00 to 11-06 20:04:00 @ 26.55 date None increment False
2024-11-06 12:35:02.779108: Adding rate rates_import: {'start': '20:05:00', 'end': '21:59:59', 'rate': 11.84} => 11-06 20:05:00 to 11-06 21:59:00 @ 11.84 date None increment False
2024-11-06 12:35:02.779844: Adding rate rates_import: {'start': '22:00:00', 'end': '00:04:59', 'rate': 26.55} => 11-06 22:00:00 to 11-06 00:04:00 @ 26.55 date None increment False
2024-11-06 12:35:02.781629: Adding rate rates_export: {'rate': 12} => 11-06 00:00:00 to 11-06 00:00:00 @ 12.0 date None increment False
2024-11-06 12:35:02.785852: Standing charge is set to 53.66 p
2024-11-06 12:35:02.846332: Adding rate rates_import_override: {'date': '2024-09-10', 'start': '13:00:00', 'end': '14:00:00', 'rate': 0} => 11-06 13:00:00 to 11-06 14:00:00 @ 0.0 date 2024-09-10 00:00:00 increment False
2024-11-06 12:35:05.461374: Rate min forward looking: now 0 at end of forecast 0.0
2024-11-06 12:35:05.461973: Import rates min 0 max 26.55 average 20.49
2024-11-06 12:35:05.532760: Export rates min 12.0 max 12.0 average 12.0
2024-11-06 12:35:05.532883: Rate thresholds (for charge/discharge) are import 26.05p (0.0) export 12.0p (0.0)
2024-11-06 12:35:05.552268: High export rate found rates in range 12.0 to 12.0
2024-11-06 12:35:05.563793: Low Import rate found rates in range 11.84 to 11.84
2024-11-06 12:35:05.742165: Today's energy import 13.3 kWh export 0.0 kWh total 13.3 kWh cost 220.22 p import 220.22 p export 0.0 p carbon 0.0 kg
2024-11-06 12:35:05.742417: Using Solcast integration from inside HA for solar forecast
2024-11-06 12:35:05.742593: PV Data for pv_forecast_today total 12.07 kWh
2024-11-06 12:35:05.742719: PV Data for pv_forecast_tomorrow total 11.72 kWh
2024-11-06 12:35:05.742858: PV Data for pv_forecast_d3 total 18.04 kWh
2024-11-06 12:35:05.742973: PV Data for pv_forecast_d4 total 13.89 kWh
2024-11-06 12:35:05.766168: PV Forecast for today is 12.07 (3.46 10% 27.4 90%) kWh and left today is 4.23 (0.85 10% 8.82 90%) kWh
2024-11-06 12:35:05.829857: PV Forecast for day tomorrow is 11.72 (3.34 10% 30.57 90%) kWh
2024-11-06 12:35:05.861285: PV Forecast for day d2 is 18.04 (3.75 10% 48.98 90%) kWh
2024-11-06 12:35:05.891121: PV Forecast for day d3 is 13.89 (3.07 10% 45.2 90%) kWh
2024-11-06 12:35:06.056028: Historical data totals for days [1, 2, 3] are [30.39, 39.15, 41.27] - min 30.39
2024-11-06 12:35:06.056245: Model filter enabled - Discarding day 1 as it is the lowest of the 3 datapoints
2024-11-06 12:35:06.057325: Historical days now [2, 3] weight [0.75, 0.5]
2024-11-06 12:35:06.154765: Today's load divergence 99.98 % in-day adjustment 99.98 % damping 0.95x
2024-11-06 12:35:06.154869: Today's predicted so far 17.23 kWh with 4.28 kWh car/iBoost excluded and 2.6 kWh import ignored and 0 forecast extra.
2024-11-06 12:35:06.154918: Today's actual load so far 17.22 kWh with 5.4 kWh Car/iBoost excluded and 2.21 kWh import ignored.
2024-11-06 12:35:06.347245: Inverter 0: Type GE GivEnergy
2024-11-06 12:35:06.347396: Inverter 0 using Rest API http://192.168.1.204:6345
2024-11-06 12:35:06.394507: Inverter 0 does not have timed pause support enabled
2024-11-06 12:35:06.395136: Invertor time 2024-11-06 12:34:45+00:00, Predbat computer time 2024-11-06 12:35:06.395040+00:00, difference -0.37 minutes
2024-11-06 12:35:06.395226: Reserve min: 4.0% Battery_min:4.0%
2024-11-06 12:35:06.395342: Inverter 0 with soc_max 21.11 kWh nominal_capacity 21.11 kWh battery rate raw 11000.0 w charge rate 11.0 kW discharge rate 11.0 kW battery_rate_min 200.0 w ac limit 11.5 kW [... cut ...]
2024-11-06 12:40:44,851 - REST - [DEBUG] - found REST response
2024-11-06 12:40:45,088 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 12:40:45,089 - read - [DEBUG] - No new writes, removing writerequest file
```
I wonder if the problem might be because I re-enabled REST in apps.yaml. I have now disabled REST in apps.yaml and turned off Read Only.

@britkat1980 Or do you think the error with enableChargeSchedule is something else?

Next charging session is due at 20:05. Let's see what happens.

[image]

LeeFarm11, 2024-11-06:
> I think I've found something, try Dev3.0.39 to see if the charge schedule status works

Hi @britkat1980 
I did install Dev3.0.39 and tried with REST enabled in predbat, and got the errors above. Now I try with REST disabled. But...

After I made the most recent changes, I see that I had left both Force AC Charge Enable and Force Charge Enable both On. predbat tried to turn enableChargeSchedule off because it was a long time before next charge event. It first tried at 18:20

```
2024-11-06 18:20:12,522 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= disable
2024-11-06 18:20:12,523 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2024-11-06 18:20:13,276 - read - [DEBUG] - Write Request recieved
2024-11-06 18:20:13,277 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'disable'}
2024-11-06 18:20:13,582 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 18:20:13,583 - read - [DEBUG] - No new writes, removing writerequest file

2024-11-06 18:20:22,661 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= disable
2024-11-06 18:20:22,663 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2024-11-06 18:20:23,407 - read - [DEBUG] - Write Request recieved
2024-11-06 18:20:23,408 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'disable'}
2024-11-06 18:20:23,711 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 18:20:23,711 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 18:20:32,715 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= disable
2024-11-06 18:20:32,715 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2024-11-06 18:20:32,748 - read - [DEBUG] - Write Request recieved
2024-11-06 18:20:32,749 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'disable'}
2024-11-06 18:20:33,056 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 18:20:33,057 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 18:20:42,761 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= disable
2024-11-06 18:20:42,761 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2024-11-06 18:20:43,088 - read - [DEBUG] - Write Request recieved
2024-11-06 18:20:43,088 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'disable'}
2024-11-06 18:20:43,396 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 18:20:43,397 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 18:20:52,434 - read - [DEBUG] - Running partial refresh
2024-11-06 18:20:52,814 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= disable
2024-11-06 18:20:52,814 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2024-11-06 18:20:56,613 - read - [DEBUG] - Data get was successful, now running handler if needed: 

2024-11-06 18:20:57,001 - read - [DEBUG] - Write Request recieved
2024-11-06 18:20:57,003 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'disable'}
2024-11-06 18:20:57,313 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 18:20:57,313 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 18:21:02,882 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= disable
2024-11-06 18:21:02,883 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2024-11-06 18:21:03,338 - read - [DEBUG] - Write Request recieved
2024-11-06 18:21:03,338 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'disable'}
2024-11-06 18:21:03,640 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 18:21:03,641 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 18:21:12,976 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= disable
2024-11-06 18:21:12,977 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2024-11-06 18:21:13,679 - read - [DEBUG] - Write Request recieved
2024-11-06 18:21:13,683 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'disable'}
2024-11-06 18:21:13,990 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 18:21:13,991 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 18:21:23,028 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= disable
2024-11-06 18:21:23,029 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2024-11-06 18:21:24,031 - read - [DEBUG] - Write Request recieved
2024-11-06 18:21:24,032 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'disable'}
2024-11-06 18:21:24,334 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 18:21:24,335 - read - [DEBUG] - No new writes, removing writerequest file
```

So it seems that the fix in Dev3.0.39 does not work. Well, not for turning enableChargeSchedule Off. I have turned both switched now to Off. Let's see if it can turn On after 20:00.

LeeFarm11, 2024-11-06:
@britkat1980 
Well, it seems Dev3.0.39 also does not succeed to turn on enableChargeSchedule either. Even with REST disabled.

```
2024-11-06 19:47:59,047 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= enable
2024-11-06 19:47:59,048 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'enable'}
2024-11-06 19:47:59,433 - read - [DEBUG] - Write Request recieved
2024-11-06 19:47:59,434 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'enable'}
2024-11-06 19:47:59,736 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 19:47:59,736 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 19:48:09,120 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= enable
2024-11-06 19:48:09,123 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'enable'}
2024-11-06 19:48:09,774 - read - [DEBUG] - Write Request recieved
2024-11-06 19:48:09,775 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'enable'}
2024-11-06 19:48:10,077 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 19:48:10,077 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 19:48:19,180 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= enable
2024-11-06 19:48:19,180 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'enable'}
2024-11-06 19:48:20,108 - read - [DEBUG] - Write Request recieved
2024-11-06 19:48:20,108 - read - [DEBUG] - Command: enableChargeSchedule was recieved: {'state': 'enable'}
2024-11-06 19:48:20,417 - read - [DEBUG] - Write Commands lengths: 1 -> 1
2024-11-06 19:48:20,418 - read - [DEBUG] - No new writes, removing writerequest file
2024-11-06 19:48:28,444 - read - [DEBUG] - Running partial refresh
2024-11-06 19:48:29,240 - mqtt - [DEBUG] - MQTT Message Recieved: GivEnergy/control/TD2343G049/enableChargeSchedule= enable
2024-11-06 19:48:29,240 - mqtt - [DEBUG] - Requesting Control Action: enableChargeSchedule - {'state': 'enable'}
```

LeeFarm11, 2024-11-06:
I've gone back to using the helper and automation and 'redirect' in apps.yaml for enable charge schedule. Charging now working again (still on Dev3.0.39).

Now trying to set up helper and automation for Inverter Mode which does not appear to be available on the 3 phase Inverter. There appear to be 4 Modes in GivTCP, and I do not know what the differences are. Can anyone explain please? -

Eco, Eco (paused), Timed Export, Timed_Demand

Timed Export I guess requires Force AC Charge Enable to be On and Force Charge Enable to be On.
Timed Demand I guess requires Force Discharge Enable to be On.
Eco I guess would be all 3 switches off
But what is Eco (paused)?

Any additional clues please? Thanks.

LeeFarm11, 2024-11-06:
> For charging and discharging, Predbat is setting charge/discharge start time, end time and then turning enable scheduled charge/discharge on.

Hi @gcoan 
The problem last night with Discharging was that predbat did NOT attempt to turn enable scheduled discharge on. INSTEAD predbat decided to set Inverter Mode to Timed Export. That is what is shown in the predbat log. And then GivTCP failed because I don't think the Inverter has the setting.

So now I have created another helper called input_select.helper_predabt_Inverter_Mode and I have created another automation -

```
alias: predbat_inverter_mode_automation
description: ""
trigger:
  - platform: state
    entity_id:
      - input_select.helper_predbat_inverter_mode
    to: null
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: input_select.helper_predbat_inverter_mode
            state: Eco
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
      - conditions:
          - condition: state
            entity_id: input_select.helper_predbat_inverter_mode
            state: Eco (Paused)
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
      - conditions:
          - condition: state
            entity_id: input_select.helper_predbat_inverter_mode
            state: Timed Demand
        sequence:
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
      - conditions:
          - condition: state
            entity_id: input_select.helper_predbat_inverter_mode
            state: Timed Export
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
mode: single
```

I have then updated apps.yaml
```
  inverter_mode:
  #  - select.givtcp_{geserial}_mode
    - input_select.helper_predbat_Inverter_Mode
```

I have set up a similar test like last night by setting a high export rate in the near future. I will report back with results.

LeeFarm11, 2024-11-06:
That seemed to work. predbat threw in a set Inverter Mode to Timed Export and then later to Eco, both of which were handled by the automation and GivTCP successfuly. It did also throw in a set enable scheduled discharge to false at the end of the discharge but does not seem to have been actioned but it was not needed. Strange.

Anyway, now to see what it does with these Hold Charges and Freeze Charges overnight.

[image]

gcoan, 2024-11-07:
Hi Lee,

My mistake, I thought Predbat doesn't directly set the battery mode, but looking again at the template apps.yaml I can see that it does have the battery mode control defined for non-rest communication.

There are in fact one additional mode you hadn't identified, Unknown, but I guess this is something the inverter can report and isn't something that Predbat sets directly so no need for it to be implemented by your automation.

You could try checking the inverter.py Predbat code to see what modes it sets and under what circumstances, but my expectation is that it only sets Eco and Timed Export.  Eco for the default (Idle mode in the predbat plan), battery charging off solar and supplying house load as necessary. No forced grid charge or discharge.

Eco (Paused) is a weird one, I have seen it set on my inverter sometimes and I think it corresponds to one of the Predbat battery hold modes but am not sure which.
Timed Demand is similar to Eco, battery supporting house load but any solar is I think exported.

It's a pain that the 3 phase doesn't have similar modbus controls to the other inverter ranges and you're having to emulate the missing modes for Predbat to operate, but looks like we are getting there

LeeFarm11, 2024-11-07:
Overnight all seemed to work OK. I don't yet know why we had a dip/spike around 00:30 but I'll try to take a look at the logs later. Seems good for now.

@gcoan 
I don't yet know how to get to the code in a .py file. I'll google it again. Thanks.

[image]

gcoan, 2024-11-07:
> I don't yet know how to get to the code in a .py file. I'll google it again. Thanks.
> 
> 

Easiest thing is if you just look at the code in github: https://github.com/springfall2008/batpred/tree/main/apps/predbat

not expecting you to be a python expert overnight but its very 'basic-like' a lot of the logic.

inverter.py is the module that talks to the inverter

britkat1980, 2024-11-08:
I'm trying to disaggregate any predbat issues with GivTCP issues. Can you confirm whether Enable Charge Schedule works as expected when turning it on/off directly using HA?

LeeFarm11, 2024-11-08:
> Enable Charge Schedule does not work. As previously discussed, on the Phase 3 Inverter it seems this setting does not exist, but it is needed for Predbat. I think it needs to switch both Force [...]

Hi @britkat1980 
Enable Charge Schedule does not work in Dev3.0.39 and does not work in Prod 3.0.4

LeeFarm11, 2024-11-08:
> Looking into the graphs a bit more, this seems very weird. The GivTCP Power Load Phase 1 Power, GivTCP Power Load Phase 2 Power, GivTCP Power Load Phase 3 Power are showing really odd results. [...]

Hi @britkat1980 
And I think this is a GivTCP issue.

LeeFarm11, 2024-11-09:
Hi @britkat1980 
I now moved the discussion of predbat issues over to the predbat area. 

The GivTCP Power issue I guess can stay here.

The issue with Enable Charge Schedule, for me, I think also stays here. To the best of my current knowledge, I do have (temporary) solution by using my helper and automation. However, I do wonder have GivEnergy given you any useful information about Force Charge Enable and Force AC Charge Enable? Or if they are able to give you any useful information? A few additional clues about what they are supposed to do would be very handy.

britkat1980, 2024-11-09:
What is confusing me is that if the individual controls for the two charge enable devices work, why doesn’t the schedule one work when all it does is exactly the same thing…

LeeFarm11, 2024-11-09:
Sorry I cannot answer that. Yes, the 2 GivTCP individual force charge controls work fine, but the enable schedule charge button achieves nothing.

LeeFarm11, 2024-11-10:
Hold that. Haven't time to look closer, but enable charge schedule is currently on. Will take a closer look later and let you know.

britkat1980, 2024-11-10:
Can you check two things, one is, does Enable Schedule “read” correctly based on the status of the other enable controls (GivTCP generates a schedule status based on the other two controls.
Second is if switching the schedule changes the other two controls.
They are different functions in GivTCP because it’s not a real register.

LeeFarm11, 2024-11-10:
Hi. Yes. It reads correctly. But it does not write.

If both Force AC Charge Enble and Force Charge Enable are on, then Enable Charge Schedule switches on.
Then switching Enable Charge Schedule to Off does nothing.

Switching either of Force AC Charge Enable and Force Charge Enable are switched Off, then Enable Charge Schedule switches Off.
With both switched Off, then switching Enable Charge Schedule to On does nothing.

In short, the switch does display On when the other 2 switches are On. But has no other visible functionality.

Do you have any information from GivEnergy about what those 2 switches are supposed to do individually?

britkat1980, 2024-11-12:
Can you try Dev3.0.41, possible error in prepping commands for enable schedule. Let me know if it makes a difference

LeeFarm11, 2024-11-12:
Dev3.0.41 Enable Charge Schedule now works as expected. Thanks.

At least until we find out what Force Charge and Force AC Charge are supposed to do individually.....

LeeFarm11, 2024-11-13:
> Looking into the graphs a bit more, this seems very weird. The GivTCP Power Load Phase 1 Power, GivTCP Power Load Phase 2 Power, GivTCP Power Load Phase 3 Power are showing really odd results. [...]

Any ideas about these strange graphs of Load by Phase?

britkat1980, 2024-11-15:
It looks like there is spurious data coming in from the inverter, which there is code to catch, but perhaps it manifests differently in the Three Phase. I need to see a recording of the raw register data over a time period to check if my validation method works or not. Perhaps running a script to grab rest data periodically?

LeeFarm11, 2024-11-15:
Do you have such a script which I could modify and use?

LeeFarm11, 2024-11-22:
Hi @britkat1980 
I found a setting which predbat uses, but which is not available in GivTCP. Is it possible to add?

When predbat is in state of 'Hold charging' it is trying to stop the SOC dropping below a certain point. 

The item outlined in Red is currently used by predbat. This is mapped to GivTCP Control Battery Power Reserve.
In the REST data, I believe it is 
Control: Battery Power Reserve
and
Invertor: battery_soc_reserve

Unfortuantely it does not do what predbat expects it to do, which is to prevent the battery SOC dropping below this point while in 'Idle' or 'Demand' without setting any time slot start and end times.

From memory I think this setting can be used when Force Exporting when timslot details are also set. But this is not relevant in this case.

The item which predbat needs to use in this case (Hold charging) is the item marked in Black.
In the REST data it is 
Invertor: load_first_stop_soc

[image]

Adding @gcoan for info.

britkat1980, 2024-11-22:
I've modified dev3.0.45 to use the same entity but point to the right register for Three phase. Can you test and let me know if it works (read and write correctly)

LeeFarm11, 2024-11-22:
Hi. Installed Dev3.0.45

Sorry. GivTCP Control Battery Power Reserve is still pointing to the item outlined in Red.

And now 
Invertor: load_first_stop_soc
has disappeared from GivTCP REST

And I can no longer find anything in GivTCP REST which correlates to the item outlined in Black.

britkat1980, 2024-11-22:
That’s odd, because I changed the register for battery_power_reserve to point to reg 1078, which was load_first_stop. So I don’t know why it’s still matching the red item…
If you change the black item in the portal what entity (if any) changes in givtcp

LeeFarm11, 2024-11-22:
Nothing I'm afraid. Nothing in Home Assistant Overview. And nothing in GivTCP REST.

LeeFarm11, 2024-12-01:
Hi @britkat1980
New settings have appeared inGiv Portal. In the Battety Control section.
An option for clearing the Scheduled Battery Calibration register, and an option for Winter Battery Conditioning (whatever that is).

[image]

gcoan, 2024-12-01:
Winter battery conditioning has appeared as a toggle option in the givenergy app.  If turned on the it will do a 10 minute charge of the battery if the battery température drops below 2 degrees to preserve battery cell life

[7 more replies]

closed 2026-10-11 by britkat1980