britkat1980/giv_tcp #288: Multiple Networks - One Default Gateway

AberDino, 2024-10-25:
My Home Assistant setup has two networks (IPv4 only - IPv6 disabled). One has a default gateway set, and the other one doesn't, because it's only for communicating with local devices including the GivEnergy kit and quite a few smart plugs (effectively my IoT network). I used [these instructions](https://community.home-assistant.io/t/setup-vlan-and-ha-tutorial/87705/111) to set up the second (VLAN) connection, including the "ipv4.never-default" option to avoid traffic routing out of this interface. Within Home Assistant (Settings -> System -> Network), the gateway address for this second interface address appears as "null".

I recently upgraded to GivTCP v3, and it won't start because it appears that it doesn't like the fact that the second network interface does not have a gateway address. Log entry:
`2024-10-25 12:30:23,092 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
Traceback (most recent call last):
  File "/app/startup.py", line 389, in <module>
    networks[i]=interface['ipv4']['gateway']+"/"+str(mask)
                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'`

If I enter a valid gateway address for the second network, GivTCP starts up without any issues, but changing the routing metrics does not appear to be working 100%, as in that scenario I do see Home Assistant traffic trying to route out through the IoT VLAN (which is then blocked on the firewall because only some devices on the IoT network are allowed to route out, e.g. the GivEnergy devices). Therefore, my preference would be to get GivTCP working without the gateway address for any additional VLAN(s). Would that be possible?

britkat1980, 2024-10-27:
Sure, I'll have to make assumptions about the subnet mask (/24) and use the ip address found to scan

AberDino, 2024-10-27:
That's great, thank you in advance!

I'm not a programmer, but hopefully you should be able to find the information about the various networks by using the equivalent of "_ip addr show_", The information returned includes the IP address and subnet mask, so you wouldn't have to make assumptions. However, as the command returns information from all the interfaces, including loopback, docker0, hassio, etc., it might be a challenge to filter out only the details which are relevant. Let me know if I can help with anything.

britkat1980, 2024-11-08:
Can you test Dev3.0.40 and let me know if it solves the issue

AberDino, 2024-11-08:
Thanks @britkat1980, I'll test that over the weekend.

In terms of the deployment steps, can you confirm I should do the following:
1. Stop and uninstall GivTCP
2. Remove gateway from second (VLAN) connection
3. Reboot
4. Install GivTCP-DEV

britkat1980, 2024-11-15:
I've found the issue with that error, fixed in Dev3.0.44. Can you test

AberDino, 2024-11-17:
Hi @britkat1980, different error(s) this time, see new log: [533ea71a_givtcp-dev_2024-11-17T20-34-10.149Z.log](https://github.com/user-attachments/files/17792330/533ea71a_givtcp-dev_2024-11-17T20-34-10.149Z.log)

I simply stopped the 'regular' GivTCP v3.0.4 an started the latest GivTCP-DEV, i.e. I did not make any changes to the network settings, as I thought it's best test with a known good network configuration first.

AberDino, 2024-11-30:
Hi @britkat1980, just wondering if I should test Dev3.0.45, or does it contain different fixes? Thanks

AberDino, 2025-03-07:
Hi @britkat1980, just to let you know I remain keen to assist you resolve this issue (i.e. be able to run multiple HA VLANs but with a default gateway defined on the main VLAN only). This morning I tried GivTCP-DEV 3.0.53 without changing my network settings and the result is attached. Issue seems to be associated with the GivGateway: 

> cannot access local variable 'power_flow_output' where it is not associated with a value

[533ea71a_givtcp-dev_2025-03-07T10-10-02.660Z.log](https://github.com/user-attachments/files/19124687/533ea71a_givtcp-dev_2025-03-07T10-10-02.660Z.log)

AberDino, 2025-04-14:
Confirmed fixed in GivTCP 3.1.6, thanks @britkat1980!

closed 2025-04-14 by AberDino