britkat1980/giv_tcp #290: Keeps rebooting home assistant

Piccyman, 2024-11-02:
Anytime I look at the config or anything with this add on, even stopping it, it causes home assistant to crash and reboot

gcoan, 2024-11-02:
> Anytime I look at the config or anything with this add on, even stopping it, it causes home assistant to crash and reboot

What are you running Home Assistant on, is it something like a Pi 3?

Piccyman, 2024-11-03:
Yes a pi3, does the add on require more power?

gcoan, 2024-11-04:
> Yes a pi3, does the add on require more power?

The add-on itself doesn't have any specific power requirements but the recommended minimum hardware for Home Assistant is a Pi 4 (or at a squeeze a Pi 3B): https://www.home-assistant.io/installation/raspberrypi/ 

You're probably just trying to run too much on your Pi

Piccyman, 2024-11-04:
Ok, thank you, I will get a pi 4

closed 2024-11-08 by britkat1980