britkat1980/giv_tcp #292: Spurious values reported for battery charge/discharge energy totals (reopen)

shuntfield, 2024-11-04:
https://github.com/britkat1980/giv_tcp/issues/265

I'm running 3.0.4 and its still doing the same thing for me.

[image]

A restart of the add on cures it as before.

britkat1980, 2024-11-08:
Hmmm, can you try the latest dev version and let me know if its still happening?

shuntfield, 2024-11-08:
> Hmmm, can you try the latest dev version and let me know if its still happening?

I'll figure out how to install the Latest Dev and report back!

britkat1980, 2024-11-15:
I'm assuming no news is good news. Will close issue

shuntfield, 2024-11-15:
Haven't noticed the problem on Dev so looks good 👍🏻 
Thanks!

shuntfield, 2024-11-25:
Unfortuantely its back, it did it again today - showed battery discharged 2400+kWh

> I'm assuming no news is good news. Will close issue

chibbert, 2025-06-20:
I'm seeing this too, wierdly for me, it is only the discharge energy: 
[image]

It always seems to go to the same strange high value: 2261.6