britkat1980/giv_tcp #294: 3.0.4 errors writing to Influx

cdrajb, 2024-11-05:
3.0.4 seems happy to publish the tempoutput dict to mqtt, but gets KeyError when it publishes it to influx

/config/GivTCP/logs # tail log_inv_1.log
2024-11-05 16:11:38,568 - read - [INFO] - Starting watch_plant loop...
2024-11-05 16:11:38,587 - read - [CRITICAL] - Detecting inverter characteristics...
2024-11-05 16:11:45,956 - read - [INFO] - Dynamic Tariff enabled so defaulting to Day Rate and waiting external control
2024-11-05 16:11:47,004 - read - [ERROR] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
/config/GivTCP/logs #

The above log is from the 3.0.4 docker image without modifications.  I enabled debug to see where the error was actually happening; if I log the KeyError it reports the key as 'Battery_to_Grid' - which is definitely in the dict, so I'm a bit stumped!

pod-la, 2025-01-04:
I'm seeing the same issue using HA addon. 

My setup has InfluxDB on a separate server connected via VLAN 1 on IP 192.168.1.250 and the invertor connected via VLAN 101 on 10.101.0.11. This works fine on v 2.4.9 but not on 3.0.4. Same errors as above in the logs. A tcpdump shows no HTTP POSTs with the bucket used for GivTCP, just the POSTs from HA itself.

```
2025-01-04 15:16:03,699 - read - [ERROR] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-04 15:16:34,910 - read - [ERROR] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-04 15:17:06,132 - read - [ERROR] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-04 15:17:38,610 - read - [ERROR] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-04 15:18:09,819 - read - [ERROR] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
```

CannyGent, 2025-03-04:
Yes - same issue here with influx on 3.0.4... if I turn influx data off then no error

closed 2025-10-23 by britkat1980