britkat1980/giv_tcp #296: Timeslot controls for Export

danmatley, 2024-11-09:
Can I check I'm not missing some controls, but is there a way to change the timeslot schedule for exporting to the grid? (separately to discharge to house)

gcoan, 2024-11-09:
> Can I check I'm not missing some controls, but is there a way to change the timeslot schedule for exporting to the grid? (separately to discharge to house)

If your inverter battery mode is set to 'Eco' then the inverter will discharge to the home to support home demand.  There isn't a concept of timed schedule to discharge to the home.

To export to the grid there are start discharge time and end discharge time controls in GivTCP and you set the battery mode to 'Timed Export'.

danmatley, 2024-11-09:
I see, the GivEnergy app and portal schedule controls distinguishes the difference between "Timed Discharge" (to cover your home's demand) and "Timed Export" (discharges as max power and the excess goes to the grid).

Perhaps this is just the way their interface manage it, but GivTCP just gives us the native controls via the discharge timeslots and Battery Mode as you've stated. And I would use off Eco mode if I want full manual control?

gcoan, 2024-11-09:
Yes correct.  GivTCP gives you the raw inverter controls as exposed by modbus.  The GivEnergy app and portal abstracts these into more easier to understand constructs

closed 2024-11-15 by britkat1980