britkat1980/giv_tcp #300: REST setBatteryPauseSlot and setBatteryPauseMode causes failure

springfall2008, 2024-11-10:
I tried calling rest setBatteryPauseSlot and then setBatteryPauseMode and it appeared to crash GivTCP:

2024-11-10 20:40:39,361 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-11-10 20:40:39,611 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-11-10 20:40:39,619 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 610)
2024-11-10 20:40:55,102 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-11-10 20:40:55,353 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-11-10 20:40:55,360 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 610)
2024-11-10 20:41:10,975 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:41:11,087 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:41:11,100 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:41:11,107 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:41:11,115 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:41:12,146 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)

springfall2008, 2024-11-10:
eventually:

2024-11-10 20:46:17,121 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:17,126 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:18,193 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:18,200 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:18,207 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:18,213 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:18,417 - startup     -  [ERROR   ] - Self Run loop process stuck. Killing and restarting...
2024-11-10 20:46:18,418 - startup     -  [INFO    ] - Restarting Invertor read loop every 30s
2024-11-10 20:46:18,831 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2024-11-10 20:46:18,832 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.0.20
2024-11-10 20:46:18,837 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2024-11-10 20:46:30,081 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:30,091 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:30,098 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:30,104 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:30,110 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:31,121 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:31,127 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:31,132 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:31,138 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)
2024-11-10 20:46:32,209 - GivTCP - read        -  [ERROR   ] - Error in Watch Loop: ('KeyError', 'read.py', 124)

springfall2008, 2024-11-11:
Okay so the two issues are:

1. I used the wrong key in the setBatteryPauseMode (mode rather than state) and this caused the infinite loop error.
2. The pause time start/end doesn't work as my inverter doesn't have these registers, but the question is how can I know this from GivTCP?

britkat1980, 2024-11-15:
I've put in a catch for the infinite error loop from Dev3.0.42 and Prod3.1.0.
I'll look at not making unavailable controls not available in GivTCP or at least making them discoverable

britkat1980, 2024-11-15:
One option to see if the control is available is to read the "Control" and "Timeslot" sections of the read cache and see if the control is listed there?

springfall2008, 2024-11-24:
In the REST data I get:

    "Timeslots": {
        "Battery_pause_end_time_slot": "00:00:00",
        "Battery_pause_start_time_slot": "00:00:00",
        "Charge_end_time_slot_1": "05:00:00",
        "Charge_start_time_slot_1": "23:00:00",
        "Discharge_end_time_slot_1": "00:00:00",
        "Discharge_end_time_slot_2": "00:00:00",
        "Discharge_start_time_slot_1": "00:00:00",
        "Discharge_start_time_slot_2": "00:00:00"
    },

It gives the Discharge time slots even though they are not present?