britkat1980/giv_tcp #303: BMS temperature = 0.0 AC3

LeeFarm11, 2024-11-12:
Hi. In section GivTCP Battery Stack 1 I have BMS temperature showing 0.0 degrees C.

Dev3.0.39

[image]

[image]

britkat1980, 2024-11-15:
What value does the Portal show? It might be that the 3PH uses a different value for the overall Battery (BMS) temp

britkat1980, 2024-11-15:
Not sure, perhaps it doesn’t report it. I can make GivTCP use an average of the cell temps as a proxy?

LeeFarm11, 2024-11-15:
Yes, makes sense. One less spurious reading....

britkat1980, 2024-11-22:
I've changed BMS_Temperature to be average of cell temps

britkat1980, 2024-11-24:
> I'm still running 2.4.9 but BMS temp is often very different to cell temp, it's much higher when charging and discharging for example, so imo I don't think average cell temp would be good for BMS temp.
> 
> Also with 2.4.9 BMS temp often freezes value when charging or discharging. Restarting GivTCP fixes it or it often fixes itself, with the frozen value being replaced with the missing data, sometime after the charge / discharge has completed. Has this been fixed in V3?

The above refers to Three Phase inverter, which doesn’t have a BMS temp available.

re: the freezing, v3 has a different data smoothing architecture so it should be better.