britkat1980/giv_tcp #304: Can't set charge/discharge rate high enough

cigriggs, 2024-11-14:
I have a Gen3 5.0kW Hybrid Invertor, with a Gen3 5.12kWh battery.

In the GivEnergy app and portal I can set the max charge and discharge rate to 3256W, but with GivTCP the home assistant controls max out at 2715W.

The screenshots show the state of the app, portal, and home assistant when I've set the Discharge Power to 3256W in the app, and the Battery Charge Rete to 2715W in home assistant.

[image]
[image]
[image]

The following is the output from readData REST endpoint:

[readData.json](https://github.com/user-attachments/files/17748303/readData.json)

I note that raw.invertor.battery_charge_limit is 50, whereas raw.invertor.battery_discharge_limit is 60.

From what I can tell setChargeRate in write.py is preventing this value being set to anything over 50.

britkat1980, 2024-11-15:
The spec says that the (dis)charge_limit raw register value is between 0-50. So in theory 60 is an illegal value... If you set it to 3256w in the Giv portal do you actually get that power output?

cigriggs, 2024-11-15:
I do appear to get close to that value. Here are the peaks rates shown in the app from yesterday.

[image]

It could be a coincidence, but I see from the datasheet for the Giv-Bat 5.12 Gen that the Charging / Discharging Current is 60A, compared to 50A for the older Giv-Bat 5.2 Gen 1.

jonnybsquared, 2025-07-10:
+1 for this - my 6kW inverter (I'm using a Polar, but it's all GivEnergy(ish) inside) can only go as high as 2560W in GivTCP. It's possible to change this on the portal to 5814W and it achieves it - until GivTCP sends it an instruction and it's limited back to 2560W (same for both charge and discharge

britkat1980, 2026-10-11:
Closing this as it's had no activity for over 8 weeks.

GivTCP 3.6 has since been released ([latest: v3.6.2](https://github.com/britkat1980/giv_tcp/releases/latest)). It moves to a new Modbus library and includes a large number of fixes, so this may well behave differently now. If you still see this on 3.6.2, please reopen it, or open a new issue, with a debug log.

closed 2026-10-11 by britkat1980