britkat1980/giv_tcp #306: Battery cell inconsistencies

LeeFarm11, 2024-11-17:
Hi. 

There are a couple of potential issues here. 
1. Cells 13 - 24 on one of my batteries showing 0. Is this a problem with one of my batteries? Or a problem with REST/GivTCP? Or neither? Or both?
2. temperatures in cells 13 - 24 not being reported.

In REST output in the "raw" section, I noticed that one of my 6 batteries is showing temperature 0 on cells 13 - 24. The other batteries are showing normal temperatures in cells 13 - 24. t_cell_13 through t_cell_24.

Then I realised that Home Assistant only shows temperatures of cells 1 -12. But shows voltage of cells 1 - 24.

Full REST output file here -
[1.txt](https://github.com/user-attachments/files/17790690/1.txt)

RAW data shows temperatures in all batteries in all cells 1-24 except in one battery showing 0 for 13 - 24. -
[image]

Battery Details section does not show any temperatures in any cell 13 - 24 in any battery -
[image]

Home screen agrees with REST Battery details section. It shows voltages for cells 1 -24, but temperatures for only cells 1 - 12.
[image]

britkat1980, 2024-11-22:
I've modified GivTCP to now expose all 24 temperature sensors. (coming in Dev3.0.46 and above)

The missing data is probably an inverter/battery problem, given the others are giving data

LeeFarm11, 2025-02-19:
Hi @britkat1980 I am now running Dev 3.0.50 but I still don't see Battery Cell Temperature for Cells 13 - 24 in HA. Or do I need to do something special to expose them?
Thanks

LeeFarm11, 2025-03-01:
Interesting to see that @trevordowling also sees one of his 6 batteries with temperature 0 on cells 13 to 24. He has posted a set of readings here -

https://github.com/springfall2008/batpred/issues/2060#issuecomment-2692330077

trevordowling, 2025-03-01:
@LeeFarm11 I've been told by Paul (GivEnergy R&D) that there is only one temperature sensor for two battery cells, so though we have 24 defined, only expect half to actually report data. 

Obviously they should not be defined, but hey...

Trevor

LeeFarm11, 2025-03-02:
> [@LeeFarm11](https://github.com/LeeFarm11) I've been told by Paul (GivEnergy R&D) that there is only one temperature sensor for two battery cells, so though we have 24 defined, only expect half to actually report data.
> 
> Obviously they should not be defined, but hey...
> 

??
If you look at your readings you will see that ..

your battery ....1 has 24 non zero readings populated.
your battery ....2 has 24 non zero readings populated.
your battery ....3 has 24 non zero readings populated.
your battery ....4 has 24 non zero readings populated.
your battery ....5 has 12 non zero readings populated. **Cells 13 to 24 show 0.**
your battery ....6 has 24 non zero readings populated.

The evidence does not support that statement about one sensor per 2 cells.

Take a look at the values you have. Do you find any pattern to support that there is one sensor per 2 cells?

I have the same apparent problem that you have -
One (only 1) of the 6 batteries has temperatures missing for cells 13 to 24.
The other 5 batteries have 24 seemingly independent temperatures for each cell.

trevordowling, 2025-03-02:
Interesting, I'd stopped looking/worrying about this once told the info
about 1 per 2 cells...

I'll review and give thought about this.

Not sure why Paul's information would be  incorrect, but possible